MKFirmware/MK4duo #2: Dual X Carriage moves both extruder

jgauchia, 2016-09-08:
Well, without errors compiling the code I've a new issue.
Both extruders moves together with tool 0 selected or tool 1 selected.
I Think that if I've a dual X carriage I've a independent extruder movement, but doesn't it.

I've tried all carriage modes but doesn't works.
With Marlin firmware, when I made X homming or movement, each extruder works independent.

jgauchia, 2016-09-08:
If you need a video comparing both firmware , please tell me

MagoKimbra, 2016-09-08:
I better .. I do not know this function, and I always copied unchanged from Marlin, maybe I forgot something. But if I see I can understand ..

jgauchia, 2016-09-08:
OK, Thanks

Movement with Marlin [https://youtu.be/sCzA_iUaKNU](url)
Movement wiht MK4Duo [https://youtu.be/msvS4nuF2wc](url)

Sorry for the video quality

MagoKimbra, 2016-09-08:
Do you have DXC_DUPLICATION_MODE??

jgauchia, 2016-09-08:
I don't find this string on code.
I'm current use this mode
#define DEFAULT_DUAL_X_CARRIAGE_MODE 0 
The same mode in both firmware Marlin and MK4Duo

jgauchia, 2016-09-08:
Sorry 
#define DEFAULT_DUAL_X_CARRIAGE_MODE 1

MagoKimbra, 2016-09-08:
Aaa ok... Now understand... I fix this!!!

jgauchia, 2016-09-08:
Thanks!
I'm testing NEXTION now

MagoKimbra, 2016-09-09:
Please test now...

jgauchia, 2016-09-09:
Still the same...
With #define DEFAULT_DUAL_X_CARRIAGE_MODE 1

MagoKimbra, 2016-09-09:
I tried it and it works ... So I thought that there was something in the settings, I finally found ...
You have also enabled X_DUAL_STEPPER_DRIVERS, what good is it if you have two engines on X in the classic operation. Now I put a check that warns that you should not put both.
Remove it from your file overall ...

jgauchia, 2016-09-09:
Thanks!
I'll try now ..

jgauchia, 2016-09-09:
I think that works... 
I'll print something and tell you .. ok?
(I need to change feedrates,etc..)

jgauchia, 2016-09-09:
Well, 
Need to fix some issues on my printer with your firmware.
But seems good!

jgauchia, 2016-09-16:
Seems works well, Close issue?

jgauchia, 2016-09-24:
Works

closed 2016-09-24 by jgauchia