MKFirmware/MK4duo #6: Feature: Nextion GFX preview of print

jgauchia, 2016-09-09:
Hi,
It's possible to make preview of print with more zoom?

jgauchia, 2016-12-05:
@MagoKimbra It's possible that you add this feature in this last version . 4.3.08 ¿?

MagoKimbra, 2016-12-05:
I add M996 S<scale> in new version...
M995 X Y Z for set origin.
For now this function is manual and not automatic...

jgauchia, 2016-12-05:
WOW!!! @MagoKimbra  you're the best!.
I'm printing now, but tomorrow i'll test the new firmware. 
Thank you!

MagoKimbra, 2016-12-05:
For example if you print a cube 20x20 in center cartesian (100,100) if you scale 4 the cube is out of display...
in start gcode after G28 (clear graphics display) you put
M995 X70 Y70
M996 S4
and the angle LOW SX is 70 70 and cube is big in the center...
Sorry for my bad english...

jgauchia, 2016-12-05:
@MagoKimbra No problem for your English, It isn't my native language too, (I'm from Spain)...  I understand, thanks! :)

closed 2016-12-17 by MagoKimbra