MKFirmware/MK4duo #13: SD Card issues

jgauchia, 2016-10-09:
SD Card is detected and files listed with M20 command.
In nextion Screen SD Files are shown but doesn't read anything.

jgauchia, 2016-10-10:
In console with M20 , M21 , M34 Gcode and file name works without problems.

jgauchia, 2016-10-10:
When I'll try to enter Menu folder always select and print restart.gcode

Echo:enqueueing "M23 restart.gcode"
Echo:enqueueing "M24"
File opened: restart.gcode Size: 1
File selected
Done printing file
Echo:enqueueing "M84 X Y Z E"
Echo:enqueueing "M31"
Echo:Print time 0s
Error:Unknown command: "M"

Letaw, 2016-10-12:
The LCD screen, press the encoder SD init. card is on the LCD screen and the computer files also become visible.
If you run the initialization of the card from the computer (command M21), the SD card is not.

16:55:16.403 : Echo:enqueueing "M21"
16:55:16.513 : SD card ok
16:55:16.513 : N1737 M20*35
16:55:16.528 : Begin file list
16:55:16.528 : 2 zwetaredblue.gcode
16:55:16.528 : Music/
16:55:16.528 : Music/imperial_march.gcode
16:55:16.528 : End file list
16:55:16.528 : ok
16:55:28.476 : N1749 M21*43
16:55:28.476 : N1750 M20*34
16:55:28.476 : N1751 M20*35
16:55:28.585 : Error:SD init fail
16:55:28.585 : ok
16:55:28.585 : Begin file list
16:55:28.585 : End file list
16:55:28.585 : ok
16:55:28.585 : Begin file list
16:55:28.585 : End file list
16:55:28.585 : ok

solder ramps 1.4 with arduino due.
That is the problem with the does not include programm pullup encoder EN1. EN2,ENC, had to solder the resistors to the legs encoder. LCD screen gadget3d.

jgauchia, 2016-10-12:
I don't have Encoders , I've a Nextion LCD ,it's a touch LCD Screen,
The SD card init correctly and files are visible on PC. In LCD are visible too.

jgauchia, 2016-12-05:
Close, works well with last version of firmware.

closed 2016-12-05 by jgauchia