MKFirmware/MK4duo #19: Compile issue (DUAL X EXTRUDER)

jgauchia, 2016-10-23:
In file included from sketch/src/../base.h:90:0,
                 from sketch/src/MK_Main.cpp:31:
sketch/src/MK_Main.cpp: In function 'bool prepare_move_to_destination_dualx()':
sketch/src/MK_Main.cpp:10322:30: error: 'inactive_extruder_x_pos' was not declared in this scope
           LOGICAL_X_POSITION(inactive_extruder_x_pos),
                              ^
sketch/src/../src/MK_Main.h:132:38: note: in definition of macro 'LOGICAL_POSITION'
 #define LOGICAL_POSITION(POS, AXIS) (POS + home_offset[AXIS] + position_shift[AXIS])
                                      ^
sketch/src/MK_Main.cpp:10322:11: note: in expansion of macro 'LOGICAL_X_POSITION'
           LOGICAL_X_POSITION(inactive_extruder_x_pos),
           ^
exit status 1
Error compilación en tarjeta Arduino/Genuino Mega or Mega 2560.

MagoKimbra, 2016-10-23:
Fix

closed 2016-10-23 by MagoKimbra

jgauchia, 2016-10-23:
Still errors

sketch/src/temperature/temperature.cpp: In function 'Temperature::manage_temp_controller() [clone .part.4]':
sketch/src/temperature/temperature.cpp:1111:1: error: unable to find a register to spill in class 'POINTER_REGS'
 }
 ^
sketch/src/temperature/temperature.cpp:1111:1: error: this is the insn:
(insn 72 70 74 3 (set (reg:SF 95 [ D.2291 ])
        (mem:SF (post_inc:HI (reg:HI 122 [ ivtmp.88 ])) [11 MEM[base: _94, offset: 0B]+0 S4 A8])) sketch/src/temperature/temperature.cpp:698 100 {*movsf}
     (expr_list:REG_INC (reg:HI 122 [ ivtmp.88 ])
        (nil)))
sketch/src/temperature/temperature.cpp:1111: confused by earlier errors, bailing out
lto-wrapper: /home/jordi/arduino-1.6.11/hardware/tools/avr/bin/avr-gcc returned 1 exit status
/home/jordi/arduino-1.6.11/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld: lto-wrapper failed
collect2: error: ld returned 1 exit status
exit status 1
Error compilación en tarjeta Arduino/Genuino Mega or Mega 2560.