MKFirmware/MK4duo #20: Compile issue

jgauchia, 2016-10-23:
sketch/src/temperature/temperature.cpp: In function 'Temperature::manage_temp_controller() [clone .part.4]':
sketch/src/temperature/temperature.cpp:1111:1: error: unable to find a register to spill in class 'POINTER_REGS'
 }
 ^
sketch/src/temperature/temperature.cpp:1111:1: error: this is the insn:
(insn 72 70 74 3 (set (reg:SF 95 [ D.2291 ])
        (mem:SF (post_inc:HI (reg:HI 122 [ ivtmp.88 ])) [11 MEM[base: _94, offset: 0B]+0 S4 A8])) sketch/src/temperature/temperature.cpp:698 100 {*movsf}
     (expr_list:REG_INC (reg:HI 122 [ ivtmp.88 ])
        (nil)))
sketch/src/temperature/temperature.cpp:1111: confused by earlier errors, bailing out
lto-wrapper: /home/jordi/arduino-1.6.11/hardware/tools/avr/bin/avr-gcc returned 1 exit status
/home/jordi/arduino-1.6.11/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld: lto-wrapper failed
collect2: error: ld returned 1 exit status
exit status 1
Error compilación en tarjeta Arduino/Genuino Mega or Mega 2560.

MagoKimbra, 2016-10-23:
This is a bug ARDUINO IDE... Do you have 2 or more extruders?

jgauchia, 2016-10-23:
Yes I've 2 extruders.
MK4duo_4.3.01_dev compiles with no problems.

MagoKimbra, 2016-10-23:
In version 4.3.1 temperature.h was not a class, now it is. If you compile for the AVR chip by mistake, if you compile for SAM with the same settings no. In fact by an error in the tools/avr /....

jgauchia, 2016-10-24:
Ok, thank you... I'll try to find info about this bug.

jgauchia, 2016-11-04:
Compiled correct with arduino 1.6.12
Close issue

closed 2016-11-04 by jgauchia