MKFirmware/MK4duo #21: Help needed - Defining E1, E2 and E3 as Z axis

hello3dprintingworld, 2016-10-27:
Hi!

I would like to set my 3 extra extruders ofmy ultratronics to be Z axis so I can have 4 motors dealing with the massive weight of a bed I am building. How can I do it?
Would be awesome if you had in configurations.h a stepper to axis set up. It would open the doors to more exotic setups!

apphead, 2016-12-30:
Override pins in configuration_overall.h

MagoKimbra, 2016-12-30:
No for now you can not put four engines to Z, only 2.
Soon I will add this option.!

apphead, 2016-12-30:
Sorry, completely misunderstood your question. Since they all run in sync, an external board might be your best option. 
Ugly but working (for those without extra stepper pins) : https://hackadaycom.files.wordpress.com/2016/06/mg_0129-comp.jpg

closed 2017-05-26 by MagoKimbra