MKFirmware/MK4duo #22: auto tool changer

hexxter, 2016-10-31:
Hi,

i will build a printer with an auto tool changer like this:
https://www.youtube.com/watch?v=unsQ9i-nBNc
What is the best way do get the gcode in MK4duo to put a tool by side and get the other tool. I found in the nozzle a feature "park" by if i understand it right it is not what i need.

desc:
first print head get a tool (level switch) for auto bed leveling
then change this tool to print with e0
then park e0 (e0 and e1 on the left head side) and move to e1 to print....

BR Hexxter

you make a nice work!

ekaggrat, 2016-11-07:
just randomly found your work. any place where you are sharing the details of the extruder head?

hexxter, 2016-11-07:
Hi,
i share it on thingiverse but today i`m building my printer. I think i publish it next week.
I will write the link here if its done.
BR Hexxter

hexxter, 2016-11-18:
Hi,

the first demo of the tool changer:
https://youtu.be/JVR-gH7dRq0

BR Hexxter

MagoKimbra, 2016-11-22:
Very good!!!

closed 2017-05-26 by MagoKimbra