MKFirmware/MK4duo #28: Compile error

vabendai, 2016-11-22:
Ciao Mago
I have this error building the latest version of MKDUO. 
What can be?
thanks

> In file included from C:\Users\io\AppData\Local\Temp\arduino_build_211046\sketch\base.h:115:0,
> 
>                  from C:\Users\io\Desktop\MK4duo\MK4duo.ino:231:
> 
> C:\Users\io\AppData\Local\Temp\arduino_build_211046\sketch\src/nextion/Nextion_lcd.h:59:71: error: default argument given for parameter 2 of 'void lcd_setstatus(const char*, bool)' [-fpermissive]
> 
>      void lcd_setstatus(const char* message, const bool persist = false);
> 
>                                                                        ^
> 
> In file included from C:\Users\io\AppData\Local\Temp\arduino_build_211046\sketch\base.h:113:0,
> 
>                  from C:\Users\io\Desktop\MK4duo\MK4duo.ino:231:
> 
> C:\Users\io\AppData\Local\Temp\arduino_build_211046\sketch\src/lcd/ultralcd.h:41:8: error: after previous specification in 'void lcd_setstatus(const char*, bool)' [-fpermissive]
> 
>    void lcd_setstatus(const char* message, const bool persist=false);
> 
>         ^
> 
> In file included from C:\Users\io\AppData\Local\Temp\arduino_build_211046\sketch\base.h:115:0,
> 
>                  from C:\Users\io\Desktop\MK4duo\MK4duo.ino:231:
> 
> C:\Users\io\AppData\Local\Temp\arduino_build_211046\sketch\src/nextion/Nextion_lcd.h:60:71: error: default argument given for parameter 2 of 'void lcd_setstatuspgm(const char*, uint8_t)' [-fpermissive]
> 
>      void lcd_setstatuspgm(const char* message, const uint8_t level = 0);
> 
>                                                                        ^
> 
> In file included from C:\Users\io\AppData\Local\Temp\arduino_build_211046\sketch\base.h:113:0,
> 
>                  from C:\Users\io\Desktop\MK4duo\MK4duo.ino:231:
> 
> C:\Users\io\AppData\Local\Temp\arduino_build_211046\sketch\src/lcd/ultralcd.h:42:8: error: after previous specification in 'void lcd_setstatuspgm(const char*, uint8_t)' [-fpermissive]
> 
>    void lcd_setstatuspgm(const char* message, const uint8_t level=0);
> 
>         ^
> 
> In file included from C:\Users\io\AppData\Local\Temp\arduino_build_211046\sketch\base.h:115:0,
> 
>                  from C:\Users\io\Desktop\MK4duo\MK4duo.ino:231:
> 
> C:\Users\io\AppData\Local\Temp\arduino_build_211046\sketch\src/nextion/Nextion_lcd.h: In function 'bool lcd_detected()':
> 
> C:\Users\io\AppData\Local\Temp\arduino_build_211046\sketch\src/nextion/Nextion_lcd.h:94:23: error: redefinition of 'bool lcd_detected()'
> 
>      FORCE_INLINE bool lcd_detected(void) { return true; }
> 
>                        ^
> 
> In file included from C:\Users\io\AppData\Local\Temp\arduino_build_211046\sketch\base.h:113:0,
> 
>                  from C:\Users\io\Desktop\MK4duo\MK4duo.ino:231:
> 
> C:\Users\io\AppData\Local\Temp\arduino_build_211046\sketch\src/lcd/ultralcd.h:52:17: error: 'bool lcd_detected()' previously defined here
> 
>      inline bool lcd_detected() { return true; }
>

MagoKimbra, 2016-11-22:
Please send me your overall.h i test this...

vabendai, 2016-11-22:
[Configuration_Overall.zip](https://github.com/MagoKimbra/MK4duo/files/606429/Configuration_Overall.zip)

MagoKimbra, 2016-11-22:
Ok Sorry, the problem is on configurator online, i fix it this night...
For now open overll and put // on #define U8GLIB_SH1106

`
//#define SAV_3DGLCD
//#define U8GLIB_SSD1306
//#define U8GLIB_SH1106
//#define SAV_3DLCD
`

closed 2016-11-22 by vabendai