MKFirmware/MK4duo #35: Temperature status LEDs compiling error

Dougwar, 2016-12-05:
`Arduino: 1.6.9 (Windows 7), Placa:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\src\../src/HAL_AVR/HAL.h:56:0,

                 from sketch\src\../base.h:86,

                 from sketch\src\MK_Main.cpp:31:

sketch\src\MK_Main.cpp: In function 'void handle_status_leds()':

sketch\src\../src/HAL_AVR/fastio.h:66:36: error: 'DIOSTAT_LED_RED_RPORT' was not declared in this scope

 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)

                                    ^

sketch\src\../src/HAL_AVR/fastio.h:89:27: note: in expansion of macro '_WRITE'

 #define WRITE(IO, v)      _WRITE(IO, v)

                           ^

sketch\src\MK_Main.cpp:11187:9: note: in expansion of macro 'WRITE'

         WRITE(STAT_LED_RED, new_led ? HIGH : LOW);

         ^

sketch\src\../src/HAL_AVR/fastio.h:55:43: error: 'DIOSTAT_LED_RED_WPORT' was not declared in this scope

                                          {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } \

                                           ^

sketch\src\../src/HAL_AVR/fastio.h:66:80: note: in expansion of macro '_WRITE_C'

 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)

                                                                                ^

sketch\src\../src/HAL_AVR/fastio.h:89:27: note: in expansion of macro '_WRITE'

 #define WRITE(IO, v)      _WRITE(IO, v)

                           ^

sketch\src\MK_Main.cpp:11187:9: note: in expansion of macro 'WRITE'

         WRITE(STAT_LED_RED, new_led ? HIGH : LOW);

         ^

In file included from sketch\src\../src/HAL_AVR/HAL.h:56:0,

                 from sketch\src\../base.h:86,

                 from sketch\src\MK_Main.cpp:31:

sketch\src\../src/HAL_AVR/fastio.h:55:72: error: 'DIOSTAT_LED_RED_PIN' was not declared in this scope

                                          {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } \

                                                                        ^

sketch\src\../src/HAL_AVR/fastio.h:38:28: note: in definition of macro 'MASK'

   #define MASK(PIN)  (1 << PIN)

                            ^

[... 432 lines cut ...]
sketch\src\../src/HAL_AVR/fastio.h:66:105: note: in expansion of macro '_WRITE_NC'

 #define _WRITE(IO, v)  do {  if (&(DIO ##  IO ## _RPORT) >= (uint8_t *)0x100) {_WRITE_C(IO, v); } else {_WRITE_NC(IO, v); }; } while (0)

                                                                                                         ^

sketch\src\../src/HAL_AVR/fastio.h:89:27: note: in expansion of macro '_WRITE'

 #define WRITE(IO, v)      _WRITE(IO, v)

                           ^

sketch\src\MK_Main.cpp:11188:9: note: in expansion of macro 'WRITE'

         WRITE(STAT_LED_BLUE, new_led ? LOW : HIGH);

         ^

exit status 1
Erro compilando para a placa Arduino/Genuino Mega or Mega 2560

Este relatório teria mais informações com
"Mostrar a saida detalhada durante a compilação"
opção pode ser ativada em "Arquivo -> Preferências"
`

MagoKimbra, 2016-12-05:
but you have define pins STAT_LED_RED and STAT_LED_BLUE??
This pin for ramps 1.4 are defined only for viki display!!

Dougwar, 2016-12-05:
Where can I define the pins I can't find it in the 33.h file
I'm trying to do this:
https://www.youtube.com/watch?v=Qmg2IL8AQ6w

Dougwar, 2016-12-05:
https://www.youtube.com/watch?v=bVbpAuM9XcU

MagoKimbra, 2016-12-05:
In configuration_pins.h write

#define STAT_LED_RED_PIN  32
#define STAT_LED_BLUE_PIN 35

Or put pin you use...

Dougwar, 2016-12-06:
thanks Alberto

Dougwar, 2016-12-10:
Any chances to get Green to work?

closed 2016-12-17 by MagoKimbra