MKFirmware/MK4duo #39: RESET

Tick007, 2016-12-08:
Mine burned DUE (dead mega16u2 + Rx uart_0) doesn't  restart on connect"(may be just mine board, Ï'm not sure) then work not via UART_0. 
I propose to add folowing:
//Defines so the device can do a self reset
//https://forum.arduino.cc/index.php?topic=345209.0
#define SYSRESETREQ (1<<2)
#define VECTKEY (0x05fa0000UL)
#define VECTKEY_MASK (0x0000ffffUL)
#define AIRCR ((uint32_t)0xe000ed0cUL) // fixed arch-defined address
#define REQUEST_EXTERNAL_RESET (AIRCR=(AIRCR&VECTKEY_MASK)|VECTKEY|SYSRESETREQ)

REQUEST_EXTERNAL_RESET can be for example call from MK_Main.cpp after FlushSerialRequestResend() in inline void gcode_M999(). 

I'm no sure that someone except me need this, but this feture gives apportunity to reset device by button (in pronterface, by M999 command). Never understand why stock button "Reset" in pronterface interface doesn't reset the board. Or does ?

apphead, 2016-12-27:
I use Radds as well. Today I switched from programming port to native port (baud set to 430600). Reset used to be working fine on disconnect / reconnect, Reset from pronterface did not do anything as far as I can remember. Since native port doesn't support reset, the question is: how is your due connected, via programming or native port?

Tick007, 2016-12-27:
As my programming port is dead, I have to connect from pronterface via UART-3 (with TTL converter). I've added the code above, but looks like it wasn't good Idea (or realization of idea), because during that kind of reset motors moves a little (that is not happen than the board resets during connect/disconnect via programing port).

Tick007, 2016-12-27:
[RAMPS-FD-Schematic.pdf](https://github.com/MagoKimbra/MK4duo/files/674551/RAMPS-FD-Schematic.pdf)
[image]

apphead, 2016-12-27:
Why don't you plug it in the native port? Set Serial_port to -1 and use 430600 baud. Works way more stable with Repetier-Server on RPi... 
PS: my technical skills are insufficient to really grasp your recommendations in the first post...

Tick007, 2016-12-27:
I did't new about native port and 430600. I have to try. All previus attemps to connect via native port left with no result. Searching the internet about it - the same.
I have no enouth technical skills too, I've just found the way in the internet how to reset DUE from code.

apphead, 2016-12-27:
I tried several times as well, today I finally succeeded 🙃 baud can be set to a lower value but why would you, so far it's stable in my setup... 
Bear in mind that there is no reset, if it doesn't come online again, you must press the physical button on the due board. Let me know if it works!

closed 2017-05-26 by MagoKimbra