MKFirmware/MK4duo #42: NEXTION Fan speed info

jgauchia, 2016-12-10:
When print with the second extruder only, layer fan info in Nextion LCD always shown 0%

MagoKimbra, 2016-12-10:
The fan info is fan for PLA not extruder. If fan is off because your print ABS or other is normally its write 0%.

jgauchia, 2016-12-10:
Ok, I'm always print with PLA, If I print with the main extruder , the info it's shown, if I "modify" GCODE file for print with second extruder, this info not appears.

MagoKimbra, 2016-12-10:
The fun go or not?

jgauchia, 2016-12-10:
yes yes, fan its ok, only the info on Nextion screen, nothing "serious". It's only for your info.

MagoKimbra, 2016-12-10:
It is strange, the fan is unic and variable is unic. Ok i check this...

jgauchia, 2016-12-10:
Yes, so strange, Ok don't worry.... 
In my printer both fans are in same output... doesn't matter..

jgauchia, 2016-12-10:
Ah! I'm printing with SD file

closed 2016-12-17 by MagoKimbra