MKFirmware/MK4duo #50: ABL not working in 4.3.09_dev

apphead, 2016-12-17:
There are at least two issues:

- G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G29 ;auto level
Resulting in console out:
12:20:53.365: G29 Auto Bed Leveling
12:20:53.369: ABL: Off (2)
12:20:53.369: Reset Bed Level
12:22:23.028: Echo:Bilinear Leveling Grid:
12:22:23.028: Echo: 0 1 2
12:22:23.032: Echo: 0 -1.11 -0.68 -0.26
12:22:23.032: Echo: 1 -0.83 -0.23 +0.36
12:22:23.036: Echo: 2 -0.56 +0.16 +0.81
12:22:23.040: X:196.000000 Y:280.000000 Z:NAN E:0.000000 Count X:39200 Y:56000 Z:26235
12:22:23.048: X:196.000000 Y:280.000000 Z:NAN E:0.000000 Count X:39200 Y:56000 Z:0
- Issue 1:
Why does it say ABL: Off? I have these settings:
#define AUTO_BED_LEVELING_FEATURE
#define AUTO_BED_LEVELING_BILINEAR
- - If I enable ABL with M320, I get
12:33:55.471: ABL: On
12:33:55.477: ABL Fade Height = 0.00
Then another G28, I get
12:34:30.060: ABL: Off
- Should a M320 be included in the Gcode before every print? Am I missing something?

- Issue 2:
G29 returns NAN for Z:
12:22:23.040: X:196.000000 Y:280.000000 Z:NAN E:0.000000 Count X:39200 Y:56000 Z:26235
As a result, prints start somewhere above the bed.

MagoKimbra, 2016-12-17:
Yes, now the values are saved in EEPROM memory.
To enable or disable the abl using the G320 or G321 command, but immediately after G28, because the home off the abl to have the right values of X, Y and Z.
G29 instead makes the abl and activates it.

apphead, 2016-12-17:
So, does that mean that EEPROM is mandatory if I wanted to use ABL? That would render RADDS unusable unless you add EEPROM like I described https://github.com/MagoKimbra/MK4duo/issues/30 . Dunno if EEPROM storage works with SD card, that would be another option.
If EEPROM is mandatory for G29, then I would suggest to add it to sanitycheck...

It turns out that my EEPROM was not responding when I tested the above. Now G28 G29 works, value for Z is not NAN anymore.

May I suggest you M117 "ABL: on" after G29? That would make things clear.

MagoKimbra, 2016-12-17:
No no, if do not use EEPROM everything remains as before.
G28
G29.
I just have to check out because after G29 is a Z = nan ...
I'm checking...

apphead, 2016-12-17:
On a another note: I cannot imagine how much time and effort you put into this project. And then there are folks like me who start opening issues which takes up more of your valueable time. Since I depend on using MK4Due on my RADDS (Repetier with ABL is a pain in the ass, so I stopped using it altogether), I am willing to offer my services as a (certified) Software Tester to you. Meaning: I know how to test and most of all, I know how important it is to describe the issue with all relevant information, so you can fix them in as little time wasted with the setup as possible. Are you interested?

MagoKimbra, 2016-12-17:
What do you mean by certified testers?

apphead, 2016-12-17:
http://www.istqb.org/certification-path-root/advanced-level/advanced-level-in-a-nutshell.html
I have passed all three modules Manager, Analyst, Technical Test Analyst
So I know a thing or two about proper testing.... (and used to work as such for more than 3 years...)

MagoKimbra, 2016-12-17:
A certificate ok in that sense, I know you wanted to be paid. I do not speak English very well and sometimes get it wrong ..

apphead, 2016-12-17:
NOOO, I dont want to be paid at all. I speak german and french too. Despite my last name being Italian (you are, right?), I dont speak Italian....

apphead, 2016-12-17:
Like I said, I am a maker like you and I appreciate your efforts A LOT. If I can be of assistance, I can give you a hand with testing (and limited Programming Skills)

MagoKimbra, 2016-12-17:
Ok, for me any help is essential also soloper test. I have not all the types of cards and all types of machines with their changes and sensors. So many things I can not try them if they go.
If texts and give me feedback to me is just less work.

MagoKimbra, 2016-12-17:
Now i fix G29 and Z=nan i think!!!
try please

apphead, 2016-12-17:
Z=NAN is gone with your latest commit
But ABL: Off is still confusing. It should say ABL: On after G29

* 14:22:24.729: G29 Auto Bed Leveling
14:22:24.733: ABL: Off
14:23:49.029: Echo:Bilinear Leveling Grid:
14:23:49.033: Echo: 0 1 2
14:23:49.033: Echo: 0 -1.00 -0.64 -0.34
14:23:49.037: Echo: 1 -0.72 -0.18 +0.27
14:23:49.037: Echo: 2 -0.47 +0.13 +5.49
14:23:49.046: X:196.000000 Y:280.000000 Z:12.093595 E:0.000000 Count X:39200 Y:56000 Z:27348

MagoKimbra, 2016-12-17:
Ok.. For now put M320 after G29...
I think how to fix it

MagoKimbra, 2016-12-17:
I fixed... Now write ABL: On when finish!!

apphead, 2016-12-17:
Perfect. Will try to test it tomorrow. Have a nice day!

MagoKimbra, 2016-12-17:
I fixed the command M320 Enabled/Disabled ABL
M320 S0 Disabled
M320 S1 Enabled
M320 V Verbose print grid
M321 Set manual valor z in grid X Y or in current position.

apphead, 2016-12-19:
- SENDING:G29 V4
G29 Auto Bed Leveling
ABL: Off
Echo:Bilinear Leveling Grid:
Echo:     0     1     2
Echo: 0 -1.92 -1.45 -1.02
Echo: 1 -1.62 -0.99 -0.39
Echo: 2 -1.36 -0.63 +0.08
ABL: On
-> during prints, the z-axis is moving to correct the plane.

- Also the following commands are working, logs from repetier server console:
M320 S0 -> log: ABL: Off
M320 S1 -> log: ABL: On
M320 V -> log:
12:30:01.062: Echo:Bilinear Leveling Grid:
12:30:01.062: Echo: 0 1 2
12:30:01.066: Echo: 0 -1.16 -0.73 -0.32
12:30:01.067: Echo: 1 -0.80 -0.20 +0.38
12:30:01.070: Echo: 2 -0.50 +0.19 +0.89

closed 2016-12-30 by apphead