MKFirmware/MK4duo #52: DIOpin_DDR not declared

TimNe0, 2016-12-19:
Hi, got the problem when compiling - am using for a K40 laser, have used fresh download, LCD is Discount Full display, laser 2 pin PWM high.  Apologies, not usually a git hub contributor :(

 
from sketch\src\laser\laser_avr\laser.cpp:43:

sketch\src\laser\laser_avr\laser.cpp: In function 'void timer3_init(int)':

sketch\src\laser\laser_avr\../../../src/HAL/HAL_AVR/fastio.h:74:29: error: 'DIOpin_DDR' was not declared in this scope

 #define _SET_OUTPUT(IO) do {DIO ##  IO ## _DDR |=  MASK(DIO ## IO ## _PIN); } while (0)

                             ^

sketch\src\laser\laser_avr\../../../src/HAL/HAL_AVR/fastio.h:97:27: note: in expansion of macro '_SET_OUTPUT'

 #define SET_OUTPUT(IO)    _SET_OUTPUT(IO)

                           ^

sketch\src\laser\laser_avr\laser.cpp:125:7: note: in expansion of macro 'SET_OUTPUT'

       SET_OUTPUT(pin);

       ^

In file included from sketch\src\laser\laser_avr\../../../src/HAL/HAL_AVR/HAL.h:56:0,

                 from sketch\src\laser\laser_avr\../../../src/HAL/HAL.h:58,

                 from sketch\src\laser\laser_avr\../../../base.h:77,

                 from sketch\src\laser\laser_avr\laser.cpp:43:

sketch\src\laser\laser_avr\../../../src/HAL/HAL_AVR/fastio.h:74:57: error: 'DIOpin_PIN' was not declared in this scope

 #define _SET_OUTPUT(IO) do {DIO ##  IO ## _DDR |=  MASK(DIO ## IO ## _PIN); } while (0)

                                                         ^

sketch\src\laser\laser_avr\../../../src/HAL/HAL_AVR/fastio.h:38:28: note: in definition of macro 'MASK'

   #define MASK(PIN)  (1 << PIN)

                            ^

sketch\src\laser\laser_avr\../../../src/HAL/HAL_AVR/fastio.h:97:27: note: in expansion of macro '_SET_OUTPUT'

 #define SET_OUTPUT(IO)    _SET_OUTPUT(IO)

                           ^

sketch\src\laser\laser_avr\laser.cpp:125:7: note: in expansion of macro 'SET_OUTPUT'

       SET_OUTPUT(pin);

       ^

In file included from sketch\src\laser\laser_avr\../../../src/HAL/HAL_AVR/HAL.h:56:0,

                 from sketch\src\laser\laser_avr\../../../src/HAL/HAL.h:58,

                 from sketch\src\laser\laser_avr\../../../base.h:77,

                 from sketch\src\laser\laser_avr\laser.cpp:43:

sketch\src\laser\laser_avr\laser.cpp: In function 'void timer4_init(int)':

sketch\src\laser\laser_avr\../../../src/HAL/HAL_AVR/fastio.h:74:29: error: 'DIOpin_DDR' was not declared in this scope

 #define _SET_OUTPUT(IO) do {DIO ##  IO ## _DDR |=  MASK(DIO ## IO ## _PIN); } while (0)

                             ^

sketch\src\laser\laser_avr\../../../src/HAL/HAL_AVR/fastio.h:97:27: note: in expansion of macro '_SET_OUTPUT'

 #define SET_OUTPUT(IO)    _SET_OUTPUT(IO)

                           ^

sketch\src\laser\laser_avr\laser.cpp:165:7: note: in expansion of macro 'SET_OUTPUT'

       SET_OUTPUT(pin);

       ^

In file included from sketch\src\laser\laser_avr\../../../src/HAL/HAL_AVR/HAL.h:56:0,

                 from sketch\src\laser\laser_avr\../../../src/HAL/HAL.h:58,

                 from sketch\src\laser\laser_avr\../../../base.h:77,

                 from sketch\src\laser\laser_avr\laser.cpp:43:

sketch\src\laser\laser_avr\../../../src/HAL/HAL_AVR/fastio.h:74:57: error: 'DIOpin_PIN' was not declared in this scope

 #define _SET_OUTPUT(IO) do {DIO ##  IO ## _DDR |=  MASK(DIO ## IO ## _PIN); } while (0)

                                                         ^

sketch\src\laser\laser_avr\../../../src/HAL/HAL_AVR/fastio.h:38:28: note: in definition of macro 'MASK'

   #define MASK(PIN)  (1 << PIN)

                            ^

sketch\src\laser\laser_avr\../../../src/HAL/HAL_AVR/fastio.h:97:27: note: in expansion of macro '_SET_OUTPUT'

 #define SET_OUTPUT(IO)    _SET_OUTPUT(IO)

                           ^

sketch\src\laser\laser_avr\laser.cpp:165:7: note: in expansion of macro 'SET_OUTPUT'

       SET_OUTPUT(pin);

       ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

MagoKimbra, 2016-12-19:
Have you set pin for laser pwm e ttl?

TimNe0, 2016-12-20:
ttl pin? I've set it as 6, PWR PIN 5

Is there another setting I need to change?

TimNe0, 2016-12-23:
Thanks Prinz12 - I've spent about three weeks trying to get it to work, though MagoKimbra has fixed loads of bugs.  I used to have the "laser only" version working just fine, but lost the firmware and accidentally uploaded a duff version, so hackspace is without a laser (not that they really had one to begin with!)

closed 2016-12-23 by MagoKimbra

TimNe0, 2016-12-23:
> Closed #52 <https://github.com/MagoKimbra/MK4duo/issues/52> via a2aedd0

Compiled clean. Can't upload till 3rd though. Cheers for the Christmas
present and have a good holiday/new year!

MagoKimbra, 2016-12-23:
I do not understand "Can't upload till 3rd though".
Merry Christmas to you too

TimNe0, 2016-12-23:
Laser & RAMPS 1.4 board are 150 miles away and I won't be back there until
3rd - I left a few days ago. Will be good to get it working again! :)

commit a2aedd09 (2016-12-23): Fix #52