MKFirmware/MK4duo #53: Support for M851 Set Z-Probe Offset

apphead, 2016-12-20:
According to http://reprap.org/wiki/G-code#M851:_Set_Z-Probe_Offset , MK4Duo supports M851, but that is not correct. M666 is used instead, which is actually used for delta endstop adjustment. The P parameter is not standard either.
Even though things work just fine the way they are, I suggest to use M851 for probe offset

apphead, 2016-12-30:
Since every firmware has its on flavour of gcode, I close the ticket.

MagoKimbra, 2016-12-30:
I'm sorry, I had not seen this. I have put M666 H command long before put him Marlin. Marlin has now opted for the M851 command, but most people who use MK4duo now use M666. I could also put the M851 command, so that it is compatible.

closed 2017-05-26 by MagoKimbra