MKFirmware/MK4duo #55: Two ENDSTOPS at Z-axis

Sanitar777, 2016-12-20:
Hello! 
Thanks for you work.

I have H-bote/CoreXY with Z dual STEPPER Drivers and Z dual EndStop at MAX positions.

If I don't put the parameter Z dual EndStop, everything works fine but of course does not work Z2_ENDSTOP.
If I put the parameter Z dual EndStop, I can't choose on which PIN to connect it.

Which PIN is assigned Z2_ENDSTOP by default? 
I have it connected as X_MAX_PIN. How do I configure the firmware that it would work correctly?

Thank you.

Sanitar777, 2017-01-04:
Hello MagoKimbra!
Thanks for your work!

I understand how to do so that it all started to work. I made some changes in the files.

1. I have a specific PIN to use by Z2_MAX_PIN in **Configuration_Overall.h.** In my case it is X_MAX_PIN.
```cpp
#define Z2_MAX_PIN ORIG_X_MAX_PIN //this is the PIN 34 on the RADDS by default for X_MAX_PIN
```
2. In **endstops.cpp** changed from "!performing_homing" to "!stepper.performing_homing"
```cpp
#if ENABLED(Z_DUAL_ENDSTOPS)

  //Pass the result of the endstop test
  void Endstops::test_dual_z_endstops(EndstopEnum es1, EndstopEnum es2) {
    byte z_test = TEST_ENDSTOP(es1) | (TEST_ENDSTOP(es2) << 1); // bit 0 for Z, bit 1 for Z2
    if (z_test && stepper.current_block->steps[Z_AXIS] > 0) {
      SBI(endstop_hit_bits, Z_MIN);
      if (!stepper.performing_homing || (z_test == 0x3))  //if not performing home or if both endstops were trigged during homing...
        stepper.kill_current_block();
    }
  }

#endif
```
3. In **stepper.cpp** I swapped "**<**"
```cpp
#if ENABLED(Z_DUAL_STEPPER_DRIVERS)
  #define Z_APPLY_DIR(v,Q) { Z_DIR_WRITE(v); Z2_DIR_WRITE(v); }
  #if ENABLED(Z_DUAL_ENDSTOPS)
    #define Z_APPLY_STEP(v,Q) \
    if (performing_homing) { \
      if (Z_HOME_DIR > 0) {\
        if (!(TEST(endstops.old_endstop_bits, Z_MAX) && (count_direction[Z_AXIS] > 0)) && !locked_z_motor) Z_STEP_WRITE(v); \
        if (!(TEST(endstops.old_endstop_bits, Z2_MAX) && (count_direction[Z_AXIS] > 0)) && !locked_z2_motor) Z2_STEP_WRITE(v); \
      } \
      else { \
        if (!(TEST(endstops.old_endstop_bits, Z_MIN) && (count_direction[Z_AXIS] < 0)) && !locked_z_motor) Z_STEP_WRITE(v); \
        if (!(TEST(endstops.old_endstop_bits, Z2_MIN) && (count_direction[Z_AXIS] < 0)) && !locked_z2_motor) Z2_STEP_WRITE(v); \
      } \
    } \
    else { \
      Z_STEP_WRITE(v); \
      Z2_STEP_WRITE(v); \
    }
  #else
    #define Z_APPLY_STEP(v,Q) do{ Z_STEP_WRITE(v); Z2_STEP_WRITE(v); }while(0)
  #endif
#else
  #define Z_APPLY_DIR(v,Q) Z_DIR_WRITE(v)
  #define Z_APPLY_STEP(v,Q) Z_STEP_WRITE(v)
#endif
```
I hope to see these changes in your firmware ;-)
Thanks.
Good luck!

MagoKimbra, 2017-01-04:
Ok for endstop.cpp i fixed, but i do not understand the change in stepper.cpp. I do not see change.

Sanitar777, 2017-01-04:
Excuse me, please. It's my fault. 
I changed this in other experiments. 
Now it is not necessary to change, you're right.

Now it remains to decide how to define a PIN for Z2 ENDSTOP in the Configurator.

Sanitar777, 2017-01-11:
> Now it remains to decide how to define a PIN for Z2 ENDSTOP in the Configurator.

Thank you.
I'll check today.

Sanitar777, 2017-01-12:
Hello Mago!

I was not able to compile the firmware. A lot of warnings. I didn't take the risk and flash it. To investigate these warnings I'll be late. 

Which version of Arduino SAM Boards should I use?

Thanks.

MagoKimbra, 2017-01-12:
Disable the warnings from the Arduino IDE settings putting on anyone.
They are warning of double definition, do not create any problem.
I use Arduino IDE 1.8, the latest version.
I compiled perfectly with your overall no problems.

Sanitar777, 2017-01-12:
Mago!

Now, all tested. 
All END STOPS works correctly. 
I close the issue.

Thank you very much!

closed 2017-01-12 by Sanitar777