MKFirmware/MK4duo #59: Laser turns on unintentionally / Raster engraving

Prinz12, 2016-12-28:
Hello, there is probably a problem with the laser. 
I use Arduino Mega 2560 with Ramps 1.4. I use the 1 pin (D5) configuration. 
All other settings are unchanged. Although the laser is deactivated, it always returns after a few seconds. 
Raster engraving also seems to have a bug. When engraving I get only lines. I use the Inkscape plugin. With another Marlinfirmware it goes without problems. 
Sorry for the computer translation.

nextime, 2017-02-14:
Can you please attach:
 * an example image and gcode generated by the inkscape plugin?
 * the complete config of the firmware?

I need to re-test and refresh the laser config as i'm using an old revision of the firmware, and many things are changed from the old MarlinKimbra, maybe something has been broken, but if you provide those examples i can test it with your exact configuration

Also:
 * which inkscape plugin are you using?
 * Are you using it on a K40 laser cutter or it's another machine?

Aaron-c, 2017-02-20:
I have a similar issue as well.  RAMPS1.4 Delta configuration, with a LED attached where the hotend normally goes.

If I do a G1 move the laser fires fine, the next G0 move after that fires the laser continuously for the move.  
The cycle repeats itself.

Prinz12, 2017-02-23:
Sorry, but I moved to the Arduino Due and the error is not there. The Mega was the problem but independent of a Gcode. Once the Mega has started the laser goes on. After the command M5, it took a few seconds and he went back to.

Prinz12, 2017-02-23:
With the Due there is unfortunately another problem. If I make a Vector cut with Arcs make the laser after synonymous with a G0 command. Only the insertion of an M649 S0 command prevents it. The M3 and M5 commands no longer work in the DEV version. I have activated laserbeam and cncrouter.

nextime, 2017-02-24:
Ok, so, probably the issue on the mega+ramps was the serial buffer size, i had to reconfigure it to 256byte to make it working as expected.

can you send me an example gcode of where you are experiencing the issue so i can test it?

closed 2017-05-26 by MagoKimbra