MKFirmware/MK4duo #60: LIN_Advance creates long delays during print

apphead, 2016-12-31:
I am not sure if my findings are specific to my setup. I use repetier server on a raspberry pi to connect to RADDS via native port (i.e. fast baud rate 460300, hardly any corrupted or lost data). I read elsewhere (https://forum.e3d-online.com/index.php?threads/new-linear-advance-extrusion-algorithm-significant-print-quality-increases.1893/#post-22135) that LIN_Advance might be restricted to printing from SDCard due to the read speed (!?)
From http://marlinfw.org/docs/features/lin_advance.html I read

> Keep in mind that enabling this feature will add extra load to the processor and also your extruder hardware. I recommend to stay at 115200 baud communication speed, specially if you get error messages in your print host software or you get wired movements of your print head. Ensure your print host software is using line numbers and checksums, this is disabled by default in Simplify3D.

Now, when I activated LIN_Advance and set K to 75, my printer started to stutter. Interestingly, the amount of time the printhead paused moving was almost always the same:

- Test Run 4 (benchy3d.stl)
X93.613 Y161.51 Z2.0
15:15:34.123: ok N10787 P0 B0
15:17:16.123: busy:processing (51)
15:17:16.402: ok N10788 P0 B0

- X123.115 Y151.186 Z2.6
15:18:02.743: ok N12753 P0 B0
15:19:44.744: busy:processing (51)
15:19:45.081: ok N12754 P0 B0

- X125.88 Y154.81 Z4.40
15:23:14.352: ok N20922 P0 B0
15:24:56.353: busy:processing (51)
15:24:56.621: ok N20923 P0 B0

- X83.3 Y156.35 Z4.80
15:25:27.928: ok N22411 P0 B0
15:27:09.929: busy:processing (51)
15:27:10.231: ok N22412 P0 B0

- 15:28:45.465: ok N26009 P0 B0
15:30:25.466: busy:processing (50)
15:30:25.685: ok N26010 P0 B0

To me, that does not look like a communication / bandwidth issue, but more as if the delay stems from the code itself. I will run another test with slower baud rate and see if the delay stays the same....

Maybe this is completely unrelated, but I noticed that in HAL.h, it says 
`define REFERENCE_F_CPU 16000000 // 16MHz MEGA2560[]`
but I could not find REFERENCE_F_CPU for Arduino Due (84 MHz) or other boards.

Macgyver001, 2017-02-01:
Hello all,

I run on release:MK4duo_4.3.1 on a arduino Due (radds 1.5) configuration core XY system. With external motor drivers. (I need 1.5us step timing) to save some clock pulses I put #define STEP_PULSE_CYCLES 110 //((MINIMUM_STEPPER_PULSE) * CYCLES_PER_MICROSECOND) (conditionals_post.h)
Set  115200 baud communication speed and enabled Lin_advance with factor 75.

I have tested cubes and they run fine.
When printing circles I notice that large circles are ok, diameter 25 mm or so. 
I have enabled the slicer such that it prints the infill with the same pattern as the outer dimension, so it keeps on printer smaller circles toward the middle of the circle to fill the layer.
You can see that the circles getting less smooth towards the middle and at some point my printer stalls.

Is this related to the issue mentioned above?
Is there some way to diagnose?

apphead, 2017-02-01:
Maybe this has been fixed since 4.3.1. I will run some tests with 4.3.11 dev later this week and post back the results.

Macgyver001, 2017-02-02:
Hi Apphead,

I tried the latest release 4.3.11 and it does better, but it still stalls at some point after about 20 minutes, compared to a few minutes in release 4.3.1. What has changed with release 4.3.11 compared to 4.3.1? Perhaps there is some optimization?

apphead, 2017-02-02:
Hi MacGyver. That's an interesting (and annoying) finding....I would rather have it stall after 20 seconds for testing purposes. But still, I will start a dry run with circular patterns (walls AND infill) in order to shed some light on the matter. Weekend is coming soon, so I should have a couple of hours for testing...
As for the changes with 4.3.11, I have as much or little understanding from the related commit https://github.com/MagoKimbra/MK4duo/commit/ff79cb71d24a2cc49b43c11fe20a5b4a06cf3eff . I hope @MagoKimbra follows this thread as well.

apphead, 2017-02-08:
If LIN_ADVANCE is enabled (LIN_ADVANCE_K 75) then my extruder stops moving entirely. I have a dual extruder setup, maybe that is the reason for the issue. Needs further testing.

Macgyver001, 2017-02-15:
Apphead, I  have a single extruder system, and have the same problem, so not related to the dual extruder setup I gess.

psavva, 2017-02-17:
Hi, please also see the same issue on the official Marlin Firmware:
https://github.com/MarlinFirmware/Marlin/issues/5699

MagoKimbra, 2017-02-17:
Yes i see...

apphead, 2017-02-24:
Still no improvement here. Now X and Y (and maybe Z too) don't move if Advance is enabled....

It is said [here](https://github.com/MarlinFirmware/Marlin/issues/3867) that slic3r now includes some kind of advance feature (no build number given, but josef prusa's team keeps pushing development of slic3r [here](https://github.com/prusa3d/Slic3r/releases) )

closed 2017-05-26 by MagoKimbra