MKFirmware/MK4duo #67: SPI support

brupje, 2017-01-23:
Hi,

Thanks for merging my pull request so quickly. I saw my message somehow got lost. I'd like to check with you how SPI is supposed to work on SAM boards. 

As fas as I know, the SAM chip can only accept pins 4,10 and 52 for hardware SPI. However, in pins.h line 94 also 59,60 and 77 are accepted. Which result in not using software SPI and SDSUPPORT does indeed not work until I enable software SPI for the Ultratronics board. 

Also in HAL.cpp for SAM, SS_PIN is directly written to for multiple functions. To me this seems incorrect as in temperature.cpp these functions are used for the MAX** chips. This will cause interference when using thermo couples and SD cards at the same time.

Can you confirm my conclusions?

MagoKimbra, 2017-01-23:
For Example Alligator board use pin 77 for SS_PIN and this is hardware...

MagoKimbra, 2017-01-23:
In theory yes ... Unfortunately I have no way to prove that it is so ...

brupje, 2017-01-23:
Hmm if I read the code correctly, the alligator board is doing hardware SPI. I guess if the hardware pin is not connected to anything, and you pull another CS pin low, it should still work. I will test that tomorrow.

MagoKimbra, 2017-01-23:
Hi, right now I'm at home with a high fever and I'm not much with his head.

In alligator the SPI_PIN is 77 and chan 0 or not?

#elif defined (ARDUINO_ARCH_SAM)
  #if (SDSS == 4) || (SDSS == 10) || (SDSS == 52)|| (SDSS == 59) || (SDSS == 60) || (SDSS == 77)
    #if (SDSS == 4)
      #define SPI_PIN         87
      #define SPI_CHAN        1
    #elif (SDSS == 10)
      #define SPI_PIN         77
      #define SPI_CHAN        0
    #elif (SDSS == 52)
      #define SPI_PIN         86
      #define SPI_CHAN        2
    #elif (SDSS == 59)
      #if defined(REPRAPWORLD_GRAPHICAL_LCD) && MB(ULTRATRONICS)
        #define DUE_SOFTWARE_SPI
      #else
        #define SPI_PIN       59
        #define SPI_CHAN      1
      #endif
    #elif (SDSS == 60)
      #define SPI_PIN         60
      #define SPI_CHAN        1
    #else
      #define SPI_PIN         77
      #define SPI_CHAN        0
    #endif
    #define MISO_PIN          74
    #define MOSI_PIN          75
    #define SCK_PIN           76
  #else
    #define DUE_SOFTWARE_SPI
    #define MOSI_PIN		      51
    #define MISO_PIN		      50
    #define SCK_PIN 		      52
  #endif

brupje, 2017-01-23:
That's unfortunate. Hope you get better soon!
In 503.h, SDSS is defined as 
#define SDSS                  77

So yes, #define SPI_PIN 77
#define SPI_CHAN 0

In hal.cpp, there are some alligator specific things going on.

In 503.h the hardware pins for SPI (4,10,52) are also defined for different purposes, so my theory should be false, otherwise it would interfere with those pins.

brupje, 2017-01-24:
I have got the SD card working with hardware SPI after a lot of testing. I am still not sure how it's supposed to work, though. I don't have an alligator board to test with, but I don't see why it wouldn't give the same error I had, an SD card init error upon start. The second M21 command does init the card correctly though. 

When I change SPI_PIN to a not connected pin, e.g. 4, the hardware SPU will work with that one. And if another CS pin happens to be LOW on the same SPI bus, that will work it appears. So I have changed pins.h, line 105 to:

#if defined(REPRAPWORLD_GRAPHICAL_LCD) && MB(ULTRATRONICS)
        #define DUE_SOFTWARE_SPI
      #else
        #define SPI_PIN       4 // Not connected
        #define SPI_CHAN      1
      #endif

I will check if other SPI devices work as well.

brupje, 2017-01-24:
I have send you a pull request. It works for me to use either hardware SPI or software SPI on UIltratronics with these commits.  Check if commit  c3ce285 doesn't break anything. Alternatively in HAL.cpp, between lines 150 and 204 all references to SS_PIN should be removed. I don't think they serve any purpose other than causing problems anyway, so maybe it's better to remove them. But I am not sure that will break anything in any use case.

setar, 2017-02-09:
hi 
check please the https://github.com/MagoKimbra/MK4duo/issues/32

u8g lib not supported hardware SPI in st7920 on Arduino Due
and firmware not have support for mixing Hardware and software SPI...

brupje, 2017-02-09:
> u8g lib not supported hardware SPI in st7920 on SAM3X8E (Arduino Due)
True, therefore I needed this to work. So with my patch you can use hardware SPI unless you have the graphical LCD, in which case you need to use software SPI. Unfortunately, I don't have time to fix u8glib to support hw SPI on DUE.

>  P.S. EEPROM function on ultratronics not working, i think it also wrong SPI mode
Ultratronics does not have EEPROM. You need to save settings to the SD card or internal flash.

setar, 2017-02-09:
ok, 
see please the link https://github.com/olikraus/u8glib/wiki/userreference#sethardwarebackup
may be use hardware backup method it good way for mixing hardware and software spi ?
```
  // set SPI backup if required
  //u8g.setHardwareBackup(u8g_backup_avr_spi);
```

sh4jonny, 2017-02-16:
can you help me with sdcard on RAMPS4DUE board ? how i must connect it? SD detect work for me, but only NO SD in menu ((((

MagoKimbra, 2017-02-16:
Ramps FD??
It is set to pin 4 SDSS
#define SDSS                     4

sh4jonny, 2017-02-16:
not FD. Hacked RAMPS1.4 for DUE
RAMPS4DUE
433 board

MagoKimbra, 2017-02-16:
A ok sorry...
The ramps4due use pin 53, but it is not a hardware spi then uses software spi ...
#define SDSS                    53

sh4jonny, 2017-02-16:
yes i have this define. also a i have 
#define MISO_PIN				74
#define MOSI_PIN				75
#define SCK_PIN					76
but they are replaced on 51 50 52 from pins.h
it as correct?

MagoKimbra, 2017-02-16:
yes

sh4jonny, 2017-02-16:
so i must wire my cardreader to aux3 ? can i use hardware spi from 74-75-76 pins ?

closed 2017-05-26 by MagoKimbra