MKFirmware/MK4duo #72: RGB_LED and TEMP_STAT_LEDS

Dougwar, 2017-02-10:
Hi, my RGB LED only get white, and M150 Rxxx Uxxx Bxxx doesn't work,
Wiring
VCC : +
#if ENABLED(RGB_LED)
  #define RGB_LED_R_PIN 4
  #define RGB_LED_G_PIN 5
  #define RGB_LED_B_PIN 6
#endif

MagoKimbra, 2017-02-10:
Do you set RGB_LED?
#define RGB_LED

Dougwar, 2017-02-10:
yes
/**************************************************************************
 ******************************** RGB LED *********************************
 **************************************************************************
 *                                                                        *
 * Support for an RGB LED using 3 separate pins with optional PWM         *
 *                                                                        *
 **************************************************************************/
#define RGB_LED
/**************************************************************************/

Dougwar, 2017-02-10:
but even with no pin setted the led get white

MagoKimbra, 2017-02-10:
You are configurate fw with configutator on line or not?

Dougwar, 2017-02-10:
not, I'm using Arduino IDE 1.6.13

MagoKimbra, 2017-02-10:
Ok i found the problem.. One minute..

MagoKimbra, 2017-02-10:
Ok try now!

Dougwar, 2017-02-11:
The RGB_LED now is working but TEMP_STAT_LEDS doesnt work.

closed 2017-05-26 by MagoKimbra