MKFirmware/MK4duo #83: Auto power on for ATX

setar, 2017-02-20:
The feature request:
If power is ATX and switched off, i think need doing power on before any move or temperature manipulation.

MagoKimbra, 2017-02-20:
Ok... Try it..

MagoKimbra, 2017-02-21:
Try MK4duo dev

setar, 2017-02-22:
```
/Users/setar/Documents/Arduino/build/sketch/src/MK_Main.cpp: In function 'void manage_inactivity(bool)':
/Users/setar/Documents/Arduino/build/sketch/src/MK_Main.cpp:12529:10: error: 'powersupply' was not declared in this scope
     if (!powersupply) powerManager.check(); // Check Power
          ^
Using library LiquidCrystal at version 1.0.5 in folder: /Applications/reprap/Arduino.app/Contents/Java/libraries/LiquidCrystal 
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
```

MagoKimbra, 2017-02-22:
Ops sorry my error..

closed 2017-02-22 by MagoKimbra

setar, 2017-02-22:
need insert a long pause for power out stabilizing (on my APX  ~5sec )

MagoKimbra, 2017-02-22:
Ok i add delay 5000 ms, try it...

setar, 2017-02-22:
strange, but pause not working.
if i doing G28 move attempt start at the same time with ps on

setar, 2017-02-22:
ok, put check to G28 cycle - maximum power consumption its first move z axis (for me)

MagoKimbra, 2017-02-22:
ok Try now...

setar, 2017-02-22:
Now its work fine
Thanks  :)

MagoKimbra, 2017-02-22:
My pleasure!!
Now i put #define DELAY_AFTER_POWER_ON 5000L // In millisecond
in configuration.h

commit cd6653f0 (2017-02-22): Fix #83