MKFirmware/MK4duo #84: FeatureRequest M600 Temperature

Dougwar, 2017-02-21:
Hi, can you check this feature? 
https://github.com/MarlinFirmware/Marlin/pull/5737

It makes the temperature to drop before retract and clean up the hotend during unload.
 
It is possible?
Thanks

MagoKimbra, 2017-02-21:
Ok i check!!!

MagoKimbra, 2017-02-21:
I did not understand the request .. That change already is in MK4duo!!!

Dougwar, 2017-02-21:
The idea is before the retract the hotend drops temperature, about 160C for abs, it makes all plastic to come out and after load again the hotend get the programmed temperature.  I tried the MK4duo and nothing happens.

MagoKimbra, 2017-02-21:
I can not understand, why go down to 160C before the filament retraction?

Dougwar, 2017-02-21:
something like this:
https://goo.gl/QXJ3wT

closed 2017-03-01 by MagoKimbra