MKFirmware/MK4duo #87: compile error with 4.3.12 dev

vabendai, 2017-02-25:
Ciao Mago
I have a compile error:

> C:\Users\io\AppData\Local\Temp\arduino_build_475957\sketch\src\lcd\ultralcd.cpp: In function 'void status_printf(uint8_t, const char*, ...)':
> 
> C:\Users\io\AppData\Local\Temp\arduino_build_475957\sketch\src\lcd\ultralcd.cpp:3455:64: error: 'vsnprintf_P' was not declared in this scope
> 
>    vsnprintf_P(lcd_status_message, 3 * (LCD_WIDTH), status, args);
> 
>                                                                 ^

[Configuration_Overall.zip](https://github.com/MagoKimbra/MK4duo/files/801397/Configuration_Overall.zip)

Can be my wrong configuration?

Thanks

yoch2015, 2017-02-26:
'Vsnprintf_P' is not declared.
If it is 'vsnprintf', you can compile it.
I do not know whether it is correct or not.

MagoKimbra, 2017-03-01:
It's fixed!!!

closed 2017-03-01 by MagoKimbra