MKFirmware/MK4duo #88: compile error branch:dev

setar, 2017-02-27:
current dev 9adbf4cb98fa5313897c8d1accffe07e11bb3d7b
has error:

`/Users/setar/Documents/Arduino/build/sketch/src/MK_Main.cpp: In function 'void gcode_M428()':
/Users/setar/Documents/Arduino/build/sketch/src/MK_Main.cpp:8612:111: error: invalid conversion from 'uint8_t {aka unsigned char}' to 'AxisEnum' [-fpermissive]
         float base = (current_position[i] > (soft_endstop_min[i] + soft_endstop_max[i]) / 2) ? base_home_pos(i) : 0,
                                                                                                               ^
/Users/setar/Documents/Arduino/build/sketch/src/MK_Main.cpp:577:33: error:   initializing argument 1 of 'float base_home_pos(AxisEnum)' [-fpermissive]
   XYZ_CONSTS_FROM_CONFIG(float, base_home_pos,  HOME_POS)
                                 ^
/Users/setar/Documents/Arduino/build/sketch/src/MK_Main.cpp:573:22: note: in definition of macro 'XYZ_CONSTS_FROM_CONFIG'
   static inline type array(AxisEnum axis) { return pgm_read_any(&array##_P[axis]); }
                      ^
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
`

MagoKimbra, 2017-02-27:
Please link your configuration for test...

setar, 2017-02-27:
[azteeg.patch.zip](https://github.com/MagoKimbra/MK4duo/files/802994/azteeg.patch.zip)

MagoKimbra, 2017-02-27:
The file is empty!!!

setar, 2017-02-27:
hmm, not empty.
ok code the patch:
`From e14df69e8d249b0f0a2a35f9b8a472b860955078 Mon Sep 17 00:00:00 2001
From: Sergey Taranenko <setar@roboforum.ru>
Date: Mon, 27 Feb 2017 11:10:39 +0300
Subject: [PATCH] 3ntProf hardware config

---
 MK4duo/Configuration_Basic.h       | 12 ++++++------
 MK4duo/Configuration_Core.h        | 40 +++++++++++++++++++-------------------
 MK4duo/Configuration_Feature.h     | 10 +++++-----
 MK4duo/Configuration_Temperature.h | 22 ++++++++++-----------
 4 files changed, 42 insertions(+), 42 deletions(-)

diff --git a/MK4duo/Configuration_Basic.h b/MK4duo/Configuration_Basic.h
index aebfa9d..890fe2e 100644
--- a/MK4duo/Configuration_Basic.h
+++ b/MK4duo/Configuration_Basic.h
@@ -59,7 +59,7 @@
  *
  * 2400,9600,19200,38400,57600,115200,250000
  */
-#define BAUDRATE 115200
+#define BAUDRATE 250000
 
 // Enable the Bluetooth serial interface
 //#define BLUETOOTH
@@ -69,11 +69,11 @@
 // User-specified version info of this build to display in [Pronterface, etc] terminal window during
 // startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
 // build by the user have been successfully uploaded into firmware.
-#define STRING_CONFIG_H_AUTHOR "(none, default config)"   // Who made the changes.
+#define STRING_CONFIG_H_AUTHOR "Trinity"   // Who made the changes.
 
 // Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
 // You can use an online service to generate a random UUID. (eg http://www.uuidgenerator.net/version4)
-#define MACHINE_UUID "00000000-0000-0000-0000-000000000000"
+#define MACHINE_UUID "EAEE364A-A5D0-4F3C-93C7-D8175A78E790"
 
 // Kill Method
 // 0 - Disable heaters, wait forever
@@ -112,7 +112,7 @@
  * See: https://github.com/MagoKimbra/MK4duo/blob/master/Documentation/Hardware.md *
  *                                                                                       *
  *****************************************************************************************/
-#define MOTHERBOARD BOARD_RAMPS_13_HFB
+#define MOTHERBOARD BOARD_AZTEEG_X3
 /*****************************************************************************************/
 
 
@@ -132,8 +132,8 @@
  * MAKERARM_SCARA - SCARA Makerfarm                                    *
  *                                                                     *
  ***********************************************************************/
-#define MECHANISM MECH_CARTESIAN
-//#define MECHANISM MECH_COREXY
+//#define MECHANISM MECH_CARTESIAN
+#define MECHANISM MECH_COREXY
 //#define MECHANISM MECH_COREYX
 //#define MECHANISM MECH_COREXZ
 //#define MECHANISM MECH_COREZX
diff --git a/MK4duo/Configuration_Core.h b/MK4duo/Configuration_Core.h
index fd3d497..90ec7ce 100644
--- a/MK4duo/Configuration_Core.h
+++ b/MK4duo/Configuration_Core.h
@@ -69,7 +69,7 @@
  * Displayed in the LCD "Ready" message.                                                 *
  *                                                                                       *
  *****************************************************************************************/
-#define CUSTOM_MACHINE_NAME "Core"
[... 186 lines cut ...]
 // shouldn't use bed PID until someone else verifies your hardware works.
 // If this is enabled, find your own PID constants below.
-//#define PIDTEMPBED
+#define PIDTEMPBED
 
 //#define BED_LIMIT_SWITCHING
 #define BED_HYSTERESIS        2 // Only disable heating if T>target+BED_HYSTERESIS and enable heating if T>target-BED_HYSTERESIS (works only if BED_LIMIT_SWITCHING is enabled)
@@ -329,11 +329,11 @@
 
 // 120v 250W silicone heater into 4mm borosilicate (MendelMax 1.5+)
 // from FOPDT model - kp=.39 Tp=405 Tdead=66, Tc set to 79.2, aggressive factor of .15 (vs .1, 1, 10)
-#define DEFAULT_bedKp   10.00
-#define DEFAULT_bedKi    0.1
-#define DEFAULT_bedKd  300.0
+#define DEFAULT_bedKp 60.70
+#define DEFAULT_bedKi 8.76
+#define DEFAULT_bedKd 105.17
 
-// FIND YOUR OWN: "M303 E-1 C8 S90" to run autotune on the bed at 90 degreesC for 8 cycles.
+// FIND YOUR OWN: "M303 H-1 C8 S90" to run autotune on the bed at 90 degreesC for 8 cycles.
 
 //#define PID_BED_DEBUG // Sends debug data to the serial port.
 /***********************************************************************/
-- 
2.6.4 (Apple Git-63)+GitX`

MagoKimbra, 2017-02-27:
But what .patch file?

setar, 2017-02-27:
easy to store different versions of firmware in different branches git (for my many printers)
and patch gives the cumulative changes from the main branch.

complete changed files : http://lab.trinitygroup.ru/owncloud/index.php/s/GUjVZ6Imf892EPv

Prinz12, 2017-02-27:
I have the same problem.
The only change is "#define MOTHERBOARDBOARD_RADDS" and I get following error message.

C:\Users\Michael\AppData\Local\Temp\arduino_build_193912\sketch\src\MK_Main.cpp: In function 'void gcode_M428()':

C:\Users\Michael\AppData\Local\Temp\arduino_build_193912\sketch\src\MK_Main.cpp:8602:111: error: invalid conversion from 'uint8_t {aka unsigned char}' to 'AxisEnum' [-fpermissive]

         float base = (current_position[i] > (soft_endstop_min[i] + soft_endstop_max[i]) / 2) ? base_home_pos(i) : 0,

                                                                                                               ^

C:\Users\Michael\AppData\Local\Temp\arduino_build_193912\sketch\src\MK_Main.cpp:576:33: error:   initializing argument 1 of 'float base_home_pos(AxisEnum)' [-fpermissive]

   XYZ_CONSTS_FROM_CONFIG(float, base_home_pos,  HOME_POS)

                                 ^

C:\Users\Michael\AppData\Local\Temp\arduino_build_193912\sketch\src\MK_Main.cpp:572:22: note: in definition of macro 'XYZ_CONSTS_FROM_CONFIG'

   static inline type array(AxisEnum axis) { return pgm_read_any(&array##_P[axis]); }

                      ^

Bibliothek Wire in Version 1.0 im Ordner: C:\Users\Michael\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.6\libraries\Wire  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Arduino Due (Native USB Port).

MagoKimbra, 2017-02-27:
Ok i check!!!

Prinz12, 2017-02-27:
Thanks, now it goes.

setar, 2017-02-28:
Work fine, Thanks

closed 2017-02-28 by setar