MKFirmware/MK4duo #94: MESH_BED_LEVELING compile error

sh4jonny, 2017-03-20:
hi! on current master branch i enable
#define MESH_BED_LEVELING
#define MANUAL_BED_LEVELING

and get this error
sketch\src\lcd\ultralcd.cpp: In function 'void _lcd_level_bed_get_z()':
sketch\src\lcd\ultralcd.cpp:1382:44: error: 'MANUAL_PROBE_Z_RANGE' was not declared in this scope
         NOLESS(current_position[Z_AXIS], -(MANUAL_PROBE_Z_RANGE) * 0.5);
                                                                   ^
sketch\src\lcd\../../src/macros.h:79:33: note: in definition of macro 'NOLESS'
 #define NOLESS(v,n) do{ if (v < n) v = n; }while(0)
                                                    ^
sketch\src\lcd\ultralcd.cpp:1383:43: error: 'MANUAL_PROBE_Z_RANGE' was not declared in this scope
         NOMORE(current_position[Z_AXIS], (MANUAL_PROBE_Z_RANGE) * 0.5);
                                                                    ^
sketch\src\lcd\../../src/macros.h:80:33: note: in definition of macro 'NOMORE'
 #define NOMORE(v,n) do{ if (v > n) v = n; }while(0)
                                                      ^
exit status 1

can you check it?

MagoKimbra, 2017-03-20:
MANUAL_PROBE_Z_RANGE is defined in configuration_cartesian or configuration_core.h

sh4jonny, 2017-03-20:
all features defined by manual in Configuration_Overall.h
with previous 4.3.14 build ok

MagoKimbra, 2017-03-20:
Ok change MESH_HOME_SEARCH_Z with MANUAL_PROBE_Z_RANGE!!!

sh4jonny, 2017-03-21:
compiled! thank you!

closed 2017-03-21 by sh4jonny