MKFirmware/MK4duo #95: compile error 4.3.15 dev

vabendai, 2017-03-21:
Hello Mago
I have a compile error.
This is my overhall
(https://github.com/MagoKimbra/MK4duo/files/859445/Configuration_Overall.zip)

Is it a problem on my configuration?

Thanks

```
C:\Users\io\AppData\Local\Temp\arduino_build_178471\sketch\src\lcd\ultralcd.cpp: In function 'void lcd_control_menu()':

C:\Users\io\AppData\Local\Temp\arduino_build_178471\sketch\src\lcd\ultralcd.cpp:2050:47: error: 'lcd_factory_settings' was not declared in this scope

     MENU_ITEM(function, MSG_RESTORE_FAILSAFE, lcd_factory_settings);

                                               ^

C:\Users\io\AppData\Local\Temp\arduino_build_178471\sketch\src\lcd\ultralcd.cpp:299:30: note: in definition of macro '_MENU_ITEM_PART_2'

         menu_action_ ## TYPE(__VA_ARGS__); \

                              ^

C:\Users\io\AppData\Local\Temp\arduino_build_178471\sketch\src\lcd\ultralcd.cpp:2050:5: note: in expansion of macro 'MENU_ITEM'

     MENU_ITEM(function, MSG_RESTORE_FAILSAFE, lcd_factory_settings);

     ^
```

MagoKimbra, 2017-03-21:
OK thanks. The problem is that you have not enabled the EEPROM. Now fix.
But I tell you that you can now activate the EEPROM because now with ultratronics saves on SD as if it were the EEPROM.

MagoKimbra, 2017-03-21:
Fixed!!!

closed 2017-03-21 by MagoKimbra

vabendai, 2017-03-22:
Hello Mago
thanks for your reply.

I succssfully compiled without eeprom but when i try to uncomment #define EEPROM_SETTINGS i have this error

```
C:\Users\io\AppData\Local\Temp\arduino_build_743380\sketch\src\eeprom\eeprom.cpp: In static member function 'static bool EEPROM::Store_Settings()':

C:\Users\io\AppData\Local\Temp\arduino_build_743380\sketch\src\eeprom\eeprom.cpp:282:9: error: return-statement with no value, in function returning 'bool' [-fpermissive]

         return;

         ^

C:\Users\io\AppData\Local\Temp\arduino_build_743380\sketch\src\eeprom\eeprom.cpp: In static member function 'static bool EEPROM::Load_Settings()':

C:\Users\io\AppData\Local\Temp\arduino_build_743380\sketch\src\eeprom\eeprom.cpp:559:9: error: return-statement with no value, in function returning 'bool' [-fpermissive]

         return;

         ^

```

Thanks