MKFirmware/MK4duo #104: Diretion Z

fabriciomoeller, 2017-04-08:
The direction of Z is not correct, it is reversed.

I have always used the signal from the minimal endStop as below:
#define Z_MIN_ENDSTOP_LOGIC true

And the direction of the Z motor to:
#define INVERT_Z_DIR true

And this always worked for my CoreXY

Now he is reversing the engine speed.

Even changing the logic is not working.

I use a 4mm inductive sensor for Z.

/**********************
 * Configuration_Core *
 **********************/
#define KNOWN_MECH
#define CUSTOM_MACHINE_NAME "COREXY"
#define CORE_FACTOR 1
#undef ENDSTOPPULLUPS
#define ENDSTOPPULLUP_XMIN
#define ENDSTOPPULLUP_YMIN
#define ENDSTOPPULLUP_ZMIN
//#define ENDSTOPPULLUP_Z2MIN
//#define ENDSTOPPULLUP_Z3MIN
//#define ENDSTOPPULLUP_Z4MIN
//#define ENDSTOPPULLUP_XMAX
//#define ENDSTOPPULLUP_YMAX
//#define ENDSTOPPULLUP_ZMAX
//#define ENDSTOPPULLUP_Z2MAX
//#define ENDSTOPPULLUP_Z3MAX
//#define ENDSTOPPULLUP_Z4MAX
//#define ENDSTOPPULLUP_ZPROBE
//#define ENDSTOPPULLUP_EMIN
#define X_MIN_ENDSTOP_LOGIC true
#define Y_MIN_ENDSTOP_LOGIC true
#define Z_MIN_ENDSTOP_LOGIC true
#define Z2_MIN_ENDSTOP_LOGIC true
#define Z3_MIN_ENDSTOP_LOGIC true
#define Z4_MIN_ENDSTOP_LOGIC false
#define X_MAX_ENDSTOP_LOGIC true
#define Y_MAX_ENDSTOP_LOGIC true
#define Z_MAX_ENDSTOP_LOGIC true
#define Z2_MAX_ENDSTOP_LOGIC true
#define Z3_MAX_ENDSTOP_LOGIC true
#define Z4_MAX_ENDSTOP_LOGIC true
#define Z_PROBE_ENDSTOP_LOGIC true
#define E_MIN_ENDSTOP_LOGIC true
//#define ENDSTOP_INTERRUPTS_FEATURE
#define Z_ENDSTOP_SERVO_NR -1
#define Z_ENDSTOP_SERVO_ANGLES {90,0} // Z Servo Deploy and Stow angles
#define Z_PROBE_FIX_MOUNTED
//#define BLTOUCH
//#define Z_PROBE_SLED
#define SLED_DOCKING_OFFSET 5
#define X_PROBE_OFFSET_FROM_NOZZLE -23
#define Y_PROBE_OFFSET_FROM_NOZZLE 0
#define Z_PROBE_OFFSET_FROM_NOZZLE -1
#define XY_PROBE_SPEED 8000
#define Z_PROBE_SPEED_FAST 200
#define Z_PROBE_SPEED_SLOW 100
//#define PROBE_DOUBLE_TOUCH
//#define Z_MIN_PROBE_REPEATABILITY_TEST
#define Z_PROBE_DEPLOY_HEIGHT 15
#define Z_PROBE_BETWEEN_HEIGHT 10
#define Z_PROBE_OFFSET_RANGE_MIN -50
#define Z_PROBE_OFFSET_RANGE_MAX  50
#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1
#define E_HOME_DIR -1
#define MIN_Z_HEIGHT_FOR_HOMING 0
#define X_ENABLE_ON 0
#define Y_ENABLE_ON 0
#define Z_ENABLE_ON 0
#define E_ENABLE_ON 0
#define INVERT_X_STEP_PIN false
#define INVERT_Y_STEP_PIN false
#define INVERT_Z_STEP_PIN false
#define INVERT_E_STEP_PIN false
#define INVERT_X_DIR true
#define INVERT_Y_DIR true
#define INVERT_Z_DIR true
#define INVERT_E0_DIR true
#define INVERT_E1_DIR true
#define INVERT_E2_DIR false
#define INVERT_E3_DIR false
#define INVERT_E4_DIR false
#define INVERT_E5_DIR false
#define DISABLE_X false
#define DISABLE_Y false
#define DISABLE_Z false
#define DISABLE_E false
//#define DISABLE_INACTIVE_EXTRUDER
#define X_MAX_POS 200
#define X_MIN_POS 0
#define Y_MAX_POS 200
#define Y_MIN_POS 0
#define Z_MAX_POS 200
#define Z_MIN_POS 0
#define E_MIN_POS 0
#define AXIS_RELATIVE_MODES {false, false, false, false}
#define Z_SAFE_HOMING
#define Z_SAFE_HOMING_X_POINT 100
#define Z_SAFE_HOMING_Y_POINT 100
//#define MESH_BED_LEVELING
#define MESH_INSET 10
#define MESH_NUM_X_POINTS 3
#define MESH_NUM_Y_POINTS 3
#define MESH_HOME_SEARCH_Z 5
//#define MESH_G28_REST_ORIGIN
//#define MANUAL_BED_LEVELING
#define MBL_Z_STEP 0.025
//#define AUTO_BED_LEVELING_3POINT
#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
//#define DEBUG_LEVELING_FEATURE
#define ABL_GRID_POINTS_X 3
#define ABL_GRID_POINTS_Y 3
#define LEFT_PROBE_BED_POSITION 25
#define RIGHT_PROBE_BED_POSITION 175
#define FRONT_PROBE_BED_POSITION 25
#define BACK_PROBE_BED_POSITION 175
#define MIN_PROBE_EDGE 10
//#define PROBE_Y_FIRST
//#define ABL_BILINEAR_SUBDIVISION
#define BILINEAR_SUBDIVISIONS 3
#define ABL_PROBE_PT_1_X 25
#define ABL_PROBE_PT_1_Y 175
#define ABL_PROBE_PT_2_X 25
#define ABL_PROBE_PT_2_Y 25
#define ABL_PROBE_PT_3_X 175
#define ABL_PROBE_PT_3_Y 25
//#define Z_PROBE_END_SCRIPT "G1 Z10 F8000\nG1 X10 Y10\nG1 Z0.5"
#define ENABLE_LEVELING_FADE_HEIGHT
//#define BED_CENTER_AT_0_0
//#define MANUAL_X_HOME_POS 0
//#define MANUAL_Y_HOME_POS 0
//#define MANUAL_Z_HOME_POS 0 // Distance between the nozzle to printbed after homing
#define DEFAULT_AXIS_STEPS_PER_UNIT {80,80,400,300,340/2,625,625,625,625}
#define DEFAULT_MAX_FEEDRATE {300,300,2,100,100,100,100,100,100}
#define MANUAL_FEEDRATE {100*60,100*60,2*60,60}
#define DEFAULT_MINIMUMFEEDRATE       0.0                       // minimum feedrate
#define DEFAULT_MINTRAVELFEEDRATE     0.0
#define MINIMUM_PLANNER_SPEED         0.05                      // (mm/sec)
#define DEFAULT_MAX_ACCELERATION {3000,3000,50,3000,3000,3000,3000,3000,3000}
#define DEFAULT_RETRACT_ACCELERATION {10000,10000,10000,10000,10000,10000}
#define DEFAULT_ACCELERATION 3000
#define DEFAULT_TRAVEL_ACCELERATION 3000
#define DEFAULT_XJERK 10
#define DEFAULT_YJERK 10
#define DEFAULT_ZJERK 0.4
#define DEFAULT_EJERK {5,5,5,5,5,5}
#define HOMING_FEEDRATE_X (100*60)
#define HOMING_FEEDRATE_Y (100*60)
#define HOMING_FEEDRATE_Z (2*60)
#define X_HOME_BUMP_MM 5
#define Y_HOME_BUMP_MM 5
#define Z_HOME_BUMP_MM 2
#define HOMING_BUMP_DIVISOR {5,5,2}
#define HOTEND_OFFSET_X {0,25,0,0}
#define HOTEND_OFFSET_Y {0,0,0,0}
#define HOTEND_OFFSET_Z {0,0,0,0}

MagoKimbra, 2017-04-08:
But what does not work exactly?

fabriciomoeller, 2017-04-08:
The direction is wrong, in reverse.

MagoKimbra, 2017-04-10:
And in normal??

fabriciomoeller, 2017-04-10:
No !!

MagoKimbra, 2017-04-10:
Check the driver o the cable to motor... If it goes in only one direction means that a coil is not working !!

fabriciomoeller, 2017-04-10:
I will test the cables, but it does not make sense, because with MarlinKimbra firware works.

MagoKimbra, 2017-04-10:
Ok check the endstop and probe with M119...

fabriciomoeller, 2017-04-10:
OK

fabriciomoeller, 2017-04-11:
I found the problem. It was Z_PROBE_DEPLOY_HEIGHT that was too high, giving a false impression that the movement was inverted.

closed 2017-05-26 by MagoKimbra