MKFirmware/MK4duo #106: eeprom

smagtech, 2017-04-10:
Hi 
Manual mesh bed leveling not saving in eeprom.

MagoKimbra, 2017-04-10:
Do you put M420 S1 for enabled MBL after G28.

smagtech, 2017-04-11:
Ok it is working after G28. but if i " Stop and Save for Restart " after restart print MBL is not working. 
and thanks for Wonderfull firmware. " Stop and Save for Restart " is great feature. Thank you once again.

smagtech, 2017-04-11:
i edeted some code in ( cardreder.cpp = line 400 ). 

    #if MECH(DELTA)
      fileRestart.write("G28\n");
    #else
      fileRestart.write(bufferCoord1);
      fileRestart.write("\n");
      fileRestart.write("G28 X Y\n");
	  fileRestart.write("M420 S1\n");   // I added this code.
    #endif
////////////////////////////////////////////////////////////////
and now
 restart code is =
G92 Z5
G28 X Y
M420 S1
T0
G92 E0
G1 E10 F300
G92 E0
G1 X0 Y120 Z0 F3600
G92 E0
M34 S585 @//mesh bed test.gcode

now it is working fine.

MagoKimbra, 2017-04-11:
I fix it...
Add

```
#if ENABLED(MESH_BED_LEVELING)
  if (mbl.active()) fileRestart.write("M420 S1\n");
#elif HAS(ABL)
  if (planner.abl_enabled) fileRestart.write("M320 S1\n");
#endif
```

smagtech, 2017-04-11:
Ok.
      but this code has to write first  =  M420 S1
then follow by this code                 =  G92 Z.....

otherwise G92 will 0 the Z axis. And M420 S1 will ( - ) the Z axis. if ther  ( - ) value in stoed in M420 S1....
 and nosel will crash into print.

smagtech, 2017-04-11:
And there is one more problem in.      current_position[Z_AXIS]. 

I "SSR" the print at  = 3.75mm. for test.  But the restart.gcode is =
/////////////////////////////////////////////////////////////
M420 S1
G92 Z8   <<<<<<<< this has to be "G92 Z8.75"
G28 X Y
T0
M109 T0 S270
M109 T1 S210
G92 E0
G1 E10 F300
G92 E0
G1 X104 Y67 Z3 F3600    <<<<<<<< this has to be  "G1 X104 Y67 Z3.75 F3600"
M106 S25 P0
G92 E2097
M34 S625336 @//skreper t2.gcode

MagoKimbra, 2017-04-11:
M420 S1 first the G28 It's useless, because g28 reset MBL or ABL
For next problem i check the code..

smagtech, 2017-04-12:
No I asked you is. M420 S1 then G92 Z......

smagtech, 2017-04-16:
Now it is working fine. after update restart.gcode is=

G92 Z6.950
G28 X Y
M420 S1
T0
G92 E0
G1 E10 F300
G92 E0
G1 X100.000 Y82.249 Z1.950 F3600
M106 S216 P0
G92 E83.495

M34 S179854 @////24by140.gcode
//////////////////////////////////////////////

Problem solved. 

Thank You.

smagtech, 2017-04-16:
But when i print with Heaters      "M420 S1" is missing. 
 
Printing gcode is=

G28 X Y
M420 S1
G92 Z4

M190 S50.000000 
M109 T0 S60.000000 
M104 T1 S70 

M106 S25 
G1 Z0.150
G1 F1800 E0.00000
G1 F2700 X92.047 Y65.535 E0.01317
G1 X92.381 Y65.145 E0.02594
G1 X92.753 Y64.753 E0.03937..................And so on.
///////////////////////////////////////////////////////////////
and restart.gocde is =.
                      <<<<<<<<<<<<<<< M420 S1 is missing
G92 Z5.450 
G28 X Y
M190 S50
T0
M109 T0 S60
M109 T1 S70
G92 E0
G1 E10 F300
G92 E0
G1 X90.917 Y65.364 Z0.450 F3600
M106 S142 P0          <<<<<<< <<<<And the Fan value is also wrong. it has to be "M106 S25 "
G92 E18.779

M34 S40094 @//24by140 with_heat.gcode

MagoKimbra, 2017-04-16:
check if you have the mesh grid in eeprom...

smagtech, 2017-04-17:
sorry i forget to "MBL" after update. now it is working fine.
restart.gode. is =
G92 Z8.300
G28 X Y
M420 S1
M190 S50
T0
M109 T0 S60
M109 T1 S70
G92 E0
G1 E10 F300
G92 E0
G1 X91.264 Y64.912 Z3.300 F3600
M106 S216 P0
G92 E147.590

M34 S321079 @//24by140 with_heat.gcode

Thank You.

MagoKimbra, 2017-04-17:
Thank you for info!!!

closed 2017-04-17 by MagoKimbra

smagtech, 2017-04-17:
I edited some code for this "restart.gcode" output  = 
G28 X Y              <<<<<<<<<<<<<<<<<
M420 S1              <<<<<<<<<<<<<<<<<
G92 Z7.400             <<<<<<<<<<<<<<<<<

M190 S50
T0
M109 T0 S60
M109 T1 S70
G92 E0
G1 E10 F300
G92 E0
G1 X108.580 Y77.606 Z2.400 F3600
M106 S216 P0
G92 E103.856

M34 S224162 @//24by140 with_heat.gcode

////////////////////////////////////////////////////////////////////////////////////////
cardreder.cpp      line 395=
////////////////////////////////////////////////////////////////////////////////////////
    #if MECH(DELTA)
      fileRestart.write("G28\n");
    #else
     // fileRestart.write(buffer_G92_Z);  <<<<<<<<<<<<<<<<<<< cut this
      fileRestart.write("G28 X Y\n");
    #endif

    #if ENABLED(MESH_BED_LEVELING)
      if (mbl.active()) fileRestart.write("M420 S1\n");
	  fileRestart.write(buffer_G92_Z);  //<<<<<<<<<<<<<<<< paste her
	  fileRestart.write("\n");                //<<<<<<<<<<<<<<<< Added this
    #elif HAS(ABL)
      if (planner.abl_enabled) fileRestart.write("M320 S1\n");
	  fileRestart.write(buffer_G92_Z); //<<<<<<<<<<<<<<<< paste her
	  fileRestart.write("\n");              //<<<<<<<<<<<<<<<< Added this
    #endif
///////////////////////////////////////////////////////////////////
out put is this starting code=
G28 X Y
M420 S1
G92 Z7.400. 

Thank You.

MagoKimbra, 2017-04-17:
Try now! I have fixed!

smagtech, 2017-04-17:
G28 X Y
M420 S1
G92 Z9.600

now it is perfect. 
Thank You for solving this problem.

smagtech, 2017-04-22:
I edited some more code for this "restart.gcode" output =

T0
M109 T0 S268
M109 T1 S210
G28 X Y
M420 S1
G92 Z39.000

G92 E0
G1 E10 F300
G92 E0
G1 X109.231 Y90.089 Z34.000 F500
M106 S25 P0
G92 E10072.640

M34 S6028412 @//////slider_left_side.gcode

////////////////////////////////////////////////////////////////////////////
cardreder.cpp  >> line 395. 
///////////////////////////////////////////////////////////////////////////
    fileRestart.open(&root, restart_name_File, O_WRITE);
    fileRestart.truncate(0);
//////>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>	 Paste
	    #if HAS(TEMP_BED)
      if (thermalManager.degTargetBed() > 0) {
        char Bedtemp[15];
        sprintf(Bedtemp, "M190 S%i\n", (int)thermalManager.degTargetBed());
        fileRestart.write(Bedtemp);
      }
    #endif

    char CurrHotend[10];
    sprintf(CurrHotend, "T%i\n", active_extruder);
    fileRestart.write(CurrHotend);

    for (uint8_t h = 0; h < HOTENDS; h++) {
      if (thermalManager.degTargetHotend(h) > 0) {
        char Hotendtemp[15];
        sprintf(Hotendtemp, "M109 T%i S%i\n", (int)h, (int)thermalManager.degTargetHotend(h));
        fileRestart.write(Hotendtemp);
      }
    }
////////////////////////////////////////////////////////////////////////
    #if MECH(DELTA)
      fileRestart.write("G28\n");
    #else
      fileRestart.write("G28 X Y\n");
    #endif

    #if ENABLED(MESH_BED_LEVELING)
      if (mbl.active()) fileRestart.write("M420 S1\n");
    #elif HAS(ABL)
      if (planner.abl_enabled) fileRestart.write("M320 S1\n");
    #endif

    fileRestart.write(buffer_G92_Z);
/* //////>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>     cut
    #if HAS(TEMP_BED)
      if (thermalManager.degTargetBed() > 0) {
        char Bedtemp[15];
        sprintf(Bedtemp, "M190 S%i\n", (int)thermalManager.degTargetBed());
        fileRestart.write(Bedtemp);
      }
    #endif

    char CurrHotend[10];
    sprintf(CurrHotend, "T%i\n", active_extruder);
    fileRestart.write(CurrHotend);

    for (uint8_t h = 0; h < HOTENDS; h++) {
      if (thermalManager.degTargetHotend(h) > 0) {
        char Hotendtemp[15];
        sprintf(Hotendtemp, "M109 T%i S%i\n", (int)h, (int)thermalManager.degTargetHotend(h));
        fileRestart.write(Hotendtemp);
      }
    }
 */
    fileRestart.write("G92 E0\nG1 E10 F300\nG92 E0\n");

///////////////////////////////////////////////////////////////////////////////////
In old restart.gcode 
stepper motor active till ''heater'' reaches temperarue. wasting energy.

I hope you agree. 

thank you.

MagoKimbra, 2017-04-22:
But if it gets warm before doing the home, do not you risk losing material on the piece?

smagtech, 2017-04-22:
no i modifed this also

    current_position[Z_AXIS] += 0.5;
    do_blocking_move_to_z(current_position[Z_AXIS]);
	
	 current_position[X_AXIS] -= 51;
    do_blocking_move_to_x(current_position[X_AXIS]);  //<< X_AXIS away from piece. 
//     I want to do "home Move" her but I dont no how.
	
	   	   current_position[Z_AXIS] += 4.5;
    do_blocking_move_to_z(current_position[Z_AXIS]);
/////////////////////////////////////////////////////////////////////////
and this
///////////////////////////////////////////////////////////////////////
MK_Main.cpp >> line 1000
/////////////////////////////////////////
  /**
   * Function for Stop e save
   */
  inline void sd_stop_e_save() {
    if (card.cardOK && card.isFileOpen() && IS_SD_PRINTING) {
      SERIAL_EM("Close file and save restart.gcode");
      card.stopSDPrint(true);
      clear_command_queue();
      quickstop_stepper();
      print_job_counter.stop();
      #if ENABLED(AUTOTEMP)
        thermalManager.autotempShutdown();
      #endif
      wait_for_heatup = false;
     // lcd_setstatus(MSG_PRINT_ABORTED, true); /////////////////////
	  enqueue_and_echo_commands_P(PSTR("M81")); //<<<< switch off after Stop e save. to save the 
                                                                                         //                                                       ''Battery''
    }
  }
///////////////////////////////////////////////////////////////////////////////

smagtech, 2017-04-22:
If we do only ( current_position[Z_AXIS] += 5.0; ) nozzle will drip some material on piece.