MKFirmware/MK4duo #108: Error Compile

pirat777knj, 2017-04-23:
Hi.
I can not compile the firmware.
Some mysterious mistakes that can not be won.
What is it and how to cure it?
Such mistakes probably about a hundred.
Arduino 1.6.9, DUE+RADDS

[image]

MagoKimbra, 2017-04-23:
Select nobody in warning. File - preferences

pirat777knj, 2017-04-24:
Thank you. Errors disappeared.
Another question:
I connect the bluetooth HC-05 with the following parameters:

> #define BLUETOOTH
> #define BLUETOOTH_PORT 1
> #define BLUETOOTH_BAUD 115200

For some reason, there is no connection to the printer. With the module it connects, but there is no response from the printer (
Plus, as long as it is enabled in the settings of the Bluetooth, the connection does not work through usb.
How can this be caused?

MagoKimbra, 2017-04-24:
Sorry, but I have to fix the bluetooth part. However if you use bluetooth as a connection port it is normal that it does not connect with the usb, either one or the other.
For now set the base connection on serial 1 and the baudrate.

pirat777knj, 2017-04-24:
We are looking forward to the appearance of the bluetooth)

closed 2017-05-26 by MagoKimbra