MKFirmware/MK4duo #109: Error compile for board Printrboard

Depoint, 2017-04-28:
I try to compile FW for Printrboard and get following error:

`Arduino: 1.8.1 (Windows 10), Board: "Printrboard"

WARNING: Category '' in library Wire is not valid. Setting to 'Uncategorized'
In file included from sketch\src/HAL/HAL.h:41:0,

                 from sketch\base.h:86,

                 from C:\Users\work\Documents\FW_all\MK4duo\MK4duo.ino:251:

sketch\src/HAL/HAL_AVR/HAL_AVR.h: In static member function 'static void HAL::serialSetBaudrate(long int)':

sketch\src/HAL/HAL_AVR/HAL_AVR.h:79:20: error: 'MKSerial' was not declared in this scope

   #define MKSERIAL MKSerial

                    ^

sketch\src/HAL/HAL_AVR/HAL_AVR.h:464:7: note: in expansion of macro 'MKSERIAL'

       MKSERIAL.begin(baud);

       ^

sketch\src/HAL/HAL_AVR/HAL_AVR.h: In static member function 'static bool HAL::serialByteAvailable()':

sketch\src/HAL/HAL_AVR/HAL_AVR.h:79:20: error: 'MKSerial' was not declared in this scope

   #define MKSERIAL MKSerial

                    ^

sketch\src/HAL/HAL_AVR/HAL_AVR.h:467:14: note: in expansion of macro 'MKSERIAL'

       return MKSERIAL.available() > 0;`

and so on.

Please tell me how can I fix it? Thank you!

[image]

MagoKimbra, 2017-04-28:
Sorry but which card do you put?

Depoint, 2017-04-28:
I make this file from 8.h
I have this board and it working fine in Marlin. Do you think it can work with you FW?

MagoKimbra, 2017-04-28:
8 is BOARD_TEENSYLU..
Where did you get the files for TEENSYLU?
I think they are old and not suitable for the new Arduino IDE.

Depoint, 2017-04-29:
Why do you think Printrboad is old? It works fine and compiles with the new Arduino IDE well.
I attached files and a screenshot.

https://yadi.sk/d/DXFJe58G3HUHqc

[image]

closed 2017-05-26 by MagoKimbra