MKFirmware/MK4duo #110: Error min_temp_error THERMISTOR 66

fabriciomoeller, 2017-05-01:
#define TEMP_SENSOR_0 66
#define TEMP_SENSOR_1 66

Cause error of min_temp_error

Configure the following variables as recommended:

// The number of consecutive low temperature errors that can occur
// before a min_temp_error is triggered. (Should not be more than 10.)
#define MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED 10

// The number of milliseconds to hotend will preheat before starting to check
// the temperature. This value should NOT be set to the time it takes the
// hot end to reach the target temperature, but the time it takes to reach
// the minimum temperature your thermistor can read. The lower the better / safer.
// This should not take more than 30 seconds (30000)
#define MILLISECONDS_PREHEAT_TIME 30000

Done the same build in Marlin and it works.

MagoKimbra, 2017-05-01:
Do you can try with one hotend please?

fabriciomoeller, 2017-05-01:
My previous setup was like this:

#define TEMP_SENSOR_0 1 // thermistor 3950
#define TEMP_SENSOR_1 66 // thermistor Dyze

In this configuration I worked.

When I switched to two, Dyze stopped working.

closed 2017-05-26 by MagoKimbra