MKFirmware/MK4duo #111: BLTOUCH throws error on homing. Reset required

apphead, 2017-05-02:
BLTOUCH stopped working
[Configuration_Overall.h.txt](https://github.com/MagoKimbra/MK4duo/files/971967/Configuration_Overall.h.txt)

 on latest 4.3.19_dev. Reverted back to V4.3.1 stable and things work fine.

Here's what I do / happens:
G28 -> x and y home correctly, then head moves to Z_SAFE_HOMING_ coordinates, the BLTOUCH pin goes down and up twice (without hitting the bed) and that is it. No movement on Z axis at all. 
This is the error in console: _STOP called because of BLTouch error - restart with M999_
After M999 the printer is able to move all axis correctly

I noticed the additional setting BLTOUCH_DELAY. What is a proper use case for it?

Find my settings for 4.3.19_dev attached.

MagoKimbra, 2017-05-02:
Most likely the problem is that you do not have an original bltouch.
Test with M119 and stow probe must not triggered. Then you do the test with the deploy probe you must always not triggered.
M280 P0 S10
M280 P0 S90

apphead, 2017-05-02:
I am pretty sure I have the original from Antslab (not the Geetech one) as I ordered it from the source antslab promoted on their kickstarter project. M119 does not work with my probe - it always says triggered.
Have you tried my Configuration_Overall.h?

MagoKimbra, 2017-05-02:
Ok reverse the logical z probe!

apphead, 2017-05-02:
Did that already. This is how I tested my bltouch in the past (I send that to antslabs as their documentation lacked this bit back then....)
Get Endstop Status(M119) Marlin
1. M280 P0 S160 
2. M280 P0 S60
3. push-pin up or down by hand
4. M119
Everything is fine, except:
21:43:47.192: x_min: NOT TRIGGERED
21:43:47.192: y_min: NOT TRIGGERED
21:43:47.193: z_min: NOT TRIGGERED
21:43:47.193: z_probe: TRIGGERED

z_min and z_probe should be the same! Where can I define the z_probe pin?

closed 2017-05-02 by apphead