MKFirmware/MK4duo #118: issue with turnkey raster

TimNe0, 2017-05-09:
I have attached g-code:
[hackspace logo.zip](https://github.com/MagoKimbra/MK4duo/files/987897/hackspace.logo.zip)

this is the output.  The raster text is unreadable, Laser does not turn off between raster finish and cut begin.  Laser then seemingly randomly turned the laser on/off during rectangular cut.

[image]

Is it the turnkey g-code or the firmware?

TimNe0, 2017-05-09:
It repeats on a box regardless of what I do

[gcode.zip](https://github.com/MagoKimbra/MK4duo/files/987972/gcode.zip)

Picture of repeated errors - changed text to a vector font instead of rastering.

[image]

TimNe0, 2017-05-09:
I have a video of the amp meter turning off during the cut...

TimNe0, 2017-05-26:
Hi, no response on this but it's still affecting everything we do?

MagoKimbra, 2017-05-27:
Sorry i closed for error...
I checking this with my friend with laser...

mayhem2408, 2017-05-27:
@TimNe0 Unfortunately my firmware has been modified and works slightly differently. I will recompile so it works with your turnkey version and see what happens. I assume you are using the original turnkey inkspace plugin to generate your gcode?

TimNe0, 2017-05-28:
Yes, didn't know there was anything else to use!

mayhem2408, 2017-05-29:
So far I have not managed to get MK4Duo working properly on my setup. Step direction only working one way, Fire pin not working. But I'll get it sorted out. I still have my original Turnkey Marlin firmware and ran your file through it. It looks like the gcode itself is OK. The outer box didn't burn all 4 sides, but I think that's my firmware. Hopefully i'll have MK4Duo working very soon and I'll test it again.

[image]

TimNe0, 2017-06-12:
Thanks very much!

TimNe0, 2017-06-14:
Any idea when that'd be sorted? I'm leaving the hackspace end of July and I've been working on this issue since November (thought it was hardware, replaced all the things) would be nice to leave it in a working state...

mayhem2408, 2017-06-15:
@TimNe0 My tests show that is not problem with the turnkey plugin itself. I am still waiting for my new Arduino Duo board so I can test MK4Duo. But it works fine with Marllin for Arduino Mego2560.

TimNe0, 2017-06-15:
Hi John,

I'm using mega2560 - have you applied a fix then? I don't have a ardunio duo

mayhem2408, 2017-06-15:
@TimNe0 I can't get MK4Duo to work at all on my Mega2560. Nothing to do with rastering. I'm having problems with the steppers not moving as expected. I am still using the original Marlin https://github.com/timschmidt/buildlog-lasercutter-marlin for this test.  I have a Duo coming which will hopefully work with MK4Duo

MagoKimbra, 2017-06-15:
I'm sorry you're having problems with raster laser, unfortunately I'm still waiting for info from who can help with arduino 8 bit, I just have arduino cards two and I have no problems.
What are the problems with the steppers?

mayhem2408, 2017-06-15:
@MagoKimbra The First problem is that the steppers motors only move in one direction and not the other. Example. The X axis will move Right when I increase the X, but does not move at all when I decrease the X. The other problem is the laser turns on at power up and stays on. I'll load up the latest firmware shortly and give some more detailed descriptions.

mayhem2408, 2017-06-16:
@MagoKimbra I'll open up another issue as the problems I am having are not related to rastering.

mayhem2408, 2017-06-16:
OK. I've managed to get MK4Duo configure enough to do some test. There is definatly a problem in MK4Duo's raster code. The exact same gcode script runs fine on the old turnkey firmware, but the results are wrong in MK4Duo.

[image]
Using MK4Duo firmware on a Mega2560+RAMP1.4

[image]
Using Turnkey Firmware on the same Mega2560 + RAMP1.4

MagoKimbra, 2017-06-16:
Which pin do you used for laser on ramps??

TimNe0, 2017-06-16:
#define LASER_PWR_PIN                   5
 #define LASER_TTL_PIN                   6

MagoKimbra, 2017-06-16:
Thank's

mayhem2408, 2017-06-16:
@MagoKimbra I'm running a single pin laser setup.
#define LASER_PWR_PIN       5
#define LASER_TTL_PIN      -1

MagoKimbra, 2017-06-16:
Ok can you try last version please?

TimNe0, 2017-06-16:
Me or John? I'll be back with the laser on Monday (it's not at home, it's
at the hackspace 150m away!)

MagoKimbra, 2017-06-16:
You and @mayhem2408 when you can.

mayhem2408, 2017-06-17:
@MagoKimbra  I've uploaded the new firmware into my mega2560. I'll fire it up in a few hours and post the results. I noticed you have removed the HIGH_TO_FIRE option. I have setups that used both. My K40 Laser pulls low to fire, and my diode laser pulls high to fire. I'll be trying it on my diode laser first.

MagoKimbra, 2017-06-17:
Ops, sorry, i copied the buildlog lasercutter version that i did not have, but if so it works, i put it back immediately ...

mayhem2408, 2017-06-17:
@MagoKimbra Pixel placement looks good now, but it is inverted, I assume because it does not have high to fire anymore. 

[image]

MagoKimbra, 2017-06-17:
Ok thank's. I look it's..

TimNe0, 2017-06-19:
Have loaded it in.  Laser isn't firing at all at the moment... but it was doing this with a replacement firmware I installed last week when I tried the turnkey firmware.

TimNe0, 2017-06-19:
have loaded old 4.3.1 software and laser is firing, so not hardware, looks like coding issue.

mayhem2408, 2017-06-20:
@MagoKimbra I have made some changes to the code on my fork. I've added an option in the configure_laser file called #define LASER_PWM_INVERT and made changes to the AVR and SAM laser.cpp files. Initial tests seems to have the laser working correctly now. I have to invert the PWM signal to make it work with my laser. I'll do a propper burn test in a few hours and see how it looks. Sorry my fork is out of sync with your code and I can't sent a pull request.

mayhem2408, 2017-06-20:
@MagoKimbra The changes I have made worked for the most part. Still have a problem with the laser turning on at bootup and staying on until signaled to do otherwise. I'll keep looking into that one.

[image]

mayhem2408, 2017-06-20:
@MagoKimbra Made a small change to the laser_init to that the laser it turned off during init sequence. Needs more testing, but is working fine on my setup.

mayhem2408, 2017-06-20:
@MagoKimbra For some unknown reason, the laser pwm appears to start intermittently pulsing causing the laser to flicker after about 30 seconds at idle. Probably some timeout somewhere interfering with the timer.

TimNe0, 2017-06-20:
Will be interested to see it working John! It's been a long 18m journey
trying to get this to work!

mayhem2408, 2017-06-20:
@MagoKimbra Still trying to find this flickering issue that occurs after the SD card print job is finished. I have timed it at it is exactly 120 seconds. Can you think what might be happening 2 minutes after the print job completes? But everything else is working at the moment.

[image]

mayhem2408, 2017-06-20:
@MagoKimbra My steppers are set to disable after 120 seconds. Will see if I can find a link there?

mayhem2408, 2017-06-20:
@MagoKimbra I have set DEFAULT_STEPPER_DEACTIVE_TIME to 0 and I am now at 15 minutes and the laser is still off. So the auto disable stepper is causing the laser flickering. If I manually disable the steppers via LCD, the problem does not occur.

MagoKimbra, 2017-06-21:
Ok !!! Good job. Thank's for feedback!

MagoKimbra, 2017-06-21:
In manage_inactivity If idle time is greater than DEFAULT_STEPPER_DEACTIVE_TIME, turn off the engines and call the laser_init routine again.

mayhem2408, 2017-06-22:
@MagoKimbra I changed laser_init to laser_extinguish in that routine and it is working OK now. After the 120 second timeout, the steppers disable and the laser turns off.

TimNe0, 2017-06-22:
mayhem2408 - can see your changes regarding the laser_extinguish in the latest from magokimbra - have you shared the other changes that got the rastering working and the odd issue with the on/off on straight lines?

mayhem2408, 2017-06-22:
@TimNe0 I have all the changed on my folk. Will hopefully get this into MK4Duo very soon. I have having some problems with git at the moment and trying to resolve them.

mayhem2408, 2017-06-22:
@MagoKimbra All the code changes look good and is working OK now.
@TimNe0 All the changes have been implemented. Let me know how you go. I try to help out where possible.

mayhem2408, 2017-06-22:
Perfect
[image]

MagoKimbra, 2017-06-23:
Ok thank's for help me...

mayhem2408, 2017-06-26:
@MagoKimbra I didn't calibrate by power levels correctly for the lighter gray areas, but pixel placement looks spot on.
[image]

MagoKimbra, 2017-06-26:
Good!!! Thank's!!!

TimNe0, 2017-06-27:
Have I missed anything obvious?

C:\Users\admin\AppData\Local\Temp\ccTJTSlW.ltrans5.ltrans.o: In function `__vector_17':

ccTJTSlW.ltrans5.o:(.text+0x616): undefined reference to `laser_fire(unsigned char)'

ccTJTSlW.ltrans5.o:(.text+0x9e4): undefined reference to `laser_fire(unsigned char)'

ccTJTSlW.ltrans5.o:(.text+0xaba): undefined reference to `laser_fire(unsigned char)'

C:\Users\admin\AppData\Local\Temp\ccTJTSlW.ltrans6.ltrans.o: In function `lcd_laser_test_fire_menu()':

ccTJTSlW.ltrans6.o:(.text+0x514): undefined reference to `laser_fire(unsigned char)'

C:\Users\admin\AppData\Local\Temp\ccTJTSlW.ltrans6.ltrans.o: In function `laser_test_fire(unsigned char, int)':

ccTJTSlW.ltrans6.o:(.text.unlikely+0x1c): undefined reference to `laser_fire(unsigned char)'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

TimNe0, 2017-06-27:
@mayhem2408 any ideas?

mayhem2408, 2017-06-28:
@TimNe0 What configuration file changes have you made. I'll try to reproduce it.

TimNe0, 2017-06-28:
See link!

https://www.cheltenhamhackspace.org/wiki/K40_Laser_Cutter#Firmware_-_MK4Duo

mayhem2408, 2017-06-28:
@TimNe0 It compiled without error for me. What version of the Arduino IDE are you running?

TimNe0, 2017-06-28:
1.8.3!

hmm, maybe I've made a mistake! I did it twice though... just noticed
@MagoKimbra was changing files last night after I posted my problem...
maybe he fixed it?  I'll try again tonight.

TimNe0, 2017-06-28:
Different computer, 1.8.3 software, fresh download from main page (assume I don't need a specific fork etc and should be using the main build) still the same error!

Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\Tim\AppData\Local\Temp\arduino_cache_429313\core\core_arduino_avr_mega_cpu_atmega2560_0c812875ac70eb4a9b385d8fb077f54c.a
C:\Users\Tim\AppData\Local\Temp\ccbI82HS.ltrans5.ltrans.o: In function `__vector_17':

ccbI82HS.ltrans5.o:(.text+0x616): undefined reference to `laser_fire(unsigned char)'

ccbI82HS.ltrans5.o:(.text+0x9e4): undefined reference to `laser_fire(unsigned char)'

ccbI82HS.ltrans5.o:(.text+0xaba): undefined reference to `laser_fire(unsigned char)'

C:\Users\Tim\AppData\Local\Temp\ccbI82HS.ltrans6.ltrans.o: In function `lcd_laser_test_fire_menu()':

ccbI82HS.ltrans6.o:(.text+0x514): undefined reference to `laser_fire(unsigned char)'

C:\Users\Tim\AppData\Local\Temp\ccbI82HS.ltrans6.ltrans.o: In function `laser_test_fire(unsigned char, int)':

ccbI82HS.ltrans6.o:(.text.unlikely+0x1c): undefined reference to `laser_fire(unsigned char)'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

TimNe0, 2017-06-28:
so the issue seems to be laser.cpp on line 180 seems to be the issue - we changed it from float to unsigned char and now it compiles, however laser intensity is now at 100% no matter what we do!  At least the laser is actually firing!

TimNe0, 2017-06-28:
[image]

Realised we needed to uncomment the invert line

Still same rastering issue I had before!

mayhem2408, 2017-06-28:
@TimNe0 I've had a look at the master branch and there is a bug.
Can you change the laser.cpp line 180 back to float,
and change laser.h line 80 from uint8_t to float.

See if that compiles. 

What image are you trying to engrave? can you share the gcode?

TimNe0, 2017-06-28:
OK have changed to float & it compiles fine.
[w000.zip](https://github.com/MagoKimbra/MK4duo/files/1110126/w000.zip)
I've created a file which literally is just the word "w000"
Still getting the same weird artefacts.

TimNe0, 2017-06-28:
[image]
Vector to the side is fine. Raster next to it. Not so much... Same speed and settings.

TimNe0, 2017-06-28:
Boxes are cutting properly again though, so that's a relief - it can be used to cut which is fine, just doesn't raster.

[38 more replies]

closed 2018-08-04 by iosonopersia