MKFirmware/MK4duo #119: Extruder/Laser/CNC selection

chromvis, 2017-05-11:
I may be doing something wrong, but I could not get laser to work with MK4duo, it perfectly working with buildlog-lasercutter-marlin though.
If I set "#define EXTRUDERS 0" in cinfig.basic.h - I am getting compilation error,
But, If it is not zero the system could not go to Laser (or CNC) mode because of the following code in MK_main.cpp 
[image]

(lines 75 - 83 in MK_main.cpp)
How anyone overcome this?

MagoKimbra, 2017-05-11:
Why? If you enable LASERBEAM printer_mode = PRINTER_MODE_LASER.
And with M450-M451-M452 you must change mode...
The extruder must set min 1 because the base it is a printer 3d...

chromvis, 2017-05-12:
Again, I may be wrong, or do not understand correctly how "if ... elif.." works, but if the first condition is TRUE (#if EXTRUDERS > 0) than it will never check next #elif statement. The first TRUE statement send program pointer to #endif. 
I.e. as far as EXTERUDERS > 0 system will never check if LASERBEAM is enabled or not.
#if ... #elif ... #elif ... construction return only one, the very first TRUE statement
Sorry for my misunderstanding, but could you please explain where I am wrong?

chromvis, 2017-05-13:
This configuration in MK_Main.cpp should work. I will check today.
[image]

closed 2017-05-26 by MagoKimbra