MKFirmware/MK4duo #125: Option to invert Z2, Z3, Z4 axis

apphead, 2017-05-19:
Could you please add
INVERT_Z2_VS_Z_DIR
INVERT_Z3_VS_Z_DIR

I tried to invert dir on the Z2 stepper (by setting INVERT_E2_DIR in my case), but it had zero effect. Which is strange anyway....

iosonopersia, 2017-08-23:
Hi, I was trying to help another user for another issue. While doing that I came across this code and I've implemented INVERT_Z2_VS_Z_DIR, INVERT_Z3_VS_Z_DIR and INVERT_Z4_VS_Z_PIN in (you can set them in Configuration_Feature.h but still not with the online configurator). Now we just need to wait that @MagoKimbra accepts my pull request.

apphead, 2017-08-23:
Hi @iosonopersia thanks for you support. I switched back to repetier but have my issues with it too, so I am tempted to use MK4Duo again - if only the current dev version would compile without errors. Will check and report back if I make the move.

iosonopersia, 2017-08-24:
Update: changes have been merged, INVERT_Z?_VS_Z_DIR is now a MK4duo feature!

iosonopersia, 2017-09-03:
@apphead this issue has to be closed, since it was fixed. In case you choose to return back to MK4duo and you have problems with the firmware, just open a new issue. Bye! @MagoKimbra

apphead, 2017-09-04:
I prefer Marlin over Repetier, but I've had trouble compiling latest dev for a few weeks now. I read that the old configuration files won't work with 4.3.27, so I tried the configurator, but firmware fails to download (syntax error). Will give MK4duo another shot once that is fixed! Thx anyways @MagoKimbra

closed 2017-09-04 by apphead

iosonopersia, 2017-09-04:
MagoKimbra has already fixed that, when you want you can give MK4duo a try. In case of problems, just open a new issue. Good luck, bye!