MKFirmware/MK4duo #126: Mesh bed leveling LCD 20x4lines do not show info

carlonb, 2017-05-19:
Ciao Mago,
Very often when I'm leveling the bed with Mesh option the display show nothing, if I touch a little the encoder knob the display will show the next info to proceed.
Anyway the mesh bed leveling is working very well.
Ciao
Carlo

iosonopersia, 2017-07-05:
Are you still experiencing this issue with MK4duo 4.3.24 ? However it's seems like a problem of the display (??).

carlonb, 2017-07-05:
If I remember well, this issue was solved with vers. 4.3.20dev. Now I'm using the 128x64 graphic LCD with 4.3.24dev so no problem, I do not know if the issue still with 20x4 LCD.

iosonopersia, 2017-07-05:
So this issue can be closed @MagoKimbra

iosonopersia, 2018-07-31:
You can safely close this very old issue, @MagoKimbra

closed 2018-07-31 by MagoKimbra