MKFirmware/MK4duo #127: RAMPS4DUE & #define ULTRA_LCD

Anry777, 2017-05-22:
Hi
I tried to configure MK4DUO for custom printer with hacked RAMPS & Due & RepRap Discount Controller
When i uncomment #define ULTRA_LCD - sketch cant compille with fpllow errors:

Arduino: 1.8.2 (Windows 10), Плата:"Arduino Due (Programming Port)"

In file included from sketch\src\lcd\ultralcd.cpp:48:0:

sketch\src\lcd\ultralcd_impl_HD44780.h:195:17: error: 'LCD_PINS_RS' was not declared in this scope

   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7

                 ^

sketch\src\lcd\ultralcd_impl_HD44780.h:195:30: error: 'LCD_PINS_ENABLE' was not declared in this scope

   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7

                              ^

sketch\src\lcd\ultralcd_impl_HD44780.h:195:47: error: 'LCD_PINS_D4' was not declared in this scope

   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7

                                               ^

sketch\src\lcd\ultralcd_impl_HD44780.h:195:60: error: 'LCD_PINS_D5' was not declared in this scope

   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7

                                                            ^

sketch\src\lcd\ultralcd_impl_HD44780.h:195:73: error: 'LCD_PINS_D6' was not declared in this scope

   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7

                                                                         ^

sketch\src\lcd\ultralcd_impl_HD44780.h:195:86: error: 'LCD_PINS_D7' was not declared in this scope

   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7

                                                                                      ^

In file included from sketch\src\lcd\ultralcd.cpp:48:0:

sketch\src\lcd\ultralcd_impl_HD44780.h: In function 'void lcd_implementation_init()':

sketch\src\lcd\ultralcd_impl_HD44780.h:387:18: error: 'LCD_PINS_RS' was not declared in this scope

       SET_OUTPUT(LCD_PINS_RS);

                  ^

sketch\src\lcd\ultralcd_impl_HD44780.h:388:18: error: 'LCD_PINS_ENABLE' was not declared in this scope

       SET_OUTPUT(LCD_PINS_ENABLE);

                  ^

exit status 1

Can anyone help me? 
Tnx

sh4jonny, 2017-07-04:
you need to add LCD pins define to the board header file. try to add
#define LCD_PINS_RS				16
#define LCD_PINS_ENABLE			17
#define LCD_PINS_D4				23
#define LCD_PINS_D5				25
#define LCD_PINS_D6				27
#define LCD_PINS_D7				29

to src/boards/1433.h

iosonopersia, 2017-07-05:
To edit the board files is now suggested to used the MK4duoBoards tool: https://github.com/iosonopersia/MK4DuoBoards . You can find every info you need in the README and wiki (Java 8 is required).

As far as this case is concerned, you need to import the 1433.h file and add in the "If blocks" section something like this (if it does work, we are going to modify it also in MK4duo):

#if ENABLED(ULTRA_LCD)
  #if ENABLED(NEWPANEL)
    // ramps-fd lcd adaptor
    #define LCD_PINS_RS              16
    #define LCD_PINS_ENABLE     17
    #define LCD_PINS_D4              23
    #define LCD_PINS_D5              25
    #define LCD_PINS_D6              27
    #define LCD_PINS_D7              29

    #if ENABLED(REPRAP_DISCOUNT_SMART_CONTROLLER)
      #define ORIG_BEEPER_PIN   37

      #define BTN_EN1           33
      #define BTN_EN2           31
      #define BTN_ENC          35

      #define SD_DETECT_PIN     49
    #endif
  #endif
#endif //ULTRA_LCD

iosonopersia, 2017-08-26:
This can be closed, issue was fixed

PR #203 (2017-08-26): Displays for Ramps4Due
Ramps4Due it's a modified Ramps board. In Marlin, display pins are the same used for the Ramps.
Added display pins to 1433.h and little fix (BEEPER_PIN -> ORIG_BEEPER_PIN) in 33.h to fix #127 and #199 still open issues.

Updated contribution.md and issue_template.md to prevent sanitycheck.h errors to be reported in the issues...

Little fixes in Radds board (1401.h)

Updated M406 code according to the latest change in Marlin bugfix-1.1.x
closed 2017-08-26 by MagoKimbra