MKFirmware/MK4duo #128: Implementation RADDS2LCD

3DSTB, 2017-05-23:
Is it possible to implement the display adapter RADDS2LCD?
Link: [https://www.thingiverse.com/thing:1740725](url)

There are already instructions for the repetier firmware in the files section:
`RADDS2LCD in Repetier-Firmware for DEV-version (1.x.x)
======================================================

Hardware
--------

Box Header: pin 1 = red conductor from ribbon cable -> side is marked with "rt" on RADDS2LCD.u / RADDS2LCD.o
on RADDS: red conductor towards MOS-FETs and screw terminals

Repetier.h
----------

at end of block beginning with:
##################################
#define NO_CONTROLLER 0

add these lines:
#########
#define CONTROLLER_RADDS_LCD_16X2 997
#define CONTROLLER_RADDS_LCD_20X4 998
#define CONTROLLER_RADDS_FGSC 999

Configuration.h
--------------

replace this:
#####################
#define FEATURE_CONTROLLER ...

by this:
######
/*
		which display ???
		- full graphics smart controller is tested!
		- row / column display is NOT TESTED - use with own testing ;)
  */
//#define FEATURE_CONTROLLER CONTROLLER_RADDS_LCD_16X2	// RADDS2LCD -> LCD with 16 columns and 2 rows
//#define FEATURE_CONTROLLER CONTROLLER_RADDS_LCD_20X4	// RADDS2LCD -> LCD with 20 columns and 4 rows
#define FEATURE_CONTROLLER CONTROLLER_RADDS_FGSC	  	// RADDS2LCD -> Full Graphics Smart Controller

ui.h
----

after these lines
#######

#if  FEATURE_CONTROLLER == CONTROLLER_RAMBO
...
#endif

add these lines
###############

// RADDS + RADDS2LCD + Full Graphics Smart Controller / other 4-bit LCD
#if (FEATURE_CONTROLLER == CONTROLLER_RADDS_FGSC || FEATURE_CONTROLLER == CONTROLLER_RADDS_LCD_16X2 || FEATURE_CONTROLLER == CONTROLLER_RADDS_LCD_20X4)
  #undef SDCARDDETECT
  #define SDCARDDETECT -1
  #undef SDSUPPORT
  #define SDSUPPORT 1
#endif // FEATURE_CONTROLLER == CONTROLLER_RADDS_FGSC

DisplayList.h
------

before or after these lines:
###################

#if FEATURE_CONTROLLER == CONTROLLER_RADDS
#undef SDSS
...
void uiCheckSlowKeys(uint16_t &action) {}
#endif

add these lines
##############

#if (FEATURE_CONTROLLER == CONTROLLER_RADDS_FGSC || FEATURE_CONTROLLER == CONTROLLER_RADDS_LCD_16X2 || FEATURE_CONTROLLER == CONTROLLER_RADDS_LCD_20X4)
  #undef SDSS
  #define SDSS            10
  #undef SPI_PIN
  #define SPI_PIN         77
  #undef SPI_CHAN
  #define SPI_CHAN        0
  #define UI_HAS_KEYS     1

  #define UI_HAS_BACK_KEY 0
  
  // decide which lcd is used
  #if (FEATURE_CONTROLLER == CONTROLLER_RADDS_FGSC)
	#define UI_DISPLAY_TYPE DISPLAY_U8G
    #define U8GLIB_ST7920
    #define UI_LCD_WIDTH 128
    #define UI_LCD_HEIGHT 64
    
    //select font size
    #define UI_FONT_6X10 //default font
    #ifdef UI_FONT_6X10
      #define UI_FONT_WIDTH 6
      #define UI_FONT_HEIGHT 10
      #define UI_FONT_SMALL_HEIGHT 7
      #define UI_FONT_DEFAULT repetier_6x10
      #define UI_FONT_SMALL repetier_5x7
      #define UI_FONT_SMALL_WIDTH 5 //smaller font for status display
      #define UI_ANIMATION false  // Animations are too slow
    #endif
  
    //calculate rows and cols available with current font
    #define UI_COLS (UI_LCD_WIDTH/UI_FONT_WIDTH)
    #define UI_ROWS (UI_LCD_HEIGHT/UI_FONT_HEIGHT)
    #define UI_DISPLAY_CHARSET 3
	
  #elif (FEATURE_CONTROLLER == CONTROLLER_RADDS_LCD_16X2)
		#define UI_DISPLAY_TYPE DISPLAY_4BIT
		#define UI_COLS 16
		#define UI_ROWS 2	
		
  #else (FEATURE_CONTROLLER == CONTROLLER_RADDS_LCD_20X4)
	  #define UI_DISPLAY_TYPE DISPLAY_4BIT
		#define UI_COLS 20
		#define UI_ROWS 4	
  #endif
  
  #define UI_DISPLAY_CHARSET 3
  
  #define BEEPER_TYPE 1
  #undef BEEPER_PIN
  #define BEEPER_PIN             41
  #define UI_DISPLAY_RS_PIN      42
  #define UI_DISPLAY_RW_PIN      -1
  #define UI_DISPLAY_ENABLE_PIN  43
  #define UI_DISPLAY_D0_PIN      44
  #define UI_DISPLAY_D1_PIN      45
  #define UI_DISPLAY_D2_PIN      46
  #define UI_DISPLAY_D3_PIN      47
  #define UI_DISPLAY_D4_PIN      44
  #define UI_DISPLAY_D5_PIN      45
  #define UI_DISPLAY_D6_PIN      46
  #define UI_DISPLAY_D7_PIN      47

  // swap these two numbers to invert rotary encoder scroll direction
  #define UI_ENCODER_A           50
  #define UI_ENCODER_B           52

  #define UI_ENCODER_CLICK       48
  #define UI_RESET_PIN           -1
  #define UI_DELAYPERCHAR 50
  #define UI_INVERT_MENU_DIRECTION 0
  #define UI_BUTTON_BACK         71 // -1
  
  #ifdef UI_MAIN
    void uiInitKeys() {
      UI_KEYS_INIT_CLICKENCODER_LOW(UI_ENCODER_A, UI_ENCODER_B); // click encoder on pins 47 and 45. Phase is connected with gnd for signals.
      UI_KEYS_INIT_BUTTON_LOW(UI_ENCODER_CLICK); // push button, connects gnd to pin
      UI_KEYS_INIT_BUTTON_LOW(UI_BUTTON_BACK);
    }
    void uiCheckKeys(uint16_t &action) {
      UI_KEYS_CLICKENCODER_LOW(UI_ENCODER_A, UI_ENCODER_B); // click encoder on pins 47 and 45. Phase is connected with gnd for signals.
      UI_KEYS_BUTTON_LOW(UI_ENCODER_CLICK, UI_ACTION_OK); // push button, connects gnd to pin
      UI_KEYS_BUTTON_LOW(UI_BUTTON_BACK, UI_ACTION_BACK);
    }
    inline void uiCheckSlowEncoder() {}
    void uiCheckSlowKeys(uint16_t &action) {}
  #endif
#endif // controller RADDS2LCD
	
DisplayList.h - (optional)
--------------------------
if you want to swap rotary encoder scroll direction (count clock / clockwise rotation)
then swap 
##########################################
#if FEATURE_CONTROLLER == CONTROLLER_RADDS_FGSC
...
#define UI_ENCODER_A           52
#define UI_ENCODER_B           50
##########################################
to
##########################################
#if FEATURE_CONTROLLER == CONTROLLER_RADDS_FGSC
...
#define UI_ENCODER_A           50
#define UI_ENCODER_B           52
##########################################`

I'm interested in using the: [http://reprap.org/wiki/RepRapDiscount_Smart_Controller]url

3DSTB, 2017-05-23:
Here is some code from a 401.h which finally may work more ore less:

`/****************************************************************************************
* 401
*
* RADDS
****************************************************************************************/

#define KNOWN_BOARD
#define BOARD_NAME "RADDS"

#ifndef ARDUINO_ARCH_SAM
  #error Oops! Make sure you have 'Arduino Due' selected from the 'Tools -> Boards' menu.
#endif

#define RADDS

#define ORIG_X_STEP_PIN         24
#define ORIG_X_DIR_PIN          23
#define ORIG_X_ENABLE_PIN       26

#define ORIG_Y_STEP_PIN         17
#define ORIG_Y_DIR_PIN          16
#define ORIG_Y_ENABLE_PIN       22

#define ORIG_Z_STEP_PIN          2
#define ORIG_Z_DIR_PIN           3
#define ORIG_Z_ENABLE_PIN       15

// Note that on the Due pin A0 on the board is channel 2 on the ARM chip
#define ORIG_X_MIN_PIN          28
#define ORIG_X_MAX_PIN          34
#define ORIG_Y_MIN_PIN          30
#define ORIG_Y_MAX_PIN          36
#define ORIG_Z_MIN_PIN          32
#define ORIG_Z_MAX_PIN          38

#define ORIG_E0_STEP_PIN        61
#define ORIG_E0_DIR_PIN         60
#define ORIG_E0_ENABLE_PIN      62

#define ORIG_E1_STEP_PIN        64
#define ORIG_E1_DIR_PIN         63
#define ORIG_E1_ENABLE_PIN      65

#define ORIG_E2_STEP_PIN        51
#define ORIG_E2_DIR_PIN         53
#define ORIG_E2_ENABLE_PIN      49

#define ORIG_E3_STEP_PIN        35
#define ORIG_E3_DIR_PIN         33
#define ORIG_E3_ENABLE_PIN      37

#define ORIG_E4_STEP_PIN        29
#define ORIG_E4_DIR_PIN         27
#define ORIG_E4_ENABLE_PIN      31

#define SDPOWER                 -1
#define SDSS                     4
#define LED_PIN                 -1

#define ORIG_BEEPER_PIN         41

#define ORIG_FAN_PIN 	           9
#define ORIG_FAN2_PIN            8

#define ORIG_PS_ON_PIN          40

#define KILL_PIN                -1

#define ORIG_HEATER_BED_PIN      7  // BED
#define ORIG_HEATER_0_PIN       13
#define ORIG_HEATER_1_PIN       12
#define ORIG_HEATER_2_PIN       11

#define ORIG_TEMP_BED_PIN        4  // ANALOG NUMBERING #57
#define ORIG_TEMP_0_PIN          0  // ANALOG NUMBERING #54
#define ORIG_TEMP_1_PIN          1  // ANALOG NUMBERING #58
#define ORIG_TEMP_2_PIN          2  // ANALOG NUMBERING #55
#define ORIG_TEMP_3_PIN          3  // ANALOG NUMBERING #56

// #define THERMOCOUPLE_0_PIN  2   // Dua analog pin #59 = A5 -> AD 2

// http://doku.radds.org/wp-content/uploads/2015/02/RADDS_Wiring.png

#if NUM_SERVOS > 0
  #define SERVO0_PIN           5
  #if NUM_SERVOS > 1
    #define SERVO1_PIN         6
    #if NUM_SERVOS > 2
      #define SERVO2_PIN      39
      #if NUM_SERVOS > 3
        #define SERVO3_PIN    40
      #endif
    #endif
  #endif
#endif

#if ENABLED(ULTRA_LCD)
  // RADDS LCD panel
  #if ENABLED(RADDS_DISPLAY)
    #define LCD_PINS_RS 		42
    #define LCD_PINS_ENABLE 	43
    #define LCD_PINS_D4 		44
    #define LCD_PINS_D5 		45
    #define LCD_PINS_D6 		46
    #define LCD_PINS_D7 		47

    #define BEEPER          41

    #define BTN_EN1         50
    #define BTN_EN2         52
    #define BTN_ENC         48

    #define BTN_BACK        71

    #undef SDSS
    #define SDSS            10 // 4,10,52 if using HW SPI.
    #define SD_DETECT_PIN   14

  #elif ENABLED(REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)
    #define LCD_PINS_RS     46
    #define LCD_PINS_ENABLE 47
    #define LCD_PINS_D4     44

    #define ORIG_BEEPER_PIN 41

    #define BTN_EN1         50
    #define BTN_EN2         52
    #define BTN_ENC         48
	
  #elif ENABLED(REPRAP_DISCOUNT_SMART_CONTROLLER)
  
	#define LCD_PINS_RS 		42
    #define LCD_PINS_ENABLE 	43
  
	#define LCD_PINS_D0 		-1
    #define LCD_PINS_D1 		-1
    #define LCD_PINS_D2 		-1
    #define LCD_PINS_D3 		-1
	#define LCD_PINS_D4 		44
    #define LCD_PINS_D5 		45
    #define LCD_PINS_D6 		46
    #define LCD_PINS_D7 		47
	
	#define ORIG_BEEPER_PIN   -1
	#define BEEPER            -1
	
	#define BTN_EN1           50
    #define BTN_EN2           52
    #define BTN_ENC           48
	
	#undef SDSS
    #define SDSS            10 // 4,10,52 if using HW SPI.
	#define SD_DETECT_PIN   -1
	
  #elif ENABLED(SSD1306_OLED_I2C_CONTROLLER)
    #define BTN_EN1         50
    #define BTN_EN2         52
    #define BTN_ENC         48
    #define BEEPER          41
    #define LCD_SDSS        10
    #define SD_DETECT_PIN   14
    #define KILL_PIN        -1

  #elif ENABLED(SPARK_FULL_GRAPHICS)
    #define LCD_PINS_D4     29
    #define LCD_PINS_ENABLE 27
    #define LCD_PINS_RS     25

    #define BTN_EN1         35
    #define BTN_EN2         33
    #define BTN_ENC         37

    #define KILL_PIN        -1
    #undef BEEPER
    #define BEEPER          -1
	#endif // SPARK_FULL_GRAPHICS
#endif // ULTRA_LCD

`

3DSTB, 2017-05-23:
as i see is the 401.h now this file:
[https://github.com/MagoKimbra/MK4duo/blob/master/MK4duo/src/boards/1401.h](https://github.com/MagoKimbra/MK4duo/blob/master/MK4duo/src/boards/1401.h)

iosonopersia, 2017-07-05:
This issue should now be solved, as I can see from 1401.h (which seems to have support for the RADDS_LCD). Is it now working for you, @3DSTB, or is it still an issue ?

iosonopersia, 2017-08-25:
Ok, sorry @3DSTB , I didn't understand well the issue... Now I'm working on it. I'll keep you updated

3DSTB, 2017-09-04:
Hi iosonopersia, my controller is

http://www.gettingmyfeetwet.com/2014/07/how-to-fix-reprapdiscount-smart.html
or as code: REPRAP_DISCOUNT_SMART_CONTROLLER

iosonopersia, 2018-09-05:
https://github.com/MarlinFirmware/Marlin/pull/11731 In Marlin they fixed this same issue justby changing value to two pins. I think we're going to do the same.