MKFirmware/MK4duo #130: Error:SD read error

Tick007, 2017-05-27:
Almost every second time than I start printing job, shortly after start, printing stops and I receive in cycle "Error:SD read error" in pronterface log:
Error:SD read error
Error:SD read error
Error:SD read error
Error:SD read error
Error:SD read error
Error:SD read error
Error:SD read error
Error:SD read error
Error:SD read error
Error:SD read error
Error:SD read error
Error:SD read error
Error:SD read error
Error:SD read error
Error:SD read error
and so on....
SD card itself is ok - I can read and write files for it.
Any suggestions to solve it ?

MagoKimbra, 2017-05-27:
Info...
Board, type SD, on board or LCD.
Version firmware..
Thx

Tick007, 2017-05-27:
Sure, sorry, I should write it beforehand.
Arduino DUE + Ramps FD 1.0
Firmware 4.3.20_dev
sd adapter build in 12864 (REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)
SD card: 4GB HC class 6, noname, FAT32

Tick007, 2017-05-27:
More log:
SD card ok
Echo:enqueueing "M23 c_verbinder.gcode"
Echo:enqueueing "M24"
File opened: c_verbinder.gcode Size: 12017017
File selected
Error:SD read error
[ERROR] Error:SD read error

Error:SD read error
[ERROR] Error:SD read error

...............................................

Error:SD read error
[ERROR] Error:SD read error

Error:SD read erт
start
Info:PowerUp
Echo:MK4duo_4.3.20_dev
Echo:Last Updated: May 27 2017 21:20:18 | Author: (none, default config)

Tick007, 2017-05-27:
beetwin File selected and Error:SD read error printer prints a couple of layers

Tick007, 2017-05-27:
USB (native port) and COM (via uart ) werer connected

closed 2017-06-20 by Tick007