MKFirmware/MK4duo #133: Compile error with dual x-carriage

atzs, 2017-05-29:
Arduino: 1.8.2 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\base.h:114:0,

                 from C:\OneDrive\Project\MK4duo-4_3_2\MK4duo\MK4duo.ino:248:

sketch\src/parser/parser.h: In static member function 'static bool GCodeParser::seen(char)':

sketch\src/parser/parser.h:129:24: warning: invalid conversion from 'const char*' to 'char*' [-fpermissive]

       if (b) value_ptr = DECIMAL_SIGNED(p[1]) ? &p[1] : NULL;

                        ^

In file included from sketch\src\../base.h:114:0,

                 from sketch\src\MK_Main.cpp:31:

sketch\src\../src/parser/parser.h: In static member function 'static bool GCodeParser::seen(char)':

sketch\src\../src/parser/parser.h:129:24: warning: invalid conversion from 'const char*' to 'char*' [-fpermissive]

       if (b) value_ptr = DECIMAL_SIGNED(p[1]) ? &p[1] : NULL;

                        ^

sketch\src\MK_Main.cpp: In function 'void update_software_endstops(AxisEnum)':

sketch\src\MK_Main.cpp:1208:24: error: 'home_offset' was not declared in this scope

     const float offs = home_offset[axis] + position_shift[axis];

                        ^

sketch\src\MK_Main.cpp:1208:44: error: 'position_shift' was not declared in this scope

     const float offs = home_offset[axis] + position_shift[axis];

                                            ^

sketch\src\MK_Main.cpp:1210:5: error: 'workspace_offset' was not declared in this scope

     workspace_offset[axis] = offs;

     ^

sketch\src\MK_Main.cpp: In function 'bool prepare_move_to_destination_dualx()':

sketch\src\MK_Main.cpp:13387:11: error: 'active_extruder_parked' was not declared in this scope

           active_extruder_parked = false;

           ^

In file included from sketch\src\../base.h:119:0,

                 from sketch\src\MK_Main.cpp:31:

sketch\src\MK_Main.cpp:13402:34: error: 'inactive_extruder_x_pos' was not declared in this scope

               LOGICAL_X_POSITION(inactive_extruder_x_pos),

                                  ^

sketch\src\../src/MK_Main.h:153:38: note: in definition of macro 'LOGICAL_X_POSITION'

   #define LOGICAL_X_POSITION(POS)   (POS)

                                      ^

sketch\src\MK_Main.cpp:13408:42: error: 'duplicate_extruder_x_offset' was not declared in this scope

               current_position[X_AXIS] + duplicate_extruder_x_offset,

                                          ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

jgauchia, 2017-06-05:
I've added this on MK_main.cpp
but i don't see the declaration :   duplicate_extruder_x_offset

#if ENABLED(DUAL_X_CARRIAGE)
  extern bool hotend_duplication_enabled;
  **extern bool active_extruder_parked;
  extern bool inactive_extruder_x_pos;**
#endif

jgauchia, 2017-06-05:
Declared duplicate_extruder_x_offset as INT , raise this error

sketch/src/kinematics/cartesian_kinematics.cpp: In member function 'void CartesianKinematics::homeaxis(AxisEnum)':
sketch/src/kinematics/cartesian_kinematics.cpp:231:54: error: 'x_home_dir' was not declared in this scope
         (axis == X_AXIS) ? x_home_dir(active_extruder) :
                                                      ^
sketch/src/kinematics/cartesian_kinematics.cpp: In member function 'void CartesianKinematics::Home(bool)':
sketch/src/kinematics/cartesian_kinematics.cpp:521:9: error: 'inactive_hotend_x_pos' was not declared in this scope
         inactive_hotend_x_pos = RAW_X_POSITION(current_position[X_AXIS]);
         ^
In file included from sketch/src/kinematics/../../base.h:41:0,
                 from sketch/src/kinematics/cartesian_kinematics.cpp:29:
sketch/src/kinematics/cartesian_kinematics.cpp:528:20: error: 'raised_parked_position' was not declared in this scope
         COPY_ARRAY(raised_parked_position, current_position);
                    ^
sketch/src/kinematics/../../src/macros.h:87:34: note: in definition of macro 'COPY_ARRAY'
 #define COPY_ARRAY(a,b)   memcpy(a, b, min(sizeof(a), sizeof(b)))
                                  ^
sketch/src/kinematics/cartesian_kinematics.cpp:529:9: error: 'delayed_move_time' was not declared in this scope
         delayed_move_time = 0;
         ^
sketch/src/kinematics/cartesian_kinematics.cpp:530:9: error: 'active_hotend_parked' was not declared in this scope
         active_hotend_parked = true;
         ^
sketch/src/kinematics/cartesian_kinematics.cpp: In member function 'void CartesianKinematics::set_axis_is_at_home(AxisEnum)':
sketch/src/kinematics/cartesian_kinematics.cpp:594:54: error: 'dual_x_carriage_mode' was not declared in this scope
       if (axis == X_AXIS && (active_extruder == 1 || dual_x_carriage_mode == DXC_DUPLICATION_MODE)) {
                                                      ^
sketch/src/kinematics/cartesian_kinematics.cpp:594:78: error: 'DXC_DUPLICATION_MODE' was not declared in this scope
       if (axis == X_AXIS && (active_extruder == 1 || dual_x_carriage_mode == DXC_DUPLICATION_MODE)) {
                                                                              ^
sketch/src/kinematics/cartesian_kinematics.cpp:595:62: error: 'x_home_pos' was not declared in this scope
         current_position[X_AXIS] = x_home_pos(active_extruder);
                                                              ^
exit status 1
Error compilación en tarjeta Arduino/Genuino Mega or Mega 2560.

jgauchia, 2017-06-05:
4.3.1 Compiles OK with DUAL_X_CARRIAGE

iosonopersia, 2017-07-05:
Are you still experiencing this issue with MK4duo 4.3.24 ? @jgauchia

jgauchia, 2017-07-15:
@iosonopersia Hi, I didn't try this version. Maybe this weekend.

jgauchia, 2017-07-15:
@iosonopersia With 4.3.24 , same Overall than actual in production raise this error:

sketch/src/endstop/endstops.cpp: In member function 'void Endstops::update_software_endstops(AxisEnum)':
sketch/src/endstop/endstops.cpp:323:34: error: 'class Cartesian_Mechanics' has no member named 'home_offset'
     const float offs = mechanics.home_offset[axis] + mechanics.position_shift[axis];
                                  ^
sketch/src/endstop/endstops.cpp:323:64: error: 'class Cartesian_Mechanics' has no member named 'position_shift'
     const float offs = mechanics.home_offset[axis] + mechanics.position_shift[axis];
                                                                ^
sketch/src/endstop/endstops.cpp:325:15: error: 'class Cartesian_Mechanics' has no member named 'workspace_offset'
     mechanics.workspace_offset[axis] = offs;
               ^
sketch/src/endstop/endstops.cpp:338:18: error: 'dual_x_carriage_mode' was not declared in this scope
         else if (dual_x_carriage_mode == DXC_DUPLICATION_MODE) {
                  ^
In file included from sketch/src/endstop/../../base.h:26:0,
                 from sketch/src/endstop/endstops.cpp:27:
sketch/src/endstop/endstops.cpp:342:87: error: 'duplicate_hotend_x_offset' was not declared in this scope
           soft_endstop_max[X_AXIS] = min(mechanics.base_max_pos[X_AXIS], dual_max_x - duplicate_hotend_x_offset) + offs;
                                                                                       ^
/home/jordi/arduino-1.6.13/hardware/arduino/avr/cores/arduino/Arduino.h:92:24: note: in definition of macro 'min'
 #define min(a,b) ((a)<(b)?(a):(b))
                        ^
exit status 1
Error compilación en tarjeta Arduino/Genuino Mega or Mega 2560.

iosonopersia, 2017-07-16:
Seems like you're using an old version of the Arduino IDE (maybe 1.6.13 ?). Please retry with the latest version (which should be 1.8.3)

iosonopersia, 2017-07-16:
OK, @jgauchia , I'm getting the same error with Arduino 1.8.3 . I'm trying to understand what's the problem

iosonopersia, 2017-07-16:
I've made a fix which should work well. We have to wait that @MagoKimbra checks my proposal. Meanwhile I suggest you to try to use always the latest version of Arduino IDE to compile MK4duo.

iosonopersia, 2017-07-16:
The fix was approved. You can now try downloading again the firmware. Your configuration_overall should work as expected. Please tell us if you succeed!
@jgauchia

jgauchia, 2017-07-16:
@iosonopersia ok, I'll try with the last Arduino IDE , otherwise , now doesn't move any axis

jgauchia, 2017-07-16:
Well, first test with 4.3.24_dev
Any movement in axis reset the printer
Set first hotend temperature ok.
Change extruder with T1 raise this

>>> T1
SENDING:T1
Echo:Home XYZ primer
No move on toolchange
Error:cold extrusion prevented
[ERROR] Error:cold extrusion prevented

Error:cold extrusion prevented
[ERROR] Error:cold extrusion prevented

jgauchia, 2017-07-16:
I think that issue is in Configuration_Pin and with my second X step driver

iosonopersia, 2017-07-18:
Does Marlin work with your setup? (I mean, does the dual_x_carriage feature work with Marlin?). Have you already tried that?

jgauchia, 2017-07-18:
Some time ago I've tested It, with some pins changed of the second stepper controller, but I'm using Nextion LCD and Marlin doesn't have support for it.
Mk4Duo 4.3.07 works perfect.

iosonopersia, 2017-07-18:
So we just need to understand what has changed after 4.3.07...

jgauchia, 2017-07-18:
I ser some changes in board declaration (board 36) un stepper definition CS and ENABLE pins are new in this version , another changes are in Configuration_pins and X2 stepper definition. Maybe problem was this new CS and ENABLE definitions?

iosonopersia, 2017-08-02:
Hi @jgauchia , a little change was made in the dual x carriage code today. Could you please try again with the latest version of the firmware?

jgauchia, 2017-08-18:
Hi @iosonopersia 
Sorry, I'm on hollydays , Now I'll test it! Thanks!

jgauchia, 2017-08-18:
HI @iosonopersia 
Compiles well, but when I power up the printer serial monitor dumps this:
start
Info:External Reset
Echo:MK4duo_4.3.26_dev
Echo:Last Updated: Aug 18 2017 17:45:01 | Author: (none, default config)
Echo:Compiled: Aug 18 2017
Echo:Free Memory: 8694 PlannerBufferBytes: 1216
okHardcoded SD Default Settings Loaded
Attempted to write invalid text to console, which could be due to an invalid baudrate
Config:  G21 ; Units in mm
Attempted to write invalid text to console, which could be due to an invalid baudrate
Config:  M92 T1 En7.442
Attempted to write invalid text to console, which could be due to an invalid baudrate
Config:  M20553900.000
Config:Maximum Acceleration (units/s2):
Config:  M201 X600 Y600 Z5
Config:  M201 T0 E3000
Config:  M201 T1 E3000
Attempted to write invalid text to console, which could be due to an invalid baudrate
Config:Advanced variables: S<min_feedrate> V<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> T* E<max_e_jerk>
Attempted to write invalid text to console, which could be due to an invalid baudrate
Config:Hotend offset (mm):
Config:  M218 H1 X0.000 Y0.000 Z0.000
Config:Auto Bed Leveling:
Config:  M320 S0
Attempted to write invalid text to console, which could be due to an invalid baudrate
Attempted to write invalid text to console, which could be due to an invalid baudrate
Attempted to write invalid text to console, which could be due to an invalid baudrate
Attempted to write invalid text to console, which could be due to an invalid baudrate
Attempted to write invalid text to console, which could be due to an invalid baudrate
8618 PlannerBufferBytes: 1216
okHardcoded SD Default Settings Loaded
Attempted to write invalid text to console, which could be due to an invalid baudrate
Config:  G21 ; Units in mm
Attempted to write invalid text to console, which could be due to an invalid baudrate
Attempted to write invalid text to console, which could be due to an invalid baudrate
Attempted to write invalid text to console, which could be due to an invalid baudrate
Config:Maximum Acceleration (units/s2):
Attempted to write invalid text to console, which could be due to an invalid baudrate
Config:  M201 X600 Y600 Z5
Config:  M201 T0 E3000
Config:  M201 T1 E3000
Attempted to write invalid text to console, which could be due to an invalid baudrate
Config:  M2000.000
Config:Advanced variables: S<min_feedrate> V<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> T* E<max_e_jerk>
Attempted to write invalid text to console, which could be due to an invalid baudrate
Attempted to write invalid text to console, which could be due to an invalid baudrate
Config:  M205 .000
Config:Hotend offset (mm):
Config:  M218 H1 X0.000 Y0.000 Z0.000
Config:Auto Bed Leveling:
Config:  M320 S0
Attempted to write invalid text to console, which could be due to an invalid baudrate
Attempted to write invalid text to console, which could be due to an invalid baudrate
Attempted to write invalid text to console, which could be due to an invalid baudrate
acceleration: 1216
okHardcoded SD Default Settings Loaded
Attempted to write invalid text to console, which could be due to an invalid baudrate
Config:  G21 ; Units in mm
Attempted to write invalid text to console, which could be due to an invalid baudrate
Attempted to write invalid text to console, which could be due to an invalid baudrate
Attempted to write invalid text to console, which could be due to an invalid baudrate

Repeatly

iosonopersia, 2017-08-18:
Ok, that's so strange, it's happening also to some other people... That's not a dual x carriage related problem, it's a more general one. We'll investigate

iosonopersia, 2017-08-18:
Are you setting the same baudrate both in the firmware and in the host?

jgauchia, 2017-08-18:
@iosonopersia yes,
The same baudrate, it's fixed in Configuration_Overall.h

I'll upload new nextion firmware ant try again

jgauchia, 2017-08-18:
@iosonopersia the same
I'm going back to 4.3.07

iosonopersia, 2017-08-18:
I'm so sorry for that... We will try to solve the problem

jgauchia, 2017-08-19:
@iosonopersia No problem
I'm very happy with this firmware!
I can wait ,
Thanks

iosonopersia, 2017-08-21:
Hi @jgauchia, sorry if I bother you again. It would be useful if you send here your Configuration_Overall.h (the one you used with 4.3.26). I'm trying to collect every possible info about the error. Thanks

jgauchia, 2017-08-22:
Hi @iosonopersia 
No problem, 
Now I'm at work, when I arrive home , I'll send you Configuration_Overall and Configuration_Pins, because I've modify It for the addicional stepper controler and Z_PROBE_PIN.

Thanks Again!

jgauchia, 2017-08-24:
here the configuration_overall & pins

[Configuration.zip](https://github.com/MKFirmware/MK4duo/files/1250283/Configuration.zip)

jgauchia, 2017-10-16:
Hi! any news about this issue?

iosonopersia, 2017-12-08:
Hi @jgauchia . Since August a lot has happened. We're now reached the `4.3.3 stable` milestone! We encourage you to try our latest version using [this configurator (link)](http://marlinkimbra.it/configurator/v4_3_3/firmware_configurator.php). We do really hope that it will work for you, since we put a lot of effort in it. Feel free to report every problem/bug you may find.

jgauchia, 2017-12-08:
Hi @iosonopersia .
Thank you!!!
I'll try now!

jgauchia, 2017-12-08:
Hi @iosonopersia , in the configurator link, my X2 stepper is in another pinout that defined in the list.
Maybe I think that I'll configure manually.

iosonopersia, 2017-12-08:
You can use the configurator and then edit the downloaded Configuration_Pins.h to fix this

jgauchia, 2017-12-08:
@iosonopersia yes, I'll try it , but downloads doesn't works

iosonopersia, 2017-12-08:
You need to enable popups in your browser's settings, otherwise download won't start

jgauchia, 2017-12-08:
mmm I'll try again, popus are enabled. OK, wait

jgauchia, 2017-12-08:
@iosonopersia return this in nav bar
"warning:%20Invalid%20argument%20supplied%20for%20foreach()%20in%20/customers/5/9/7/marlinkimbra.it/httpd.www/configurator/v4_3_3/config.php%20on%20line%201552Warning:%20Invalid%20argument%20supplied%20for%20foreach()%20in%20/customers/5/9/7/marlinkimbra.it/httpd.www/configurator/v4_3_3/config.php%20on%20line%201552../tmp/builder_1512756813-5a2ad64d03054/zip/MK4duo_08-12-2017_191333.zip"

Otherwise, Upload of my previous Configuration_overall doesn't read the actual configuration

iosonopersia, 2017-12-08:
Ok, you seem to be a very unlucky person xD. On my pc it works perfectly. Try with a manual configuration but be aware that something has changed (for example the temperature sensor settings). @MagoKimbra, the error is this: "Warning: Invalid argument supplied for foreach() in /customers/5/9/7/marlinkimbra.it/httpd.www/configurator/v4_3_3/config.php on line 1552"

jgauchia, 2017-12-08:
@iosonopersia JAJAJAAJ don't worry I'll made manual configuration...

iosonopersia, 2017-12-08:
@jgauchia I would like to understand what it's wrong with the configurator. How did you exactly get that error message? What configs did you modify (and how) before getting that error? Please give us every little detail

jgauchia, 2017-12-08:
@iosonopersia Ok

First of all, Firefox in Debian 9
OK,
I upload my Configuration_Overall (from 4.3.07) and my values doesn't appears. No problem, I modify all the steps in the configuration Website. 
All steps are Ok. when I click in Download Firmware appear this error

iosonopersia, 2017-12-08:
Ok, in the meanwhile, could you please send here that Overall.h from 4.3.07? I'll try doing like you did, to see if that error comes up

jgauchia, 2017-12-08:
[Configuration.zip](https://github.com/MKFirmware/MK4duo/files/1543541/Configuration.zip)

jgauchia, 2017-12-08:
this is my Configuration_Overall and Configuration_Pin

iosonopersia, 2017-12-08:
Thanks (I'm hoping very hard that you won't have other problems, this issue is getting too old ;-) )

MagoKimbra, 2017-12-08:
O_O But this file does not contain the json database, you created it by hand and it can not be imported!

jgauchia, 2017-12-08:
@MagoKimbra Well, Sorry I don't know it.
In the website says import Configuration_Overall I suposed that manual configuration works.
Sorry, :(

MagoKimbra, 2017-12-08:
Attention! Only Configuration.h or Configuration_Overall.h files generated by this configuration tool can be uploaded!

generated by this configuration tool can be uploaded...

jgauchia, 2017-12-08:
:( sorry again

MagoKimbra, 2017-12-08:
do not problem!!! 👍

jgauchia, 2017-12-08:
Finally I've the firmware, but I'll do some manual modifications

jgauchia, 2017-12-08:
Well after some manual modifications in configuration pin & overall (my printer is "special" ) and without NEXTION
[homming.zip](https://github.com/MKFirmware/MK4duo/files/1543905/homming.zip)
, I've some issues in homming both extruders.
There are independent extruder
I've defined X_TWO_STEPPER & DUAL_X_CARRIAGE, DXC_AUTO_PARK_MODE , 
[homming.zip](https://github.com/MKFirmware/MK4duo/files/1543904/homming.zip)
but homming is not correct .
I've tried some configuration without success
Video : 
[
https://mega.nz/#!SdYTSKxZ
](url)

MagoKimbra, 2017-12-08:
Sorry but i do not understand where is the problem!!!

jgauchia, 2017-12-08:
In previous version of the firmware, when homming the X axis, first home extruder 0 and later, in opposite movement the second.
In this version both extruders are moving at the same time. In this case the second extruder is not parked outside the impression area .
This video show how actually works X homming (filmed when I tested the 4.3.07 version for first time)

https://mega.nz/#!2URV0DzZ!nMN5BpbW1diciyJ5Sv96z4_wqMWQKLFJu1iNS8O6k1o

MagoKimbra, 2017-12-08:
A ok it's normal. Now home the second extrudet (to the right), remember this extruder's position for later tool change, home the 1st (left) extruder and consider the active extruder to be parked.
When you change tools move the second extruder to park position and move the first extruder...

jgauchia, 2017-12-08:
Ah, ok, 
Now I'm printing, Tomorrow I'll upload the new firmware and test some printing.

jgauchia, 2018-01-18:
Hi, 
I've tested and tested but doesn't works well, 
Once extruder are parked, and start to print both extruders starts to move in wrong directions.
I'll try to upload the new firmware and test again when I'll have a little time to do it.

Thank you!

apphead, 2018-02-27:
@jgauchia
With latest 4.3.5, dualx homing works fine (have you noticed INVERT_X2_VS_X_DIR ?), but when I switch to extruder 2, it moves to right border of the bed, but x=0 where it should be X2_HOME_POS(?).
Have you made progress? If so, would you mind sharing your config?

iworld4us, 2018-02-28:
Set x>0 for extruder 2 only in Cura for correct printing.

[12 more replies]

closed 2018-11-22 by MagoKimbra