MKFirmware/MK4duo #149: Totall error

Tick007, 2017-06-24:
1. Compilation error. In file core_mechanics on line 54: axis_steps = stepper.position((axis);
I've fixed this. 
2. After building and uploading all power otput set to on. I don't often see hot end temperature higher than 400 celsius degrees. Didn't know it was possible.
This last firmware almost killed my printer.

Tick007, 2017-06-25:
So I've made fresh copy of latest master revision. I've played with INVERT_FAN_PIN and other invert keys - still no result - all power outputs is always on.
I've rollback to last V4.3.2.

MagoKimbra, 2017-06-26:
I use the latest version on Wanhao D6 board 8 bit, and a delta with 32 bit board and I have no problem getting out on always.
What card do you use?

Tick007, 2017-06-26:
Hello! I use DUE + RAMPS-FD.

MagoKimbra, 2017-06-26:
Ok try now!

Tick007, 2017-06-27:
Hello!
I've just tested:
bed is ok
Cooler and hotend is on.  INVERT_COOLER_PIN and INVERT_HEATER_PIN does not affect.

Tick007, 2017-06-27:
Just updated to 24c8077c913c445a28f42ad67ceba01596cd633b
Up to now:
1. cooler - is on on startup, but can be controlled invertedly - 100% - off, off - 100%. Doesn't affected by INVERTED_COOLER_PIN.
2. bed - is off, can't be controlled, always off
3. hotend - OK with #define INVERTED_HEATER_PINS uncommented.

MagoKimbra, 2017-06-27:
cooler is not fan... Cooler is the liquid cooler for hotend... Have you Cooler??
Check for bed!

Tick007, 2017-06-27:
Sorry, I thought that were the same. Let make it clear. I talk about FAN and fan pin. 
```
// FAN pin
#define FAN_PIN             10//ORIG_FAN_PIN
```
So that power output is ON by default.

In previus firmware versions I had to make appropriate changes to conditionals_post.h to make fan work as I need with INVERTED_COOLER_PIN var.
For example in 4.3.2 there was 
```
#if ENABLED(INVERTED_FAN_PINS) 
    #define _WRITE_FAN(pin, v) WRITE(pin, !v)
  #else
    #define _WRITE_FAN(pin, v) WRITE(pin, v)
  #endif
```
So to make it work I've changed it to 
```
#if ENABLED(INVERTED_FAN_PINS) || ENABLED(INVERTED_COOLER_PIN) 
    #define _WRITE_FAN(pin, v) WRITE(pin, !v)
  #else
    #define _WRITE_FAN(pin, v) WRITE(pin, v)
  #endif
```
up to now bed doesn't correspond to commands and always set to OFF.
fan (that I called cooler before) is corresponding in inverted mode.

MagoKimbra, 2017-06-27:
Noo do you need define INVERTED_FAN_PINS you found it in configuration_feature.h

Tick007, 2017-06-27:
I tried - no result. Still responds in inverted mode.

Tick007, 2017-06-27:
And about cooler: in russian we understand fan as fan, and cooler as fan + heat sink (radiator). So there are  both the fan and the heat sink on hot end , so I've called it as "cooler", thinking about controlling the fan via power output 10 (FAN_PIN). As I was wrong about that, may be you should add a small note about INVERTED_COOLER_PIN ?

Tick007, 2017-06-27:
The code 
```
  #if ENABLED(INVERTED_FAN_PINS)
    #define _WRITE_FAN(pin, v) WRITE(pin, !v)
  #else
    #define _WRITE_FAN(pin, v) WRITE(pin, v)
  #endif

  #if HAS_FAN0
    #define WRITE_FAN(v) _WRITE_FAN(FAN_PIN, v)
    #define WRITE_FAN0(v) WRITE_FAN(v)
  #endif
  #if HAS_FAN1
    #define WRITE_FAN1(v) _WRITE_FAN(FAN1_PIN, v)
  #endif
  #if HAS_FAN2
    #define WRITE_FAN2(v) _WRITE_FAN(FAN2_PIN, v)
  #endif
  #if HAS_FAN3
    #define WRITE_FAN3(v) _WRITE_FAN(FAN3_PIN, v)
  #endif
  #define WRITE_FAN_N(n, v) WRITE_FAN##n(v)
```
defenetly worked in previus firmwares. Why it does not now - i don't know.
I've tried to change
```
#define _WRITE_FAN(pin, v) WRITE(pin, v)
```
for
```
#define _WRITE_FAN(pin, v) WRITE(pin, !v)
```
to atchive invercion in any case, but the fan still works in inverted mode.

MagoKimbra, 2017-06-27:
Sorry my error, I forgot to put #define INVERTED_FAN_PINS in configuration_feature.h

Tick007, 2017-06-27:
I see it. But the code above still doesn't chande behivor of power output. Does it  work on your printer ?

MagoKimbra, 2017-06-27:
The problem is the ramps-fd that has the outputs with inverted logic, with all the other boards I have no problems ... Only in the latest version i have inserted the hardware PWM, and it works fine with the normal logic but i can not Find the solution with inverted logic ...

Tick007, 2017-06-27:
Ok, so what can you recommend for me ? Continue to use firmware 4.3.2. ?

Tick007, 2017-06-27:
I've installed git for windows some time ago and it was very comfortable to receive new version of firmware every time I use my printer. But..... you know what happened lately. But 4.3.2 looks OK for me.

Tick007, 2017-06-27:
Ok, thank you. Bed problem - has appeared in last firmware too. 
So if is it all for now - please close this issue.
I will sum up. Best decision for RAMPS_FD - previos firmware 4.3.2 (branch).

commit 4f4879c9 (2017-06-29): Fix #149

I think fexed it!
closed 2017-06-29 by MagoKimbra