MKFirmware/MK4duo #150: error Z_ENDSTOP_SERVO_NR

youssifgh, 2017-06-26:
This error appeared after Update V4_3_23_dev

Generating function prototypes...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ 
In file included from C:\Users\youss\AppData\Local\Temp\arduino_build_674022\sketch\base.h:83:0,

                 from D:\Meus docs\Projetos\Arduino\Sketchbook Arduino\MK4duo\MK4duo.ino:248:

C:\Users\youss\AppData\Local\Temp\arduino_build_674022\sketch\src/sanitycheck.h:1667:4: error: #error DEPENDENCY ERROR: At least one of the Z_ENDSTOP_SERVO_NR is required for DEACTIVATE_SERVOS_AFTER_MOVE.

   #error DEPENDENCY ERROR: At least one of the Z_ENDSTOP_SERVO_NR is required for DEACTIVATE_SERVOS_AFTER_MOVE.

    ^

Usando a biblioteca LiquidCrystal na versão 1.0.5 na pasta: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal 
exit status 1
Erro compilando para a placa Arduino/Genuino Mega or Mega 2560

MagoKimbra, 2017-06-26:
do you set Z_ENDSTOP_SERVO_NR?

youssifgh, 2017-06-26:
yes
// Z Servo Endstop
// Remember active servos in Configuration_Feature.h
// Define nr servo for endstop -1 not define. Servo index start 0
#define Z_ENDSTOP_SERVO_NR 0
#define Z_ENDSTOP_SERVO_ANGLES {139,56} // Z Servo Deploy and Stow angles

commit 55c39b17 (2017-06-26): Fix #150
closed 2017-06-26 by MagoKimbra