MKFirmware/MK4duo #156: SSD1306 Display

szymonm2, 2017-07-03:
I get a blank screen when I enable the ssd1306 by commenting the following lines on an arduino due (no shield installed)
> #define U8GLIB_SSD1306

> #define DOGLCD

The display is connected to pins 21 and 20 and when I run a graphics test from the u8glib library everything works? What am I doing anything wrong?

iosonopersia, 2017-12-08:
Hi @szymonm2 , I'm very sorry for my late response. We would like to inform you that a new version of the firmware came out this days: `4.3.3 stable` ! We encourage you to try our latest version using  [this configurator (link)](http://marlinkimbra.it/configurator/v4_3_3/firmware_configurator.php). We do really hope that it will work for you, since we put a lot of effort in it. Feel free to report every problem/bug you may find.

jf1452, 2018-01-01:
Hello all, sorry to piggy back onto this post (should I create a new one?).

I am having a related issue with the MKS 12864 OLED display; I have used 4.3.3 stable but get the following compilation issue..

sketch\src\lcd\ultralcd_impl_DOGM.h:203:42: error: 'DOGLCD_MOSI' was not declared in this scope

U8GLIB_SH1106_128X64 u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);      // 8 stripes

sketch\src\lcd\ultralcd_impl_DOGM.h:203:55: error: 'DOGLCD_CS' was not declared in this scope

U8GLIB_SH1106_128X64 u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);      // 8 stripes

sketch\src\lcd\ultralcd_impl_DOGM.h:203:66: error: 'DOGLCD_A0' was not declared in this scope

Any help or advice would be very much appreciated.

Thanks

Jon

MagoKimbra, 2018-01-01:
On board??

jf1452, 2018-01-01:
Apologies, I have Arduino Due with Ramps-FD modified to V2 level.

Many thanks

MagoKimbra, 2018-01-01:
Ok this board havent't mapped pin for this lcd... If you write me the pins i added to file...

jf1452, 2018-01-01:
Perfect, thank you very much. I will go and get all the schematics and work out the connections; I will add the data in a few days (I need to print some of the data out and re-draw the circuits).

Many thanks for your help, and have a very happy New year!

Cheers

Jon

jf1452, 2018-01-03:
Hi MagoKimbra,

Thank you very much for your patience and support. I have traced the pin connections from the MKS LCD as follows:

MKS_OLED_PIN, EXP1_PIN, ORIGINAL_LCD_FUNCTION, RAMPS-FD_AUX4_PIN, ARDUINO_PIN

**MOSI**, 3, E, 17, **D17**
**CS**, 4, RS, 18, **D16**
**CLK**, 5, D4, 16, **D23**
**DC**, 6, D5, 15, **D25**
**RES**, 7, D6, 14, **D27**

I guess you are only interested in the **BOLD** items.

Thank you

Jon

iosonopersia, 2018-07-31:
You can safely close this very old issue, @MagoKimbra . These pins were already put in 1404.h!

closed 2018-07-31 by MagoKimbra