MKFirmware/MK4duo #160: Compile Error: leveling fade height

Kuro-Maii, 2017-07-07:
while compiling the current build of Marlin Kimbra via the [Configurator](http://marlinkimbra.it/configurator/v4_3_x/firmware_configurator.php) arduino fails to compile with the following error:
```
Auto-set to: AVR_MEGA2560
In file included from sketch/base.h:83:0,
                 from /home/angel/projects/others/ultratronics/MK4duo/MK4duo.ino:257:
sketch/src/sanitycheck.h:739:4: error: #error "ENABLE_LEVELING_FADE_HEIGHT requires Bed Level"
   #error "ENABLE_LEVELING_FADE_HEIGHT requires Bed Level"
    ^
exit status 1
Error compiling for board Arduino Due (Programming Port).
```
As I need to manually level the bed I looked the `#define ENABLE_LEVELING_FADE_HEIGHT` up and commented it out.
recompiling results in the following error:
```
sketch/src/motion/stepper.cpp:558:40: error: missing binary operator before token "("
     #define _COUNT_STEPPERS_4 INCREMENT(_COUNT_STEPPERS_3)
                                        ^
sketch/src/motion/stepper.cpp:563:30: note: in expansion of macro '_COUNT_STEPPERS_4'
   #define CYCLES_EATEN_XYZE (_COUNT_STEPPERS_4 * 5)
                              ^
sketch/src/motion/stepper.cpp:564:51: note: in expansion of macro 'CYCLES_EATEN_XYZE'
   #define EXTRA_CYCLES_XYZE (STEP_PULSE_CYCLES - (CYCLES_EATEN_XYZE))
                                                   ^
sketch/src/motion/stepper.cpp:620:9: note: in expansion of macro 'EXTRA_CYCLES_XYZE'
     #if EXTRA_CYCLES_XYZE > 20
         ^
sketch/src/motion/stepper.cpp:558:40: error: missing binary operator before token "("
     #define _COUNT_STEPPERS_4 INCREMENT(_COUNT_STEPPERS_3)
                                        ^
sketch/src/motion/stepper.cpp:563:30: note: in expansion of macro '_COUNT_STEPPERS_4'
   #define CYCLES_EATEN_XYZE (_COUNT_STEPPERS_4 * 5)
                              ^
sketch/src/motion/stepper.cpp:564:51: note: in expansion of macro 'CYCLES_EATEN_XYZE'
   #define EXTRA_CYCLES_XYZE (STEP_PULSE_CYCLES - (CYCLES_EATEN_XYZE))
                                                   ^
sketch/src/motion/stepper.cpp:652:9: note: in expansion of macro 'EXTRA_CYCLES_XYZE'
     #if EXTRA_CYCLES_XYZE > 20
         ^
sketch/src/motion/stepper.cpp:558:40: error: missing binary operator before token "("
     #define _COUNT_STEPPERS_4 INCREMENT(_COUNT_STEPPERS_3)
                                        ^
sketch/src/motion/stepper.cpp:563:30: note: in expansion of macro '_COUNT_STEPPERS_4'
   #define CYCLES_EATEN_XYZE (_COUNT_STEPPERS_4 * 5)
                              ^
sketch/src/motion/stepper.cpp:564:51: note: in expansion of macro 'CYCLES_EATEN_XYZE'
   #define EXTRA_CYCLES_XYZE (STEP_PULSE_CYCLES - (CYCLES_EATEN_XYZE))
                                                   ^
sketch/src/motion/stepper.cpp:655:11: note: in expansion of macro 'EXTRA_CYCLES_XYZE'
     #elif EXTRA_CYCLES_XYZE > 0
           ^
sketch/src/motion/stepper.cpp:558:40: error: missing binary operator before token "("
     #define _COUNT_STEPPERS_4 INCREMENT(_COUNT_STEPPERS_3)
                                        ^
sketch/src/motion/stepper.cpp:563:30: note: in expansion of macro '_COUNT_STEPPERS_4'
   #define CYCLES_EATEN_XYZE (_COUNT_STEPPERS_4 * 5)
                              ^
sketch/src/motion/stepper.cpp:564:51: note: in expansion of macro 'CYCLES_EATEN_XYZE'
   #define EXTRA_CYCLES_XYZE (STEP_PULSE_CYCLES - (CYCLES_EATEN_XYZE))
                                                   ^
sketch/src/motion/stepper.cpp:743:9: note: in expansion of macro 'EXTRA_CYCLES_XYZE'
     #if EXTRA_CYCLES_XYZE > 20
         ^
sketch/src/motion/stepper.cpp:558:40: error: missing binary operator before token "("
     #define _COUNT_STEPPERS_4 INCREMENT(_COUNT_STEPPERS_3)
                                        ^
sketch/src/motion/stepper.cpp:563:30: note: in expansion of macro '_COUNT_STEPPERS_4'
   #define CYCLES_EATEN_XYZE (_COUNT_STEPPERS_4 * 5)
                              ^
sketch/src/motion/stepper.cpp:564:51: note: in expansion of macro 'CYCLES_EATEN_XYZE'
   #define EXTRA_CYCLES_XYZE (STEP_PULSE_CYCLES - (CYCLES_EATEN_XYZE))
[... 87 lines cut ...]
sketch/src/motion/stepper.cpp:1448:22: error: 'count_position' was not declared in this scope
   const long  xpos = count_position[X_AXIS],
                      ^
sketch/src/motion/stepper.cpp:1458:25: error: 'SERIAL_M' was not declared in this scope
     SERIAL_M(MSG_COUNT_X);
                         ^
sketch/src/motion/stepper.cpp:1460:16: error: 'SERIAL_V' was not declared in this scope
   SERIAL_V(xpos);
                ^
sketch/src/motion/stepper.cpp:1469:12: error: 'ypos' was not declared in this scope
   SERIAL_V(ypos);
            ^
sketch/src/motion/stepper.cpp:1478:12: error: 'zpos' was not declared in this scope
   SERIAL_V(zpos);
            ^
sketch/src/motion/stepper.cpp:1480:3: error: 'SERIAL_E' was not declared in this scope
   SERIAL_E;
   ^
exit status 1
Error compiling for board Arduino Due (Programming Port).
```

I have included my config file in case it may contain some information required that I am overlooking

[Configuration_Overall.h.txt](https://github.com/MagoKimbra/MK4duo/files/1130615/Configuration_Overall.h.txt)

iosonopersia, 2017-07-07:
That's so strange... It says there are problems in src/motion/stepper.cpp

But the "motion" folder doesn't even exist and stepper.cpp is located in src/stepper/stepper.cpp

Kuro-Maii, 2017-07-07:
then there are probably some bugs in the configuration site as well... as the folder structure is correct with the error messages and the zip file

iosonopersia, 2017-07-07:
Are you using the latest version of Arduino IDE? (1.8.3)

Kuro-Maii, 2017-07-07:
I missed your edit on your previous post and tried it out after posting my comment, and this gave me different errors:
```
In file included from sketch/base.h:83:0,
                 from /home/angel/projects/others/ultratronics/MK4duo/MK4duo.ino:257:
sketch/src/sanitycheck.h:739:4: error: #error "ENABLE_LEVELING_FADE_HEIGHT requires Bed Level"
   #error "ENABLE_LEVELING_FADE_HEIGHT requires Bed Level"
    ^
sketch/src/sanitycheck.h:2034:4: error: #error DEPENDENCY ERROR: You have to set at least one HOTEND_?_AUTO_FAN_PIN to a valid pin if you enable HOTEND_AUTO_FAN
   #error DEPENDENCY ERROR: You have to set at least one HOTEND_?_AUTO_FAN_PIN to a valid pin if you enable HOTEND_AUTO_FAN
    ^
exit status 1
Error compiling for board Arduino Due (Programming Port).
```

then I saw your most resent post and updated the IDE to the current version. thanks for pointing this out to me.
( I wish the ubuntu repo would update its arduino version, safes me from manually checking. but those are reports for some where else. )

sadly the bug is the same regardless of arduino version.

Kuro-Maii, 2017-07-07:
however commenting out `#define ENABLE_LEVELING_FADE_HEIGHT` resolves that issue and leaves me with `DEPENDENCY ERROR: You have to set at least one HOTEND_?_AUTO_FAN_PIN to a valid pin if you enable HOTEND_AUTO_FAN`

iosonopersia, 2017-07-07:
If you don't use the BED LEVELING feature, you have to be sure that no one of the BED LEVELING options are activated, because them being active while BED LEVELING is not will of course cause an error. You must check with a lot of attention the configurator, for example you can activate the bed leveling feature in order to make the configurator show you the bed leveling options: even if you did not activate bed leveling is possible that some of its sub-options are active (nothing is perfect in this world, after all).

Speaking of the error you have with the HOTEND AUTO FAN feature, I don't know why that happened. You can try defining that pin in your Configuration Overall:
#define H0_AUTO_FAN_PIN <pinNumber>

I hope you'll be able to get MK4duo compiling

Kuro-Maii, 2017-07-07:
after doing some thorough searching I found that the missing link is in Configuration_Pins.h this file contains:
```
#if ENABLED(HOTEND_AUTO_FAN)
  #define H0_AUTO_FAN_PIN ORIG_FAN1_PIN
  #define H1_AUTO_FAN_PIN -1
  #define H2_AUTO_FAN_PIN -1
  #define H3_AUTO_FAN_PIN -1
#endif
```
which was all set to `-1` inspecting earlier configurator builds it was indeed set to the correct pin `ORIG_FAN1_PIN`

so currently the git repo is compiling, but only after I have overwritten the files `Configuration_Overall.h` and `Configuration_Pins.h` with the ones I get from the configurator.
I hope I am not missing any "hidden" configs that may have gone unnoticed thanks to this way of patching.

MagoKimbra, 2017-07-07:
But you set it as original fan pin 1, so the compiler put it in, the fact that you have to know if your card has the fan1 or not.
Also for the fade height problem, you set it up and then disable the bed level this way but the fade height is set! Checking things that do the configurator does not have an error check otherwise it becomes another program ...

Kuro-Maii, 2017-07-07:
I want to make a few things clear to prevent confusion:
- I have been running MarlinKimbra via the configurator as off 2017-03-22 17:13:52 ( per date time stamp from the downloaded zip )
- that was the day that I installed a [ultratornics pro](https://reprapworld.com/products/electronics/ultratronics/ultratronics_pro_v1_0/) on my printer with new [RAPS128](https://reprapworld.com/products/electronics/stepper_drivers/raps128_1_128_stepresolution_stepper_driver/) and [Graphical LCD](https://reprapworld.com/products/printers_printing/autonomous_printing/graphical_lcd_screen_v1_0/)
- yesterday I installed an [E3D v6](https://reprapworld.com/products/e3d/e3d_v6/kits/e3d_v6_hotend_full_kit_3mm_direct_12v/) to solve hardware issues
- after the installation was complete I ran the configurator
( fast forwarding a bit )
- `ENABLE_LEVELING_FADE_HEIGHT` started happening
- `Configuration_Overall.h` input file for configurator was fixed `ENABLE_LEVELING_FADE_HEIGHT` no longer an issue
- the following [configurator compile error](https://github.com/MagoKimbra/MK4duo/files/1132070/configurator_issue.txt) started to happen and has *not* been sovled
- successful workaround made with git repo (commit id `c0972ddc31c4e6624b8e28f8968edaa995cd0495`) and the following configurator files: `Configuration_Overall.h` and `Configuration_Pins.h`

Kuro-Maii, 2017-08-03:
closing this issue as the original issue was due to obscured user error

closed 2017-08-03 by Kuro-Maii

Kuro-Maii, 2017-08-03:
was going to make a new issue for the remainder of the problem.
then I decided to run a diff on the contents of the zip and the extracted folder that was already there and at a point it actually compiles, results of the test:
- files where moved to new locations
- files where renamed
conclutions:
- me extracting over the already existing folder doubled up on files
- compilers and linkers get confused
what I learned:
- always empty out the dir before extracting a new version when using the configurator site

sorry for the inconvenience and confusion that I may have caused.