MKFirmware/MK4duo #165: "CONTROLLERFAN" Function is not reacting on the Steppers state.

demblack, 2017-07-09:
Hi,
I wanted to get the "controller fan function" going with this special printer i've built. 

void controllerFan() {

in MK_Main.cpp

However as with the other instances i've read online unfortunately the fan could be ether constantly "on" or "off". 
After looking into the code and monitoring the corresponding variables i've noticed that none of the flags "X_ENABLE_READ, Y_ENABLE_READ, ..."(all axis the same) will ever be high or not equal to"0". Due to the routine of monitoring if the Steppers are "on" (Enabled) or "off" is always adjusting "lastMotorOn" from very beginning which means it does not matter if the Steppers are on or off the clock is ticking...

if (X_ENABLE_READ == X_ENABLE_ON || Y_ENABLE_READ == Y_ENABLE_ON ...
...     
      ) {
        lastMotorOn = ms; //... set time to NOW so the fan will turn on
      }
However the line:
controller_fanSpeeds = (!lastMotorOn || ELAPSED(ms, lastMotorOn + (CONTROLLERFAN_SECS) * 1000UL)) ? 0 : CONTROLLERFAN_SPEED;
is working absolutely fine as soon as  lastMotorOn stop bein updated constantly.
Now, the problem i have,  i could not fine the way to get these flags "X_ENABLE_READ..." properly reflecting the steppers State neither directly reading the ENABLE_PIN of the driver.

For example:
bool stepperX_state;
stepperX_state = READ(X_ENABLE_PIN);
...
For some reason it comes always "low" back for all axes regardless if Steppers are enabled or disabled or moving.
Probably solution is easy... Many thanks in advance for some clue.

iosonopersia, 2017-09-21:
Hi @demblack , sorry for the delay... Is this still happening with the latest version of the firmware?

demblack, 2017-09-21:
Hi **iosonopersia** , thanks a lot for your replay.  I haven't try the latest Firmware yet as I've been trying to find the root cause myself. Honestly it was not successful as I couldn't figured out where exactly the code enables/disables the Steppers. it bugged me so deep that I've spent many hours playing and understanding the code(well... trying). It must be because of the HAL layer and the mechanism I do not yet understand how the motors being controlled thru it. Is there any docs or something one can read to this topic. I believe it belongs to the topic "SAM3X8E (Arduino Due) Pin IO registers"
perhaps hier:
http://www.atmel.com/Images/Atmel-11057-32-bit-Cortex-M3-Microcontroller-SAM3X-SAM3A_Datasheet.pdf
Interesting, all functions which are trying to reach the state of the Stepper always return LOW... even when Stepper is moving, it could be only because they not talking to the controller directly... however controller PIN is proper "ON"... mystery...

Anyways I'll try the latest software and I'll leave a comment here then.  As a mater of the interest, what exactly have you changed? If I may...

iosonopersia, 2017-12-08:
Hi @demblack , I'm very sorry for my late response. We would like to inform you that a new version of the firmware came out this days: `4.3.3 stable` ! We encourage you to try our latest version using  [this configurator (link)](http://marlinkimbra.it/configurator/v4_3_3/firmware_configurator.php). We do really hope that it will work for you, since we put a lot of effort in it. Feel free to report every problem/bug you may find.