MKFirmware/MK4duo #177: Compilation-Error with TMC2130

tekknix, 2017-07-17:
Hi,

trying to adapt this Firmware for my MKS-GEN v1.4 Board and Prusa-i3 Style Printer.

I'm using 3 TMC2130-Stepperdrivers for X/Y/Z-Axis running in 16/256-Mode.

Compiling with TMC2130 isn't possible.

```
sketch\src\MK_Main.cpp: In function 'void gcode_M906()':

sketch\src\MK_Main.cpp:8231:60: error: invalid initialization of reference of type 'TMC2130Stepper&' from expression of type 'uint16_t {aka unsigned int}'

       if (values[X_AXIS]) tmc2130_set_current(values[X_AXIS], stepperX, 'X');

                                                            ^

sketch\src\MK_Main.cpp:8181:15: note: in passing argument 1 of 'void tmc2130_set_current(TMC2130Stepper&, char, int)'

   static void tmc2130_set_current(TMC2130Stepper &st, const char name, const int mA) {

               ^

sketch\src\MK_Main.cpp:8235:60: error: invalid initialization of reference of type 'TMC2130Stepper&' from expression of type 'uint16_t {aka unsigned int}'

       if (values[Y_AXIS]) tmc2130_set_current(values[Y_AXIS], stepperY, 'Y');

                                                            ^

sketch\src\MK_Main.cpp:8181:15: note: in passing argument 1 of 'void tmc2130_set_current(TMC2130Stepper&, char, int)'

   static void tmc2130_set_current(TMC2130Stepper &st, const char name, const int mA) {

               ^

sketch\src\MK_Main.cpp:8239:60: error: invalid initialization of reference of type 'TMC2130Stepper&' from expression of type 'uint16_t {aka unsigned int}'

       if (values[Z_AXIS]) tmc2130_set_current(values[Z_AXIS], stepperZ, 'Z');

                                                            ^

sketch\src\MK_Main.cpp:8181:15: note: in passing argument 1 of 'void tmc2130_set_current(TMC2130Stepper&, char, int)'

   static void tmc2130_set_current(TMC2130Stepper &st, const char name, const int mA) {

               ^

exit status 1
```

I'm using latest Arduino IDE with latest TMC2130-Stepper-Library

Regards
MoS-tekknix

tekknix, 2017-07-17:
Mhh, aftet looking around it should work in modifying MK_main.cpp in that way:

Remove

```
  inline void gcode_M906() {
    uint16_t values[NUM_AXIS];
    LOOP_XYZE(i)
      values[i] = parser.seen(axis_codes[i]) ? parser.value_int() : 0;

    #if ENABLED(X_IS_TMC2130)
      if (values[X_AXIS]) tmc2130_set_current(values[X_AXIS], stepperX, 'X');
      else tmc2130_get_current(stepperX, 'X');
    #endif
    #if ENABLED(Y_IS_TMC2130)
      if (values[Y_AXIS]) tmc2130_set_current(values[Y_AXIS], stepperY, 'Y');
      else tmc2130_get_current(stepperY, 'Y');
    #endif
    #if ENABLED(Z_IS_TMC2130)
      if (values[Z_AXIS]) tmc2130_set_current(values[Z_AXIS], stepperZ, 'Z');
      else tmc2130_get_current(stepperZ, 'Z');
    #endif
    #if ENABLED(E0_IS_TMC2130)
      if (values[E_AXIS]) tmc2130_set_current(values[E_AXIS], stepperE0, 'E');
      else tmc2130_get_current(stepperE0, 'E');
    #endif

    #if ENABLED(AUTOMATIC_CURRENT_CONTROL)
      if (parser.seen('S')) auto_current_control = parser.value_bool();
    #endif
  }
```

Insert

```
 inline void gcode_M906() {
    uint16_t values[XYZE];
    LOOP_XYZE(i)
      values[i] = parser.intval(axis_codes[i]);

    #if ENABLED(X_IS_TMC2130)
      if (values[X_AXIS]) tmc2130_set_current(stepperX, 'X', values[X_AXIS]);
      else tmc2130_get_current(stepperX, 'X');
    #endif
    #if ENABLED(Y_IS_TMC2130)
      if (values[Y_AXIS]) tmc2130_set_current(stepperY, 'Y', values[Y_AXIS]);
      else tmc2130_get_current(stepperY, 'Y');
    #endif
    #if ENABLED(Z_IS_TMC2130)
      if (values[Z_AXIS]) tmc2130_set_current(stepperZ, 'Z', values[Z_AXIS]);
      else tmc2130_get_current(stepperZ, 'Z');
    #endif
    #if ENABLED(E0_IS_TMC2130)
      if (values[E_AXIS]) tmc2130_set_current(stepperE0, 'E', values[E_AXIS]);
      else tmc2130_get_current(stepperE0, 'E');
    #endif

    #if ENABLED(AUTOMATIC_CURRENT_CONTROL)
      if (parser.seen('S')) auto_current_control = parser.value_bool();
    #endif
  }
```

MagoKimbra, 2017-07-17:
Thank's

closed 2017-07-17 by MagoKimbra

tekknix, 2017-07-17:
Hi,

currently i would suggest to remove TMC-Support, cause there are so many minssconfigurations inside. And even my suggested Fix is only a litte step.

Being working on it.

Cheers
MoS-tekknix

Von: MagoKimbra
Gesendet: Montag, 17. Juli 2017 21:29
An: MKFirmware/MK4duo
Cc: tekknix; Author
Betreff: Re: [MKFirmware/MK4duo] Compilation-Error with TMC2130 (#177)

Thank's

commit c44fc214 (2017-07-17): fix #177