MKFirmware/MK4duo #182: FAN PWM error

Windaub57, 2017-07-20:
Installed update 4.3.25, previously was using 4.3.20.

When printer starts if hotend_auto_start is enabled I get in the com windows of simplify3D:

assertion "duty <= pPwm->PWM_CH_NUM[ul_channel].PWM_CPRD" failed: file "../source/pwmc.c", line 272, function: PWMC_SetDutyCycle

Printer stays stuck after that.

iosonopersia, 2017-12-08:
Hi @Windaub57 , I'm very sorry for my late response. We would like to inform you that a new version of the firmware came out this days: `4.3.3 stable` ! We encourage you to try our latest version using  [this configurator (link)](http://marlinkimbra.it/configurator/v4_3_3/firmware_configurator.php). We do really hope that it will work for you, since we put a lot of effort in it. Feel free to report every problem/bug you may find.

iosonopersia, 2018-07-31:
@MagoKimbra I believe this old issue is not directly related to MK4duo code. I feel like you can close this issue.

closed 2018-07-31 by MagoKimbra