MKFirmware/MK4duo #183: Language translation and Pre-Heating mod

carlonb, 2017-07-21:
Hi all, 
I've modified the english language: from "End" to "HotEnd", obviously for all 3 preheating materials (PLA, ABS, GUM).
```
#ifndef MSG_PREHEAT_1_END
  #define MSG_PREHEAT_1_END                   MSG_PREHEAT_1 _UxGT(" HotEnd")
#endif
....
....
```

After that I've modified the file ultralcd.cpp in order to have the "ALL" in menu Prepare->Preheat when we have only one HotEnd and a hot bed, I will display "Preheat PLA All" instead of only "Preheat PLA", of course I've mod. also for ABS and GUM.
```
      #if HOTENDS == 1
        #if HAS_TEMP_BED
          MENU_ITEM(function, MSG_PREHEAT_1_ALL, lcd_preheat_m1_h0); 
          MENU_ITEM(function, MSG_PREHEAT_1_END, lcd_preheat_m1_h0_only);
        #else
.....
.....
```
All these seems to work fine with my 128x64 graphic LCD.
What do you think about it? Am I completely out of the track?
Ciao
Carlo

iosonopersia, 2017-08-15:
Hi Carlo, can you please open a Pull Request with your modifications so that we can understand better what you're suggesting

carlonb, 2017-08-15:
Ciao Simone,
I never tried to open a PR but I will try. Anyway may modifications are very simple and you can see it here on top.
I've modded the english "language_en.h" (I love use the english language) changing:
MSG_PREHEAT_1   _UxGT(" End")              ---->   MSG_PREHEAT_1   _UxGT(" HotEnd")
MSG_PREHEAT_2   _UxGT(" End")              ---->   MSG_PREHEAT_2   _UxGT(" HotEnd")
MSG_PREHEAT_2   _UxGT(" End")              ---->   MSG_PREHEAT_3   _UxGT(" HotEnd")
HotEnd is better than only End. This is done for PLA, ABS and GUM preheat.

After that, as I use the hot bed and only one HotEnd, I've modified the file "ultralcd.cpp" in order to show on display "Preheat PLA All" instead of "Preheat PLA", to me this is more understandable as means preheat both. This must be done also for ABS and GUM preheat. 
If you find these lines on file "ultralcd.cpp":
```
      #if HOTENDS == 1
        #if HAS_TEMP_BED
          MENU_ITEM(function, MSG_PREHEAT_1_ALL, lcd_preheat_m1_h0); 
          MENU_ITEM(function, MSG_PREHEAT_1_END, lcd_preheat_m1_h0_only);
        #else
.....
```
You can understand better.

iosonopersia, 2017-08-15:
Ok, if you prefer I can do it for you ;)

carlonb, 2017-08-15:
Sure, do it if I'm correct :)
Thanks

iosonopersia, 2017-08-15:
MagoKimbra: 

> E' stata fatta questa scelta per la lunghezza, non so se va fuori dal display..
> Secondo per avere lo stesso numero di caratteri ed aver cosi un incolonnamento ALL BED END
> Quando hai più hotend c'è da mettere il numero quindi diventa cosi e se conti sono 20 caratteri, ma ci devi mettere lo spazio per il selettore >
> Preheat PLA Hotend 1
> 
> MSG_PREHEAT_1_ALL è il messaggio quando devi riscaldare tutto quando hai più hotend.
> Mentre se hai un solo hotend va bene come messaggio Preheat PLA..

iosonopersia, 2017-08-15:
This issue can be closed.

carlonb, 2017-08-16:
OK, understood.
I like my idea so I keep it in my printer :).
About the lenght of message on display, what happen with the italian language ?
Ciao

closed 2017-08-16 by carlonb