MKFirmware/MK4duo #184: Ramps4Due- UltraLCD

jmlcatarino, 2017-07-21:
Trying to compile for my laser and after changing to REPRAP_DISCOUNT_SMART_CONTROLLER, gives me this error:

sketch/src/lcd/ultralcd.cpp:56:0:
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_26025/sketch/src/lcd/ultralcd_impl_HD44780.h:276:34: error: 'const byte clock [8]' redeclared as different kind of symbol
 const static PROGMEM byte clock[8] = {
                                  
Thank you in advance

jmlcatarino, 2017-07-22:
This is after trying solution for issue #127 that I was having also.

jmlcatarino, 2017-07-31:
Any help here?

iosonopersia, 2017-08-15:
@jmlcatarino Does this issue still occur? Can you please send here your Configuration_Overall.h file?

jmlcatarino, 2017-08-15:
Hi there, yes it still happens and I have nothing on Configuration_Overall.h. It's clear. The issue happens when I try to config this kind of LCD/ controller. 

This is the complete error message:

In file included from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_359408/sketch/src/lcd/ultralcd.cpp:56:0:
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_359408/sketch/src/lcd/ultralcd_impl_HD44780.h:276:34: error: 'const byte clock [8]' redeclared as different kind of symbol
 const static PROGMEM byte clock[8] = {
                                  ^
In file included from /Users/catarino/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/stdlib.h:11:0,
                 from /Users/catarino/Library/Arduino15/packages/arduino/hardware/sam/1.6.11/cores/arduino/Arduino.h:24,
                 from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_359408/sketch/src/lcd/../../base.h:28,
                 from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_359408/sketch/src/lcd/ultralcd.cpp:23:
/Users/catarino/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/time.h:44:12: error: previous declaration of 'clock_t clock()'
 clock_t    _EXFUN(clock,    (void));
            ^
In file included from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_359408/sketch/src/lcd/ultralcd.cpp:56:0:
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_359408/sketch/src/lcd/ultralcd_impl_HD44780.h: In function 'void lcd_set_custom_characters()':
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_359408/sketch/src/lcd/ultralcd_impl_HD44780.h:352:37: error: cannot convert 'clock_t (*)() {aka long unsigned int (*)()}' to 'const byte* {aka const unsigned char*}' for argument '2' to 'void createChar_P(char, const byte*)'
   createChar_P(LCD_CLOCK_CHAR, clock);
                                     ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_359408/sketch/src/lcd/ultralcd_impl_HD44780.h: In function 'void lcd_bootscreen()':
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_359408/sketch/src/lcd/ultralcd_impl_HD44780.h:517:9: error: 'printer' was not declared in this scope
         printer.safe_delay(DELAY); \
         ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_359408/sketch/src/lcd/ultralcd_impl_HD44780.h:533:11: note: in expansion of macro 'CENTER_OR_SCROLL'
           CENTER_OR_SCROLL(STRING_SPLASH_LINE2, 2000);
           ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_359408/sketch/src/lcd/ultralcd_impl_HD44780.h:517:9: error: 'printer' was not declared in this scope
         printer.safe_delay(DELAY); \
         ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_359408/sketch/src/lcd/ultralcd_impl_HD44780.h:549:9: note: in expansion of macro 'CENTER_OR_SCROLL'
         CENTER_OR_SCROLL(STRING_SPLASH_LINE1, _SPLASH_WAIT_1);
         ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_359408/sketch/src/lcd/ultralcd_impl_HD44780.h:517:9: error: 'printer' was not declared in this scope
         printer.safe_delay(DELAY); \
         ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_359408/sketch/src/lcd/ultralcd_impl_HD44780.h:551:11: note: in expansion of macro 'CENTER_OR_SCROLL'
           CENTER_OR_SCROLL(STRING_SPLASH_LINE2, 1500);
           ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_359408/sketch/src/lcd/ultralcd_impl_HD44780.h: In function 'void lcd_implementation_status_screen()':
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_359408/sketch/src/lcd/ultralcd_impl_HD44780.h:833:26: error: 'printer' was not declared in this scope
     duration_t elapsed = printer.print_job_counter.duration();
                          ^

iosonopersia, 2017-08-15:
@MagoKimbra in library <time.h> (which is included in some strange way by <Arduino.h> or <stdlib.h>, I don't know) line 164 you can find this: `extern clock_t clock(void);`. That's a problem because in "ultralcd_impl_HD44780.h" a const byte PROGMEM array is called "clock" (line 276: `const static PROGMEM byte clock[8] = {`). This should be fixed, every occurrence of the word "clock" should be changed to something else.

It also seems to be some issue with the declaration of "printer"

closed 2017-08-16 by jmlcatarino

iosonopersia, 2017-08-17:
Hi, could you please tell us how did you managed to solve the issue? Thanks

jmlcatarino, 2017-08-17:
Hi @iosonopersia, thank you for your help. I just saw that there was an update/fix error and just downloaded and installed again. I was reading the fixes and saw there was a new lib, printer.h, so just decided to make a fresh install.  I'm not any close to an expert or whatever to code but I just try and read a lot. I am trying to upload the firmware to a Laser machine and I wanted to have it simple, with just a non graphical LCD. This time it just worked but now that other new error.

iosonopersia, 2017-08-17:
Ok, that's great!