MKFirmware/MK4duo #187: Heatedbed wont hold tempture while printing.

pyromanci, 2017-08-04:
I seem to have a problem. When i'm printing the heated bed continues to heat past the set temperature. HOWEVER it will stop at and maintain the temperature if i preheat the bed. I noticed this when I turned my back while printing a calibration print after installing a BLTouch and It stop mid print with a heatedbed over temp warning. After resetting and i watched it and just saw the heated bed temp continue to climb.

I'm not sure whats wrong. This is actually my first printer ever and I choose to go the hard route by doing a DIY clone kit just had to supply the RAMPS board. Everything is printing fine (beyond this problem a a problem with communicating with Simplify3D (oddly cura and repetier host work)).

Here is my Overall Config file.  Can someone take a look and see whats wrong with my heated bed settings. I thought i had them set right, but it's obvious I don't.

Thanks.

[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/1198982/Configuration_Overall.txt)

MagoKimbra, 2017-08-04:
Do I use the bed?
Can I know why?
However, did you make auto adjustment for the bed pid?

pyromanci, 2017-08-05:
I do have it enabled. Only because PIDTEMP was enabled and I don't have problem with the temp on the extruder.

I did not change any of the settings.

MagoKimbra, 2017-08-05:
But I'm talking about the pidbed. They're active ... You have to do Pidbed autotune.

iosonopersia, 2018-07-31:
This issue can now be safely closed, @MagoKimbra

closed 2018-07-31 by MagoKimbra