MKFirmware/MK4duo #199: HAL_spi_Due.cpp compile error

jmlcatarino, 2017-08-16:
Hi  there, after downloading updated version and with last error solved, got this new error.
I'm still trying to use the same LCD/controller.

/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_181672/sketch/src/HAL/HAL_DUE/HAL_spi_Due.cpp: In function 'void spiSend(const uint8_t*, size_t)':
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_181672/sketch/src/HAL/HAL_DUE
/HAL_spi_Due.cpp:117:51: error: 'void spiSend(const uint8_t*, size_t)' was declared 'extern' and later 'static' [-fpermissive]
   static void spiSend(const uint8_t* buf, size_t n) {
                                                   ^
In file included from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_181672/sketch/src/HAL/HAL_DUE/../../../base.h:85:0,
                 from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_181672/sketch/src/HAL/HAL_DUE/HAL_spi_Due.cpp:40:
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_181672/sketch/src/HAL/HAL_DUE/../../../src/HAL/HAL.h:47:6: error: previous declaration of 'void spiSend(const uint8_t*, size_t)' [-fpermissive]
 void spiSend(const uint8_t* buf, size_t n);

iosonopersia, 2017-08-17:
src/HAL/HAL_DUE/HAL_spi_Due.cpp line 117: 
  static void spiSend

should be only

  void spiSend

iosonopersia, 2017-08-20:
While waiting that @MagoKimbra approves my pull request, you can try

- downloading the firmware from my repository (https://github.com/iosonopersia/MK4duo)

- manually applying the edit you can find in my commit named "Fix for #199" (src/HAL/HAL_DUE/HAL_spi_due.h, line 117)

jmlcatarino, 2017-08-20:
@iosonopersia I tried your repository and it worked but it was missing a fix for "ultralcd_impl_HD44780.h", which I copied from @MagoKimbra. Now I have several others that don't make sense. But I think I know what it is. I'll just go and test a theory.

iosonopersia, 2017-08-24:
Update: the fix for this issue has been merged. Can you please update us about your situation?

jmlcatarino, 2017-08-24:
Hi there iosonopersia, I'll download the firmware again and test it again and come back with the results.

jmlcatarino, 2017-08-24:
@iosonopersia Hi there, this is the new compile error: 

In file included from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd.cpp:56:0:
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd_impl_HD44780.h:195:17: error: 'LCD_PINS_RS' was not declared in this scope
   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7
                 ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd_impl_HD44780.h:195:30: error: 'LCD_PINS_ENABLE' was not declared in this scope
   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7
                              ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd_impl_HD44780.h:195:47: error: 'LCD_PINS_D4' was not declared in this scope
   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7
                                               ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd_impl_HD44780.h:195:60: error: 'LCD_PINS_D5' was not declared in this scope
   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7
                                                            ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd_impl_HD44780.h:195:73: error: 'LCD_PINS_D6' was not declared in this scope
   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7
                                                                         ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd_impl_HD44780.h:195:86: error: 'LCD_PINS_D7' was not declared in this scope
   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7
                                                                                      ^
In file included from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd.cpp:56:0:
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd_impl_HD44780.h:276:34: error: 'const byte clock [8]' redeclared as different kind of symbol
 const static PROGMEM byte clock[8] = {
                                  ^
In file included from /Users/catarino/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/stdlib.h:11:0,
                 from /Users/catarino/Library/Arduino15/packages/arduino/hardware/sam/1.6.11/cores/arduino/Arduino.h:24,
                 from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/../../base.h:26,
                 from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd.cpp:23:
/Users/catarino/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/time.h:44:12: error: previous declaration of 'clock_t clock()'
 clock_t    _EXFUN(clock,    (void));
            ^
In file included from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd.cpp:56:0:
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd_impl_HD44780.h: In function 'void lcd_set_custom_characters()':
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd_impl_HD44780.h:352:37: error: cannot convert 'clock_t (*)() {aka long unsigned int (*)()}' to 'const byte* {aka const unsigned char*}' for argument '2' to 'void createChar_P(char, const byte*)'
   createChar_P(LCD_CLOCK_CHAR, clock);
                                     ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd_impl_HD44780.h: In function 'void lcd_implementation_init()':
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd_impl_HD44780.h:408:18: error: 'LCD_PINS_RS' was not declared in this scope
       SET_OUTPUT(LCD_PINS_RS);
                  ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd_impl_HD44780.h:409:18: error: 'LCD_PINS_ENABLE' was not declared in this scope
       SET_OUTPUT(LCD_PINS_ENABLE);
                  ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd.cpp: In function 'void lcd_tune_menu()':
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd.cpp:231:41: error: invalid conversion from 'uint16_t* {aka short unsigned int*}' to 'int16_t* {aka short int*}' [-fpermissive]
         menu_action_ ## TYPE(__VA_ARGS__); \
                                         ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd.cpp:274:9: note: in expansion of macro '_MENU_ITEM_PART_2'
         _MENU_ITEM_PART_2(type, label, ## __VA_ARGS__); \
         ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd.cpp:285:57: note: in expansion of macro 'MENU_MULTIPLIER_ITEM'
[... cut ...]
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/../../src/macros.h:83:25: note: in definition of macro '_BV'
   #define _BV(b) (1 << (b))
                         ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/../../src/lcd/ultralcd.h:181:38: note: in expansion of macro 'EN_C'
       #define LCD_CLICKED (buttons & EN_C)
                                      ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_592381/sketch/src/lcd/ultralcd.cpp:3804:9: note: in expansion of macro 'LCD_CLICKED'
     if (LCD_CLICKED) {

iosonopersia, 2017-08-24:
Errors caused by the lack of support for the display by the board:
- error: 'BLEN_C' was not declared in this scope
- error: 'LCD_PINS_RS' was not declared in this scope
- error: 'LCD_PINS_ENABLE' was not declared in this scope
- error: 'LCD_PINS_D4' was not declared in this scope
- error: 'LCD_PINS_D5' was not declared in this scope
- error: 'LCD_PINS_D6' was not declared in this scope
- error: 'LCD_PINS_D7' was not declared in this scope

Other errors:
- error: 'const byte clock [8]' redeclared as different kind of symbol
const static PROGMEM byte clock[8] = {

jmlcatarino, 2017-08-24:
I had the solution for this, but because of all the changes done lately, I can't find it or remember where I got it. But I think it was on original Marlin4Due repo issues.

iosonopersia, 2017-08-24:
#127

jmlcatarino, 2017-08-24:
Nope, no joy here. Same error. :(

MagoKimbra, 2017-08-25:
Please attach your configuration...

jmlcatarino, 2017-08-25:
@MagoKimbra what part of the config file you want, the basic or one of the others I use, like laser or features? 
@iosonopersia this is the new compile error:

In file included from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_137880/sketch/src/pins.h:33:0,
                 from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_137880/sketch/base.h:81,
                 from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_modified_sketch_872392/MK4duo.ino:261:
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_137880/sketch/src/boards/1433.h:19:6: error: stray '#' in program
  -   #define BOARD_NAME "RAMPS4DUE"
      ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_137880/sketch/src/boards/1433.h:20:5: error: stray '#' in program
  +  #define BOARD_NAME "RAMPS4DUE"
     ^
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_137880/sketch/src/boards/1433.h:19:2: error: expected unqualified-id before '-' token
  -   #define BOARD_NAME "RAMPS4DUE"
  ^

iosonopersia, 2017-08-25:
That error makes absolutely no sense to me... It says there's a # in a place where there isn't any #. Have you manually modified that file?

Please attach every Configuration file you have, plus your src/boards/1433.h

Thanks

jmlcatarino, 2017-08-25:
Yes I have Arduino IDE 1.8.3 and I did copy/paste the 1433.h file to my sketch file, instead of downloading everything again. 

This is what I have now, after changing:
/****************************************************************************************
* 1433
* Arduino Due pin assignment
* for RAMPS4DUE (http://forums.reprap.org/read.php?219,479626,page=1)
****************************************************************************************/

  //###CHIP
  #if DISABLED(ARDUINO_ARCH_SAM)
 -  #error Oops!  Make sure you have 'Arduino Due' selected from the 'Tools -> Boards' menu.
 +  #error Oops!  Make sure you have 'Alligator 3D Printer Board' selected from the 'Tools -> Boards' menu.
  #endif
  //@@@
  
  #define KNOWN_BOARD 1
  
  //###BOARD_NAME
  #if DISABLED(BOARD_NAME)
 -   #define BOARD_NAME "RAMPS4DUE"
 +  #define BOARD_NAME "RAMPS4DUE"
  #endif
  //@@@
  
 
 //###X_AXIS
 #define ORIG_X_STEP_PIN 54
 #define ORIG_X_DIR_PIN 55
 #define ORIG_X_ENABLE_PIN 38
 #define ORIG_X_CS_PIN -1
 
 //###Y_AXIS
 #define ORIG_Y_STEP_PIN 60
 #define ORIG_Y_DIR_PIN 61
 #define ORIG_Y_ENABLE_PIN 56
 #define ORIG_Y_CS_PIN -1
 
 //###Z_AXIS
 #define ORIG_Z_STEP_PIN 46
 #define ORIG_Z_DIR_PIN 48
 #define ORIG_Z_ENABLE_PIN 62
 #define ORIG_Z_CS_PIN -1
 
 //###EXTRUDER_0
 #define ORIG_E0_STEP_PIN 26
 #define ORIG_E0_DIR_PIN 28
 #define ORIG_E0_ENABLE_PIN 24
 #define ORIG_E0_CS_PIN -1
 #define ORIG_SOL0_PIN -1
 
 //###EXTRUDER_1
 #define ORIG_E1_STEP_PIN 36
 #define ORIG_E1_DIR_PIN 34
 #define ORIG_E1_ENABLE_PIN 30
 #define ORIG_E1_CS_PIN -1
 #define ORIG_SOL1_PIN -1
 
 //###EXTRUDER_2
 #define ORIG_E2_STEP_PIN -1
 #define ORIG_E2_DIR_PIN -1
 #define ORIG_E2_ENABLE_PIN -1
 #define ORIG_E2_CS_PIN -1
 #define ORIG_SOL2_PIN -1
 
 //###EXTRUDER_3
 #define ORIG_E3_STEP_PIN -1
 #define ORIG_E3_DIR_PIN -1
 #define ORIG_E3_ENABLE_PIN -1
 #define ORIG_E3_CS_PIN -1
[... 2014 lines cut ...]
 *****************************************************************************************/
//#define CUSTOM_USER_MENUS

#define USER_SCRIPT_DONE "M117 User Script Done"

#define USER_DESC_1 "Home & ABL"
#define USER_GCODE_1 "G28\nG29"

#define USER_DESC_2 "Preheat for PLA"
#define USER_GCODE_2 "M140 S" STRINGIFY(PREHEAT_1_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_1_TEMP_HOTEND)

#define USER_DESC_3 "Preheat for ABS"
#define USER_GCODE_3 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_2_TEMP_HOTEND)

#define USER_DESC_4 "Heat Bed/Home/Level"
#define USER_GCODE_4 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nG28\nG29"

#define USER_DESC_5 "Home & Info"
#define USER_GCODE_5 "G28\nM503"
/*****************************************************************************************/

#endif /* _CONFIGURATION_FEATURE_H_ */

I only selected laser, cartesian, board 1433, changed the table/printer size and the name. 
I'm using a MacBook so when I need to edit the files I use text editor app.

iosonopersia, 2017-08-25:
Please, please, please, and please
Delete your last comments: you can attach a file by drag and dropping in inside the comment area... It will be created a link to that file.
The way you did is unreadable...

jmlcatarino, 2017-08-25:
I tried that but it keeps saying it doesn't support that kind of file. Dang this makes me mad because I want to give all info I can to get the help I need. If I copy paste the files as text, would that be good or should I just zip them all?

iosonopersia, 2017-08-25:
Zip or change their extensions to .txt

jmlcatarino, 2017-08-25:
[Copy of config files.zip](https://github.com/MKFirmware/MK4duo/files/1252438/Copy.of.config.files.zip)

iosonopersia, 2017-08-25:
Ok, now it's way better! Well, I didn't read your configuration files yet but... in 1433.h you have made a mess...

 You now have to cancel out every line that starts with a "-" and then, delete every "+" at the beginning of the remaining lines. It would be faster to download that file again from the github repository. 

The last error you posted should be fixed with this. But you will have the same old errors caused by the lack of support for the LCDs. I have opened a pull request to add that support, be patient...

jmlcatarino, 2017-08-25:
Ok I just went to your hardware issue #202 and just copy/ paste the code. Maybe that went wrong. I'll just get the original one.

jmlcatarino, 2017-08-25:
Done, old error in place...

iosonopersia, 2017-08-25:
I've already told you that...

> The last error you posted should be fixed with this. But you will have the same old errors caused by the lack of support for the LCDs. I have opened a pull request to add that support, be patient...

Try using this version of the file (is the one I've modified for the pull request):
https://raw.githubusercontent.com/iosonopersia/MK4duo/e24554cdf83bf0284e2a1d5f213d4e6b1b3c03dc/MK4duo/src/boards/1433.h

You will have some trouble also with Configuration_Pins.h, I bet.
In lines 206 and 207 you have:
```
#define LASER_PWR_PIN            12       ORIG_LASER_PWR_PIN
#define LASER_PWM_PIN            13       ORIG_LASER_PWM_PIN
```

Replace with this:
```
#define LASER_PWR_PIN            12
#define LASER_PWM_PIN            13
```

jmlcatarino, 2017-08-25:
Ok, tried the fix from your link and it worked but there was a #endif without an #if, so I deleted one of them at the end. After deleting and running compile again, the same old error about pins not being declared.

iosonopersia, 2017-08-25:
Thank you so much for the valuable info, it seems I forgot a piece of code while opening the pull request. I will fix as soon as possible!

iosonopersia, 2017-08-25:
Please, retry:
https://raw.githubusercontent.com/iosonopersia/MK4duo/bdd3e7d0a2fe3144813979494ad3b44969add9ca/MK4duo/src/boards/1433.h

jmlcatarino, 2017-08-25:
Nope, now I have the other one I had before, the one "ultralcd_impl_HD44780.h", 'const byte clock [8]' redeclared as different kind of symbol
 const static PROGMEM byte clock[8] = { 

and so on...

iosonopersia, 2017-08-25:
That's good, we've fixed the majority of the errors: this last one is different from the others, it could require more time to solve, we will see...

jmlcatarino, 2017-08-25:
Do you need the entire error log? I've just posted the first lines, because it seemed the same as the before, which in fact we solved if I remember...

iosonopersia, 2017-08-25:
Put it in a .txt file and attach it here...

jmlcatarino, 2017-08-25:
[compile error.txt](https://github.com/MKFirmware/MK4duo/files/1253090/compile.error.txt)

iosonopersia, 2017-08-28:
My new pull request fixes the "clock variable error". A lot of other problems occur, however, caused by ultralcd_impl_HD44780.h. 
@MagoKimbra I don't know what's happening here... I think they are the same of #204.
[error_messages.txt](https://github.com/MKFirmware/MK4duo/files/1256473/error_messages.txt)

iosonopersia, 2017-09-03:
Hi @jmlcatarino , sorry for the delay. I've rechecked your configuration a few minutes ago and I've found that some recent edit by MagoKimbra fixed the compilation error you had. Now MK4duo compiles well, at least on my computer. Could you please try again and confirm me that (using the latest version, of course)?

Well, a firmware compiling is not necessarily a firmware working, but we're getting there...

jmlcatarino, 2017-09-03:
Hi @iosonopersia , thank you very much for your answer and help. It really made me happy.
As for your request, I did downloaded a new version and configured as usual. The old errors gone but now a new one. Should I close this Issue and open a new one or can I post it here?

iosonopersia, 2017-09-03:
Post it here

jmlcatarino, 2017-09-03:
@iosonopersia here it goes:

In file included from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_192990/sketch/src/commands/../gcode/gcode.h:42:0,
                 from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_192990/sketch/src/commands/commands.cpp:30:
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_192990/sketch/src/commands/../gcode/config/m200.h: In function 'void gcode_M200()':
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_192990/sketch/src/commands/../gcode/config/m200.h:45:5: error: 'tools' was not declared in this scope
     tools.volumetric_enabled = (parser.value_linear_units() != 0.0);
     ^
In file included from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_192990/sketch/src/commands/../gcode/gcode.h:50:0,
                 from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_192990/sketch/src/commands/commands.cpp:30:
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_192990/sketch/src/commands/../gcode/config/m221.h: In function 'void gcode_M221()':
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_192990/sketch/src/commands/../gcode/config/m221.h:37:28: error: 'tools' was not declared in this scope
   if (parser.seenval('S')) tools.flow_percentage[TARGET_EXTRUDER] = parser.value_int();
                            ^
In file included from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_192990/sketch/src/commands/../gcode/gcode.h:51:0,
                 from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_192990/sketch/src/commands/commands.cpp:30:
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_192990/sketch/src/commands/../gcode/config/m222.h: In function 'void gcode_M222()':
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_192990/sketch/src/commands/../gcode/config/m222.h:39:5: error: 'tools' was not declared in this scope
     tools.density_percentage[TARGET_EXTRUDER] = parser.value_int();
     ^

iosonopersia, 2017-09-07:
Those errors should now be fixed... Please try again, I really hope this time will work !!!

jmlcatarino, 2017-09-07:
Hi there @iosonopersia , here is the thing: If I choose 0 extruders there is still an error, but if I choose 1, for example, it gives me another type of error. 
This is the one for Zero extruders, as this is going for a laser machine:

In file included from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_451155/sketch/base.h:103:0,
                 from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_modified_sketch_202303/MK4duo.ino:261:
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_451155/sketch/src/mechanics/mechanics.h:112:61: error: too many initializers for 'float [0]'
               retract_acceleration[EXTRUDERS]       = { 0.0 },
                                                             ^

I'm getting you guys such a hard work on this one... :(

iosonopersia, 2017-09-07:
Ok, I think this can be fixed... Could you please send also the errors you get with EXTRUDERS > 0 ?

jmlcatarino, 2017-09-07:
Ok the first error was for the extruders 0, drivers 0, the next one is for extruders 1, drivers 1. 

[compile error new.txt](https://github.com/MKFirmware/MK4duo/files/1284421/compile.error.new.txt)

iosonopersia, 2017-09-07:
Well, that's an error caused by an external library (NewliquidCrystal), so we cannot do anything about that. Try installing again that library or updating it through Arduino IDE...

jmlcatarino, 2017-09-07:
Thank you @iosonopersia I'll check that out. But still, if the first one can solved, I'll appreciate. 
I'll get back to you after testing.

jmlcatarino, 2017-09-07:
@iosonopersia Ok, if I disable the eeprom and leave zero extruders and drivers, this is what comes up:In file included from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_451155/sketch/base.h:103:0,
                 from /var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_modified_sketch_276059/MK4duo.ino:261:
/var/folders/vf/g2tpy0pd48ldwvhk09n2b74r0000gn/T/arduino_build_451155/sketch/src/mechanics/mechanics.h:112:61: error: too many initializers for 'float [0]'
               retract_acceleration[EXTRUDERS]       = { 0.0 },
                                                             ^

But if I set to one extruder and one driver all the errors go away. So something with eeprom support I guess.

iosonopersia, 2017-09-07:
Ok, I talked with MagoKimbra and he told me that the firmware is still not ready 100% to be used with EXTRUDERS = 0, this is the cause of your error.

He suggests a workaround: you have to set 1 extruder and then you have to set the temp sensor relative to Hotend0 as a "Dummy thermistor", i'll post some screenshots of the online configurator just to be clear. This way, MK4duo will think that you've got an extruder which stays constantly at exactly 25 degrees celsius, so it won't never complain.

The eeprom errors you're receiving are very similar to these ones : #222, we will try as soon as possible to fix them.

**Now, could you please disable temporarily EEPROM and just check if you manage to get the firmware compiling and working on your machine? It would be a great satisfaction... I'm just curious :-)**

iosonopersia, 2017-09-07:
[image]
[image]

jmlcatarino, 2017-09-07:
Ok, I've already didi that and explained it before. If I disable eeprom and set extruder and driver to 1, everything compiles ok. Now, I'm trying to configuring a laser, so it doesn't make me much sense setting a dummy temp sensor. But I'll try that too. :)

iosonopersia, 2017-09-07:
I know, it's just a workaround to fool the firmware and make everything work as expected until full support will be added...

jmlcatarino, 2017-09-07:
Ok, now it compiles. Extruder 1, driver 1, dummy temp sensor and no eeprom.

jmlcatarino, 2017-09-07:
Well, I've uploaded and it seems to me that everything, except the buzzer, is ok. 
When I press the encoder button to access the menu, it just beeps continuously until I change within the menu. Now I have to put it attached to a frame a test it. But I'm sure it works ok. :)
Thank you very much for all your help. Should I close this issue or leave it until a permanent solution is found?

iosonopersia, 2017-09-07:
I think you should close this issue and open a new one reporting again only the eeprom problem you had, so we can focus better on that particular issue. I'm glad everything works fine, have a nice laser experience! ;-)

MagoKimbra, 2017-09-07:
Thanks @jmlcatarino , we will work to ensure that extruders can also be set to 0. But work is so far the only solution is that given by @iosonopersia .

jmlcatarino, 2017-09-07:
@iosonopersia , @MagoKimbra  guys, I can only thank you for your work in order to put this to work. I really didn't want to mess with Repetier firmware and this is the best solution for all the boards around to work with Due.

closed 2017-09-07 by jmlcatarino

MagoKimbra, 2017-09-07:
Thanks to you, and make good publicity, more feedback more quality to can get.

jmlcatarino, 2017-09-07:
Already did it on the RepRap facebook group and as long as I can. I only publicise things I like or know that work. ;)

iosonopersia, 2017-09-08:
Hi @jmlcatarino, the eeprom error is only caused by the fact that RAMPS4DUE has no EEPROM in it. No error in the code. Now, I'm not an electronics guy (I don't know what SPI and I2C are) but there are two ways to connect to an EEPROM with MK4duo: via SPI (defining SPI_EEPROM) and via I2C (defining I2C_EEPROM). Looking at the code, I suppose you'll need to use SPI_EEPROM, but you'll also need to define some other pins. When you'll need to connect your external EEPROM to your board, open a new issue and we'll try to help you defining the required pins. Enjoy our firmware!

jmlcatarino, 2017-09-08:
@iosonopersia ,I remember to have seen somewhere how to adapt an external I2C EEPROM to Ramps, because it has dedicated pins for that. I just have to search how to implement that on firmware. SPI and I2C are just ways of communication that allow us to have more devices connected  with less pins or wires. I'll give a look at some old post on repraps forum for that.

iosonopersia, 2017-09-08:
Another user in another issue told us about this: https://www.instructables.com/id/Adding-an-24LC256-EEPROM-to-the-Arduino-Due/
Maybe it can help you